JMRI/JMRI #5199: PanelPro Zoom seems not to handle large panels correctly

simongray67, 2018-04-04:
Running 4.10+R419243e on OSX High Sierra (10.13.3) MacBook Pro 15" (2880x 1800)

I have a large Layout Panel (4000w x 2250h) describing the whole layout. 
When viewed at Zoom factor 1.0x (or No Zoom), the scroll bars can be used to move around and see the whole panel correctly rendered.
When I zoom out, by say, selecting Zoom factor 0.5x - the displayed track does indeed shrink (as you would expect), but only the top left quadrant of the panel is rendered, with the rest being white space.

Using the location co-ordinates from the layout editor screen, the last bottom right corner pixel that seems to be rendered (note this is approximate as 'eye'-balled from screen) is as follows:

@0.75x {3010,1690}
@0.5x {2006,1126}
@0.25x {1660, 872}

Prior versions of layout editor did not display this behaviour; it was possible to apply zoom to fit the whole layout on the screen.  It seems like the zoom factor is being applied to the panel size to determine the rendered portion - eg 0.75x 4000 = ~3000 etc.

Is this a bug with the new release, or is my elderly layout xml file past its sell-by date?

Many thanks for any assistance.

bobjacobsen, 2018-04-04:
Please try 4.11.4 which has lots of changes in this area.

>  is my elderly layout xml file past its sell-by date?

If 4.11.4 doesn’t fix it, could you please try:

 - make a backup copy of your panel file
 - load the file, and store it back out (to make a new version)
 - see if restarting PanelPro and loading that new file is better.

Thanks, and sorry for the trouble.

simongray67, 2018-04-04:
Bob

Many thanks - have downloaded 4.11.4 - rendering is much crisper, but issue remains.

I also re-stored the file, but no improvement.

I attach the xml file in case that helps you at all.

Simon

This email and any attachments contain information which is confidential and may also be privileged. It is for the exclusive use of the intended recipient(s). Any confidentiality or privilege is not waived or lost if this email has been sent to you by mistake. If you are not the intended recipient, you must not disclose or use the information contained in it and should notify the sender immediately by return email and delete all copies of this email and all attachments to it from your systems. This email is also subject to copyright and no part of it should be reproduced, adapted or communicated without the written consent of the copyright owner. Any personal information in this email must be handled in accordance with any relevant personal data protection laws and regulations. Opinions, conclusions and other information in this email and all attachments which do not relate to the official business of the firm are neither given by nor endorsed by it.??

Emails may be interfered with, may contain computer viruses or other defects and may not be successfully replicated on other systems. We give no warranties and accept no liability in relation to these matters.??

Arcus Infrastructure Services LLP is a limited liability partnership registered in England and Wales with registered number OC344797, registered office 2nd Floor, 6 St Andrew Street, London EC4A 3AE and is authorised and regulated by the FCA. Please email info@arcusip.com if you need assistance.

simongray67, 2018-04-05:
Bob

Playing with 4.11.4 last night, if I 'zoom to fit', the top left quadrant is the only part displayed as per prior message.  However, if I turn on grid lines,  the whole layout is rendered correctly.  Turning off gridlines reverts to only rendering the top left quadrant.  

Thank you for looking at the issue.
Simon

simongray67, 2018-06-14:
Bob

In an attempt to move this on, I had some time and took a look at the code for the layout editor.  in particular I was trying to understand the different flow for the rendered panel with grid (which works at all zoom levels) and the rendered panel without (which does not as above).

In 'drawpanel'  round about 9492, there is a test for the grid and if needed a call to 'drawpanelgrid'.
Looking at drawpanelgrid, round about 10034, there is a call to set the graphics2D clip (having calculated it). 

If there is no grid to render, I can't find a similar call for the straight 'element rendering' chain.  Is this 'setclip' call and its absence in a no-grid situation a factor in (or cause of) the issue highlighted above?

geowar1, 2018-06-14:
That set clip code was added by me in an (unsuccessful) attempt to fix an underclipping problem. You’ll probably notice that the scroll bars are getting overdrawn now (when the grid is visible).

closed 2021-04-14 by stale[bot]