JMRI/JMRI #5204: JMRI locks up

Ted-IC, 2018-04-06:
I have been having a recurring problem with JMRI locking up during operating sessions.  A dispatcher panel using Panel Pro (version 4.10+R419243e with version 1.8.0 for Java) is used to throw mainline switches and to keep track of train locations using manually activated sensors.  The track diagram is on an external monitor for a laptop (Windows 10) that communicates at 9600 baud with an NCE DCC system having the latest EPROM.  There have been anywhere from six to nine throttles (ProCab-Rs) in use when the lockup occurs. 

I had an operating session on 04/04/18 with 6 throttles operating.  JMRI locked up at 3.5 hr into the session.  As per Dave’s suggestion in an email on 03/10/18, I recycled the power to the NCE command station, waiting about 20 sec to turn the power back on.  JMRI did not respond.  The laptop and NCE DCC system were still working as normal.  I had to use the Task Manager in Windows 10 to end JMRI.  Because JMRI did not close normally and the Com port was still “open”, I restarted the laptop and then opened JMRI.  JMRI then worked to the end of the session.

The locking up does not seem to correlate to the number of throttles in use, presence of shorts, or volume of activity at the dispatcher panel.

Do you have any other suggestions to prevent the locking up of JMRI?

Ted-IC

ClockLoop, 2018-04-07:
I have a dcs240 and if anything happens to the serial connection (i pull the usb cable out of the command station or computer, jmri will look like its operating properly, but no serial data will move.

And JMRI dosen't even do anything about the disconnection or connection issue.  
If I plug the usb cable back in without first turning off jmri, jmri will act flaky.

You are right in that jmri dosen't deal with serial connection issues very well.
I would suggest trying a powered usb hub, some computers internal usb hubs are messed up.

The usb hub will allow you to remove the usb hub from the system once a problem starts, possibly helping you from having to restart the pc, only needing to close jmri, unplug the hub, and plug it back in and restart jmri.

I too wish jmri had better serial port problem "handling" but I suspect this might be kinda related to how java talks to the os when it comes to serial interfaces...  

Most programs have a way to open and close the serial port with a button in the program, for exactly these reasons, to do more ACTIVE control of the serial port from the interface not being that great to begin with.

See if you can find/borrow a self powered usb hub and try it, or if you are already using a usb hub, remove it, or get a different one to test with.   Make sure its self powered!!!!  (this means it has its own wall wart power supply)

How long and what kind of quality of usb cable are you using?    Is the pc powered by the same outlet strip (and ground) that your layout is connected to, or are you hooking up your pc and layout to wall plugs that have to go all the way back to the power panel to make a link?  (use the pc and entire layout on the same power strip)  

Have you tried using power strips or even power filtering devices like https://www.amazon.com/Tripp-Lite-Interactive-Ultra-Compact-AVR550U/dp/B000BMBT4I   (and make sure any pc device or device on the layout or anything powering any part of the layout all must connect at the same power strip, or eventually all connect to the same single outlet in the wall using a chain of power strips (observing current limits naturally)

IF I even bump my usb cable connection, jmri will not recover from the connection loss, and requires a full program shutdown, PLUS I must unplug the usb cable, and power off my DCS240, then power on the dcs240, and then plug in the usb cable and then start jmri.

I never have had to restart the pc to get the usb port working again tho.
I don't use windows10 I use windows 7.
I suspect this is more a win10 subsystem issue, rather than jmri.

Its possible java.exe is still running in the processes list even tho jmri has been closed, which would be what is keeping the serial port in a locked up state, making it look like the pc needs a restart, when in fact the java.exe program needs to be terminated in the task list after you close jmri...

BUT having a way to close/open the serial interface in jmri with a button might help here, then the OP might find that simply pressing the close port button forces jmri to run code that will for sure close the JAVA SIDE port, instead of leaving it open, even upon a restart of jmri, thus jmri sees the port is in use and the pc requires a restart.

This might also make it possible to restart jmri, and have it see the port is in use and give the user a warning and even an option to force it closed?   I dont know java programming so i don't know what is possible here with the limits of java and serial port interfacing/control.

dheap, 2018-04-07:
If you pull the USB Cable the driver is unloaded and the port disappears altogether. 

That's a pretty drastic OS-level event that pushes up through the PJC library to JMRI. I'm doubtful JMRI code could be written to recover fully from that in the current session.

Dave

bobjacobsen, 2018-04-08:
> requires a full program shutdown, PLUS I must unplug the usb cable, and power off my DCS240, then power on the dcs240, and then plug in the usb cable and then start jmri.

That sequence pretty clearly demonstrates that the problem isn't in JMRI.

Ted-IC, 2018-04-08:
ClockLoop and Dave,

Thanks for your comments.

The next time JMRI locks up, I will end JMRI, unplug the USB cable from the laptop, plug the USB cable back in, and restart JMRI.  Maybe this will work better than what I have been doing.

I am looking into a self-powered USB 2.0 Hub.

For an operating session, the cable from the command station to the laptop consists of a 6 ft RS232 extension cable, a 6 ft RS232 to USB Keyspan cable, a 6 ft USB extension cable, and a 32 ft BlueRigger USB 2.0 active extension cable.  The laptop is powered from a circuit that is separate from that for the DCC system.

Ted-IC

bobjacobsen, 2018-04-08:
> For an operating session, the cable from the command station to the laptop consists of a 6 ft RS232 extension cable, a 6 ft RS232 to USB Keyspan cable, a 6 ft USB extension cable, and a 32 ft [...]

Interesting setup!  I have no reason to suspect that the BlueRigger extension cable is the problem, but it is a bit unusual setup.  Is there a reason you need to use that instead of extending the RS232 side?

Ted-IC, 2018-04-08:
Bob,

The 6 ft RS232 extension cable is plugged into the command station because the RS232 end of the RS232 to USB Keyspan cable will not plug into the command station because of the size/shape of the command station chassis.  I went with USB cables from there on because I thought that the USB might be more robust than the RS232.

Ted-IC

Ted-IC, 2018-05-04:
Dave in Australia,
I had an operating session on Wednesday, May 2 and did not experience any lockup of JMRI.  Three observations:
1.  A few days prior to the operating session, I installed a 4-port USB 3.0 hub with power adapter at the USB end of the Keyspan cable.  The cabling from the command station to the laptop now consists of a 6 ft RS232 extension cable, a 6 ft Keyspan RS232 to USB cable, the USB hub, a 6 ft USB extension cable, and a 32 ft BlueRigger USB 2.0 active extension cable.
2.  During the operating session, there were 7 wireless throttles active (normally there are 8).
3.  Some of the operators were switching in areas that required minimum dispatcher involvement, so the level of communication between the laptop and the command station was less than usual.
I will continue to monitor the lockup problem and will let you know if it occurs again.
Your help is greatly appreciated.
Ted-IC

closed 2018-09-23 by bobjacobsen