JMRI/JMRI #5210: Move sensor from Internal to LCC sensor generates stack dump

KenC57, 2018-04-06:
Here is the dump. It prevents further interaction with the panel. The move seems to show correctly in the table and when you write the panel file.

2018-04-06 18:19:39,612 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.IllegalArgumentException: name cannot be empty in getNamedBeanHandle
	at jmri.NamedBeanHandleManager.getNamedBeanHandle(NamedBeanHandleManager.java:61)
	at jmri.jmrit.display.layoutEditor.LayoutBlock.getOccupancy(LayoutBlock.java:609)
	at jmri.jmrit.display.layoutEditor.LayoutBlock.getBlockColor(LayoutBlock.java:454)
	at jmri.jmrit.display.layoutEditor.LayoutTrack.getColorForTrackBlock(LayoutTrack.java:146)
	at jmri.jmrit.display.layoutEditor.LayoutTrack.setColorForTrackBlock(LayoutTrack.java:159)
	at jmri.jmrit.display.layoutEditor.LayoutTrack.setColorForTrackBlock(LayoutTrack.java:166)
	at jmri.jmrit.display.layoutEditor.TrackSegment.draw1(TrackSegment.java:2095)
	at jmri.jmrit.display.layoutEditor.LayoutEditor.draw1(LayoutEditor.java:9807)
	at jmri.jmrit.display.layoutEditor.LayoutEditor.drawLayoutTracksBlockLines(LayoutEditor.java:9743)
	at jmri.jmrit.display.layoutEditor.LayoutEditor.draw(LayoutEditor.java:9504)
	at jmri.jmrit.display.layoutEditor.LayoutEditorComponent.paint(LayoutEditorComponent.java:41)
	at javax.swing.JComponent.paintChildren(Unknown Source)
	at javax.swing.JComponent.paint(Unknown Source)
	at javax.swing.JLayeredPane.paint(Unknown Source)
	at jmri.jmrit.display.Editor$TargetPane.paint(Editor.java:643)
	at javax.swing.JComponent.paintChildren(Unknown Source)
	at javax.swing.JComponent.paint(Unknown Source)
	at javax.swing.JViewport.paint(Unknown Source)
	at javax.swing.JComponent.paintChildren(Unknown Source)
	at javax.swing.JComponent.paint(Unknown Source)
	at javax.swing.JComponent.paintChildren(Unknown Source)
	at javax.swing.JComponent.paint(Unknown Source)
	at javax.swing.JComponent.paintChildren(Unknown Source)
	at javax.swing.JComponent.paint(Unknown Source)
	at javax.swing.JLayeredPane.paint(Unknown Source)
	at javax.swing.JComponent.paintChildren(Unknown Source)
	at javax.swing.JComponent.paintToOffscreen(Unknown Source)
	at javax.swing.RepaintManager$PaintManager.paintDoubleBuffered(Unknown Source)
	at javax.swing.RepaintManager$PaintManager.paint(Unknown Source)
	at javax.swing.RepaintManager.paint(Unknown Source)
	at javax.swing.JComponent.paint(Unknown Source)
	at java.awt.GraphicsCallback$PaintCallback.run(Unknown Source)
	at sun.awt.SunGraphicsCallback.runOneComponent(Unknown Source)
	at sun.awt.SunGraphicsCallback.runComponents(Unknown Source)
	at java.awt.Container.paint(Unknown Source)
	at java.awt.Window.paint(Unknown Source)
	at javax.swing.RepaintManager$4.run(Unknown Source)
	at javax.swing.RepaintManager$4.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at javax.swing.RepaintManager.paintDirtyRegions(Unknown Source)
	at javax.swing.RepaintManager.paintDirtyRegions(Unknown Source)
	at javax.swing.RepaintManager.prePaintDirtyRegions(Unknown Source)
	at javax.swing.RepaintManager.access$1200(Unknown Source)
	at javax.swing.RepaintManager$ProcessingRunnable.run(Unknown Source)
	at java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$500(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)

KenC57, 2018-04-07:
If I hide the LE panel, so I can't mouse over it, but then go to save after making some moves (Internal to LCC), when I do the save panel, I get:

2018-04-06 20:15:30,553 configurexml.ErrorHandler             ERROR - Exception thrown while storing to file in addConfigStore Exception: java.lang.NullPointerException
See http://jmri.org/help/en/package/jmri/configurexml/ErrorHandler.shtml for possibly more information. [AWT-EventQueue-0]
java.lang.NullPointerException
	at jmri.managers.configurexml.AbstractSensorManagerConfigXML.store(AbstractSensorManagerConfigXML.java:63)
	at jmri.managers.configurexml.InternalSensorManagerXml.store(InternalSensorManagerXml.java:55)
	at jmri.configurexml.AbstractXmlAdapter.store(AbstractXmlAdapter.java:81)
	at jmri.configurexml.ConfigXmlManager.elementFromObject(ConfigXmlManager.java:525)
	at jmri.configurexml.ConfigXmlManager.elementFromObject(ConfigXmlManager.java:505)
	at jmri.configurexml.ConfigXmlManager.addConfigStore(ConfigXmlManager.java:287)
	at jmri.configurexml.ConfigXmlManager.storeUser(ConfigXmlManager.java:479)
	at jmri.implementation.JmriConfigurationManager.storeUser(JmriConfigurationManager.java:155)
	at jmri.configurexml.StoreXmlUserAction.actionPerformed(StoreXmlUserAction.java:56)
	at javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
	at javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
	at javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
	at javax.swing.DefaultButtonModel.setPressed(Unknown Source)
	at javax.swing.AbstractButton.doClick(Unknown Source)
	at javax.swing.plaf.basic.BasicMenuItemUI.doClick(Unknown Source)
	at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(Unknown Source)
	at java.awt.Component.processMouseEvent(Unknown Source)
	at javax.swing.JComponent.processMouseEvent(Unknown Source)
	at java.awt.Component.processEvent(Unknown Source)
	at java.awt.Container.processEvent(Unknown Source)
	at java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$500(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)
2018-04-06 20:15:41,747 configurexml.ErrorHandler             ERROR - Exception thrown while storing to file in addConfigStore Exception: java.lang.NullPointerException
See http://jmri.org/help/en/package/jmri/configurexml/ErrorHandler.shtml for possibly more information. [AWT-EventQueue-0]
java.lang.NullPointerException
	at jmri.managers.configurexml.AbstractNamedBeanManagerConfigXML.storeUserName(AbstractNamedBeanManagerConfigXML.java:89)
	at jmri.managers.configurexml.AbstractNamedBeanManagerConfigXML.storeCommon(AbstractNamedBeanManagerConfigXML.java:41)
	at jmri.managers.configurexml.AbstractTurnoutManagerConfigXML.store(AbstractTurnoutManagerConfigXML.java:73)
	at jmri.configurexml.AbstractXmlAdapter.store(AbstractXmlAdapter.java:81)
	at jmri.configurexml.ConfigXmlManager.elementFromObject(ConfigXmlManager.java:525)
	at jmri.configurexml.ConfigXmlManager.elementFromObject(ConfigXmlManager.java:505)
	at jmri.configurexml.ConfigXmlManager.addConfigStore(ConfigXmlManager.java:287)
	at jmri.configurexml.ConfigXmlManager.storeUser(ConfigXmlManager.java:479)
	at jmri.implementation.JmriConfigurationManager.storeUser(JmriConfigurationManager.java:155)
	at jmri.configurexml.StoreXmlUserAction.actionPerformed(StoreXmlUserAction.java:56)
	at javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
	at javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
	at javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
	at javax.swing.DefaultButtonModel.setPressed(Unknown Source)
	at javax.swing.AbstractButton.doClick(Unknown Source)
[... 8 lines cut ...]
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
[... cut ...]
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)

bobjacobsen, 2018-04-07:
Could you provide  `diff` of the before and after panel files, please?

closed 2021-04-14 by stale[bot]