JMRI/JMRI #5216: Some NamedBean classes improperly implement toString()

bobjacobsen, 2018-04-08:
(See #5180 for initial issue, PR#5215 for an initial fix)

The NamedBean class documents toString() as providing the system name, and lots of code counts on that.  But some classes re-implement the AbstractNamedBean method to do their own thing.

These classes (and perhaps others) need to have their toString methods removed and appropriate other code updated (including tests) so that AbstractNamedBean.toString can be made 'final':

 - [ ] jmri/implementation/AbstractIdTag
 - [ ] jmri/implementation/DefaultRailCom
 - [ ] jmri/implementation/DefaultSignalSystem
 - [ ] jmri/jmrit/logix/Portal
 - [ ] jmri/jmrit/audio/AbstractAudioSource

pabender, 2018-04-09:
@bobjacobsen 

Changing the IdTag objects (and RailCom objects, because they are IdTag objects ) is going to necessitate a change in some of the Reporter report handling functionality, both for the tables and panels.

As you know, Reporters report Objects, and the first implementation (the LocoNet implementation) used Strings as the objects reported.

Of the 9 Reporter types in jmri.jmrix, 6 use a derivative of the IdTag interface as their reports.  ( the exceptions are LocoNet, RPS, and the JMRI Client, which all use strings).  

The current functionality of the IdTag is that toString (which ends up as the report) will present the user name (typically reporting marks, but not always) if available and the system name (which is the the tag id with a prefix) if there is no user id.

Railcom objects are a bit more complicated, and can carry more information, so they display that information as part of their string representation.

Incidentally, IdTag objects are never stored in panel files.  We store them in separate files for the same reasons rosters are separate from layout data.

bobjacobsen, 2018-04-09:
@pabender Thanks!  I agree that it's going to be some work.  

Defining the toString to get the system name is being used in the various picking tools, the standard table, etc.  So either way, it's going to take a bit of work to get something consistent.

Not sure what the right direction is.  Maybe IdTags shouldn't be NamedBeans?  Not everything is; perhaps they really are more similar to RosterEntries than NamedBeans.  Or maybe the Report-to-IdTag connection is the place to work?  

Right now, the GUI issue for Reporters is fixed, so further work in this area might not be a priority until somebody encounters something that's behaving sufficiently oddly.

bobjacobsen, 2018-04-09:
@pabender That pretty much makes the case for NamedBean-ness then.

Another alternative would be to change the NamedBean.toString() contract, and change all the code that uses toString() to get a system name to instead use getSystemName().  The problem is finding them all:  The usage in JTables, for example, is from feeding an `Object extends NamedBean` to a Swing-standard Renderer class, which takes the value from Object.toString().  We could certainly provide NamedBean-specific replacements for those, and then toString() would be available for other uses.  We'd just have to track them all down somehow...

geowar1, 2018-04-09:
On Apr 8, 2018, at 9:24 PM, Bob Jacobsen <notifications@github.com> wrote:
> @pabender <https://github.com/pabender> That pretty much makes the case for NamedBean-ness then.
> Another alternative would be to change the NamedBean.toString() contract, and change all the code [...]

XXX.toString() is used by NetBeans when mousing over variables while debugging. The getDisplayName method is preferred for this over getSystemName() (for the classes that have usernames).

-- 
Enjoy,
George Warner
eMail, Message, or FaceTime: <geowar1@mac.com>

closed 2021-04-14 by stale[bot]