JMRI/JMRI #5223: can't read Loconet stationary decoder in Ops Mode in 4.11.5

n3ix, 2018-04-09:
Installed the latest Jenkins build (4.11.5ish) and opened the roster entry for a QuadLN_S Loconet Stationary Decoder in the Operations Mode (Main Track) programmer.  The READ buttons are disabled on all the tabs except for the CVs tab.  This decoder supports Ops Mode READ over Loconet and the READ buttons should be active on all tabs.

bobjacobsen, 2018-04-10:
Decoders that work via a LocoNet cable, i.e. don't need transponding receivers, should have the following in their decoder definition to indicate that:
```
    <programming direct="no" paged="yes" register="no" ops="no">
        <mode>LOCONETOPSBOARD</mode>
    </programming>
```

Alternately, it should work to go into the preferences and say that transponding hardware is installed.

I'd be happy to change the files. Is it all of these?
```
xml/decoders/TamValleyDepot_QuadLn.xml
xml/decoders/TamValleyDepot_QuadLn11.xml
xml/decoders/TamValleyDepot_QuadLn_S_11.xml
xml/decoders/TamValleyDepot_QuadLn_S_11.xsl
```
Any others?

n3ix, 2018-04-10:
Yes I tried that.

bobjacobsen, 2018-04-10:
Just to see whether it works, could you try the RR-CirKits Motorman decoder?  That has the same clause.  Thanks

n3ix, 2018-04-10:
When I open a decoder with the changes in the definition, I do see this message in the system console:

2018-04-10 08:43:22,580 tabbedframe.PaneOpsProgFrame          INFO  - Programming mode selected: LocoNet-Attached Board (LOCONETOPSBOARD) [AWT-EventQueue-0]

However the READ buttons are not enabled. 

Your question about the Motorman seems to have uncovered another problem.  Sorry.  It seems that after Roster->CreateEntry, select _any_ decoder type and then click the Open Programmer button I get an NPE.  For example:

2018-04-10 08:48:40,675 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.NullPointerException
	at jmri.jmrit.symbolicprog.VariableTableModel.setButtonModeFromProgrammer(VariableTableModel.java:628)
	at jmri.jmrit.symbolicprog.tabbedframe.PaneProgPane.setCvListFromTable(PaneProgPane.java:2191)
	at jmri.jmrit.symbolicprog.tabbedframe.PaneProgPane.makeCvTable(PaneProgPane.java:2180)
	at jmri.jmrit.symbolicprog.tabbedframe.PaneProgPane.newColumn(PaneProgPane.java:1533)
	at jmri.jmrit.symbolicprog.tabbedframe.PaneProgPane.<init>(PaneProgPane.java:268)
	at jmri.jmrit.symbolicprog.tabbedframe.PaneProgFrame.newPane(PaneProgFrame.java:1215)
	at jmri.jmrit.symbolicprog.tabbedframe.PaneProgFrame.readConfig(PaneProgFrame.java:959)
	at jmri.jmrit.symbolicprog.tabbedframe.PaneProgFrame.loadProgrammerFile(PaneProgFrame.java:789)
	at jmri.jmrit.symbolicprog.tabbedframe.PaneProgFrame.<init>(PaneProgFrame.java:465)
	at jmri.jmrit.symbolicprog.tabbedframe.PaneNewProgAction$1$1.<init>(PaneNewProgAction.java:76)
	at jmri.jmrit.symbolicprog.tabbedframe.PaneNewProgAction$1.startProgrammer(PaneNewProgAction.java:74)
	at jmri.jmrit.symbolicprog.CombinedLocoSelPane.openNewLoco(CombinedLocoSelPane.java:611)
	at jmri.jmrit.symbolicprog.CombinedLocoSelPane.openButton(CombinedLocoSelPane.java:566)
	at jmri.jmrit.symbolicprog.CombinedLocoSelPane$5.actionPerformed(CombinedLocoSelPane.java:268)
	at javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
	at javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
	at javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
	at javax.swing.DefaultButtonModel.setPressed(Unknown Source)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(Unknown Source)
	at java.awt.AWTEventMulticaster.mouseReleased(Unknown Source)
	at java.awt.Component.processMouseEvent(Unknown Source)
	at javax.swing.JComponent.processMouseEvent(Unknown Source)
	at java.awt.Component.processEvent(Unknown Source)
	at java.awt.Container.processEvent(Unknown Source)
	at java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$500(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)

bobjacobsen, 2018-04-10:
What LocoNet connection variety do you have defined?  What command station type?

n3ix, 2018-04-10:
DCS240 as Command Station with PR3 as the System Connection.

n3ix, 2018-04-10:
I tried a Loconet Sim as the Connection and I still get the NPE when trying to create a new entry.  This new problem appears to be completely separate from the subject of this Issue, maybe I should create a separate one?

n3ix, 2018-04-10:
Back on the Ops Mode issue.  I opened an existing RR Cirkits TC64 roster entry in Ops Mode.  The "Read changes on sheet" and "Read full sheet" buttons are disabled on all tabs that have those buttons.  These were the only buttons I had been checking previously.    Looking further I see that the "Read changes on all sheets" and "Read all sheets" buttons are enabled.  I re-checked the QuadLN_S and QuadLN and the but behavior is the same as the TC64.  So the issue is the "Read changes on sheet" and "Read full sheet" buttons.

n3ix, 2018-04-10:
More data.  When I open the roster entry for the Decoder the "Read changes on sheet" and "Read full sheet" are disabled (grayed out) on all the tabs.  However "Read changes on all sheets" and "Read all sheets" are enabled.  If I click "Read changes on all sheets" the two Read buttons that were disabled are now enabled!  Now if I navigate to other tabs all the buttons are enabled.  Thought this info might be useful.

n3ix, 2018-04-11:
@bobjacobsen  I think this will help narrow it down even more.  While supporting a customer I just tried using a Loconet Sim profile.  The problem is not present here -  all buttons come up enabled.

bobjacobsen, 2018-04-11:
Thanks for all the work on this!  It's starting to look like a sequencing or timing issue.  I'll check it in a day or two when I get back to hardware. Should have a fix before this weekend's release.

PUlvestad, 2018-04-11:
I found that a recent change elsewhere requires a change in preferences Additional Connection Settings to enable Transponding.

bobjacobsen, 2018-04-11:
Thanks! Robin’s PR #5229 (just merged, Jenkins dev builds soon) updated those.

PUlvestad, 2018-04-11:
I just updated to the latest build from a few minutes ago and no longer see the issue. I can now have transponding present button set to the default NO.

n3ix, 2018-04-15:
Tested Jenkins build 2422 and 2424.  2422 runs ok, shows same issue with Read buttons as before.  2424 installed ok, the first time I attempted to open decoder for programming it had some type of roster index exception  which took maybe 60 seconds.  Subsequent restarts did not show this exception.  Still shows same issue with Read buttons as before.

n3ix, 2018-05-02:
Tested latest Jenkins build 2448.  In addition to the same issue with the buttons noted above, the Read Sheet is now failing.  The Loconet Monitor shows the expected responses from the decoder.  Tried setting Transponding option to ON and the Read Sheet still failed.  Defaults are all set correctly to LocoNet.

n3ix, 2018-05-16:
@bobjacobsen  That is exactly what I see.  BTW if I click on "Read changes on all sheets" then both of previously disabled buttons become enabled.  I will test this as soon as there is a new Jenkins build.  As always, thank you for taking care of this!

Just FYI Issue 5300 covers a different Ops Mode Read glitch I ran into when track power is off.  That one also seems to affect the Loconet Stationary Interrogate behavior during JMRI start up.

n3ix, 2018-05-16:
@bobjacobsen I just opened a QuadLN_S roster entry in the Ops Mode programmer and noticed that the ToolTip help on all the fields has "(Hardware connect read)" appended to it.  If this is meant to indicate that you cannot do a Read on that field then it is incorrect.

closed 2018-09-17 by bobjacobsen