JMRI/JMRI #5226: HDL LocoIO definition

silverailscolo, 2018-04-10:
I'd like to expand the DIY Hans Deloof LocoIO/LocoServo [Public_Domain_HDL_LocoIO.xml](https://github.com/JMRI/JMRI/blob/master/xml/decoders/Public_Domain_HDL_LocoIO.xml) decoder definition. Just assembled a new board so can test. Plan to base it on the ca 2015 work by @bobjacobsen and @nigelcliffe script. Will definitely have some questions on the SV1 message format.
Planned steps:

- [x] Document LocoIO set-up options and corresponding SV values
- [ ] Build Ports pane and attach ``enum`` lists (in progress)
- [ ] Compose LocoNet message (CV > SV) from values
- [ ] Read board values and display in pane (possible using Ops mode?)

Would like to learn from what others have already discovered in this area.

nigelcliffe, 2018-04-10:
Back, circa 2015 (can’t remember the date),  I made a start on them, but got bogged down on the conditions for the SV values.
The SVs within the LocoIO (and LocoServo) are combinations of multiple factors, which quickly got out of hand when I was trying to make things work within the constraints of a Decoder Definition file, and my knowledge of what was possible.

So, I gave up, and went back to my old Excel spreadsheet for calculating SV values for the boards, and just sending those changes via the single CV/SV programmer window.  If you’d like a copy of my spreadsheet, I’ll look around my old computer for it.

The nature of these boards is that I build a few when doing work on a layout, then they are installed and never need changing.  Once a layout is constructed its rare to add a board.  So, a burst of interest when doing work on a layout’s infrastructure, then nothing else.  Somewhat different to loco decoders where new locos are added to most people’s layouts reasonably frequently.    Going forwards, I expect my next LocoIO’s might be done using the Arduino options, several examples on the internet.  I don’t know if those have differences in how SVs work compared to the Hans De Loof designs.

Regards

Nigel

silverailscolo, 2018-04-10:
Thanks, @nigelcliffe. I recognize the dynamics of programming/configurement interest. I've found your spreadsheet, as you may remember I played with just the interface a couple of years before. I've also updated your script for the current JMRI reqs. If that's OK with you I might add it here to make these items retreavable in one place any time in the future.

silverailscolo, 2018-04-13:
Using a split value to fill in 2 SV values from one enum drop down is not working. It picks up the lower byte, but does not update the high byte. This is the structure under development:
[image]
Here's a snippet of the xml:
```
      <!-- Port 1 -->
      <variable item="Port1 Type" CV="3" mask="VXXXXXXX" default="0">
        <!-- note: the value is shifted to the left to fit in the mask, so "1" + mask = 128 -->
        <xi:include href="http://jmri.org/xml/decoders/parts/enum-InputOutput.xml"/>
        <label>Port Type:</label>
        <label xml:lang="nl">Poorttype:</label>
      </variable>

      <!-- Input example options -->
      <variable item="Port1 Input" CV="5" mask="VVVVXXXX">
        <splitVal highCV="3" factor="128"/>
        <enumVal>
          <enumChoice choice="Input Toggle Direct" value="1793"> <!-- 0000011100000001 -->
            <choice>Toggle switch</choice>
          </enumChoice>
          <enumChoice choice="Input Toggle Indirect" value="3841"> <!-- 0000111100000001 -->
            <choice>Toggle switch - Indirect</choice>
          </enumChoice>
        </enumVal>
        <label>Attached to:</label>
        <label xml:lang="nl">Aansluiting:</label>
      </variable>
```
CV5 is filled in, but CV3 isn't. Should I always use an offset or factor to copy the upper 8 bits from my 2 byte value into CV3? See my [loco-io](https://github.com/silverailscolo/JMRI/blob/eb-loco-io/xml/decoders/Public_Domain_HDL_LocoIO.xml) branch. Any help appreciated.

bobjacobsen, 2018-04-13:
I think `splitVal` and `enumVal` are mutually exclusive within a single `variable` element.  We may have to think of another approach...

Why do those two values set so many bits?  It looks like there's only one with content.

nigelcliffe, 2018-04-14:
How an individual value (SV) is constructed depends on adjacent SV values.  The complexity is only seen by looking over the full set of possible settings for the LocoIO.

My memory of doing this some years ago is that a single setting is possible with the constructions available in decoder files.  But, that arrangement would then be incompatible with another setting.    And there is a comment in my code about the incompatibility of using splitVal and enumVal in the same variable

Hence my decision to go back to my spreadsheet – even defining what new code would be needed within the decoder files was going to be complicated.
I did produce a script (below) which works for me, but still needs the spreadsheet.  That gave me a user interface for setting the SV’s associated with board address, and also the three SV’s associated with each of the 16 in/out channels.  Worked in conjunction with the LocoNet monitor to read results.  But it’s a long way from a bit of JMRI production code !

-          Nigel

--------Jython script follows---------------

# Script to set SV values in LocoIO and LocoServo boards
#
# First version works at a basic level; will set any SV value.
# But user has to calculate correct Decimal Value for all settings manually.
# Ideally one would have a tool with checkboxes/drop downs to determine the correct values.
#
# Copyright Nigel Cliffe 2010.
# Code in part derived from scripts in JMRI Jython directory.

import java
import javax.swing
import jarray
import jmri

class LocoNetCV(jmri.jmrit.automat.AbstractAutomaton) :
    # init() is called exactly once at the beginning to do
    # any necessary configuration.
    def init(self):

                                return

    # handle() will only execute once here, to run a single test
    # Modify this to do your calculation.
    def handle(self):

                                return 0

    # define what button does when clicked and attach that routine to the button
    def whenSendClicked(self,event) :
                                return

# [E5 10 50 52 01 08 01 30 00 00 10 01 00 00 00 21]  LocoBuffer => LocoIO@52/1 Write SV48=0x80
# [E5 10 50 52 01 00 01 03 00 0F 10 01 00 00 00 15]  LocoBuffer => LocoIO@52/1 Write SV3=0xf
# e5 10 50 51 00 04 00 04 00 00 01 00 12 00 00 00

    def whenBuildClicked(self,event) :
                                # Work out the packets which we will be sending
                                boardAdd1 = int(self.locoIOadd1.text)
                                boardAdd2 = int(self.locoIOadd2.text)
                                svNum = int(self.svNumber.text)
                                svHiVal = 0
                                svVal = 0
                                svReadWrite = 2
                                if (self.writeState.selected) :
                                                svVal = int(self.svValue.text)  & 127
                                                svHiVal = ( int(self.svValue.text) / 128 ) * 8
                                                svReadWrite = 1
                                packet = jmri.jmrix.loconet.LocoNetMessage(16)  # probably 16 ??
                                packet.setElement(0, 229)  # Op-Code to cause LocoIO Read/Write message
                                packet.setElement(1, 16)  # Low-bytes of address of sender ( = 1 )
                                packet.setElement(2, 80)  # High-bytes of sender ( = 80 )
                                packet.setElement(3, boardAdd1)  #  this should be a variable, depending which board....
                                packet.setElement(4, boardAdd2)   # again depends which board...
                                packet.setElement(5,svHiVal)  # related to hi bytes... becomes 8 if SV Val is > 127
                                packet.setElement(6,svReadWrite)  # Read a SV = 2, write a SV = 1
                                packet.setElement(7,svNum)  # SV number to read
                                packet.setElement(8,0)
                                packet.setElement(9,svVal)  # Value to write, or 0 if read
                                packet.setElement(10,16)  # appears to be a constant.
[... 131 lines cut ...]

                                                # end of stuff
                                return

# Get the LocoNetTrafficController.Instance going..
my_LnTrafContInstance = jmri.jmrix.loconet.LnTrafficController.instance()
my_LnTrafContInstance.addLocoNetListener(0xFF,MyLnListener())

# create one of these
a_ln_sv = LocoNetCV()

# set the name, as a example of configuring it
a_ln_sv.setName("LocoNet SV Tool")

# create a few variables
a_ln_sv.my_last_SVNum = 0
a_ln_sv.my_last_SVVal = 0
a_ln_sv.sv_array = {}

# and show the initial panel
a_ln_sv.setup()

--------------------end of jython script-----------------------

Nigel Cliffe :    nigel.cliffe@outlook.com<mailto:nigel.cliffe@outlook.com>  or  nigel@fleetwith.com<mailto:nigel@fleetwith.com>

silverailscolo, 2019-03-23:
Recent progress in the xml using compositeVal type:
[image]

silverailscolo, 2019-04-16:
The first 5 ports for LocoIO (versions 1.35 - 1.54), LocoBooster (v003-004) and LocoServo (v101-106) are now available in JMRI since PR #6768. Opcode1/2 and special options (port 1-4) were added in PR #6776.

closed 2019-04-16 by silverailscolo