JMRI/JMRI #5234: Oaktree not showing in sensors or turnouts

KenC57, 2018-04-13:
User reports it worked back in 3.8 but failing in 4.11.4. I tried with a dummy connection and it does complain about not finding all the nodes in the profile, which figures. But in the tables, the Oaktree system isn't offered, only Internal. Trying to load any panel file fails on any turnout or sensor entries.

The XML verify passes the files.
-ken c

KenC57, 2018-04-13:
Here is the profile and panel files from the user.
[profile485441011133136015.zip](https://github.com/JMRI/JMRI/files/1905741/profile485441011133136015.zip)

bobjacobsen, 2018-04-13:
If you can build a local copy of JMRI 3.8 (start with `git checkout v3.8`) and run it with those files, then `git bisect` will rapidly track down what change caused it to start failing.  See http://jmri.org/help/en/html/doc/Technical/GitFAQ.shtml and click on the 'git bisect' item.

silverailscolo, 2018-04-13:
Please try 4.11.5(ish). The OakTree systemconnectionmemo configureManagers() was updated to show up in the tables. Saw similar effect in the other systems migrated after we took out deprecated instance() in their managers.

KenC57, 2018-04-13:
I went to testing against the HEAD and things are much better.

Note 1: switching between a connection and simulator loses the node list. But I've seem this happen with other node based systems.
Note 2: on simulator, you get warnings about your nodes
WARN  - Timeout can't be handled due to missing node (index 1) [oaktree.SerialTrafficController Transmit thread]

But now the connection shows in the tables and the panel loads. I will advise the user that 4.11.5ish is working.
-ken c

KenC57, 2018-04-13:
However, some exceptions still happen with the simulator loaded.
2018-04-13 06:51:38,953 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [OakTree Simulator]
java.lang.NullPointerException
	at jmri.jmrix.oaktree.simulator.SimulatorAdapter.generateReply(SimulatorAdapter.java:263)
	at jmri.jmrix.oaktree.simulator.SimulatorAdapter.run(SimulatorAdapter.java:211)
	at java.lang.Thread.run(Unknown Source)
2018-04-13 06:51:50,006 jmrix.AbstractMRTrafficController     WARN  - sendMessage: Exception: In jmri.jmrix.oaktree.SerialTrafficController port warn:  [oaktree.SerialTrafficController Transmit thread]
java.io.IOException: Read end dead
	at java.io.PipedInputStream.checkStateForReceive(Unknown Source)
	at java.io.PipedInputStream.receive(Unknown Source)
	at java.io.PipedOutputStream.write(Unknown Source)
	at java.io.DataOutputStream.write(Unknown Source)
	at java.io.FilterOutputStream.write(Unknown Source)
	at jmri.jmrix.AbstractMRTrafficController.forwardToPort(AbstractMRTrafficController.java:630)
	at jmri.jmrix.AbstractMRTrafficController.transmitLoop(AbstractMRTrafficController.java:416)
	at jmri.jmrix.AbstractMRTrafficController$1.run(AbstractMRTrafficController.java:711)
	at java.lang.Thread.run(Unknown Source)

silverailscolo, 2018-04-13:
I noticed, and would like the help of people with OakTree hardware as to what messages are normally seen in such cases. The OakTree message console could provide a log.

KenC57, 2018-04-14:
The user also asks if it might be possible to add a signal head/mast specific for the Oaktree boards. He notices that the comm channel gets stuffed with only a couple of flashing heads. The boards support some other forms. Maybe somebody familiar with this can review the documents.
[IO48USER.pdf](https://github.com/JMRI/JMRI/files/1909900/IO48USER.pdf)
Additional comments from user:
If you make signal 1 flash that is on card 3 then the inputs for 
that card no longer function. This is the same for any card that 
is flashing a signal.

It looks to me like JMRI is making the output flash by
by sending the on then off output command over and over.
This could be jamming the card with so much data that it can not
get the command for the input data to the computer????

The cards can handle the flashing on there own with a different 
command for making an output Blink and then another to turn it off.

Command 23 -- Blink
Function – This command causes the alternate high/low output to occur on a single point. The duration value is
expressed in increments of 0.01 seconds. Each state of the output is for an equal length of time, and can range from
0.01 seconds to 2.55 seconds. This command is primarily intended for slow flashing (e.g. warning flasher).
Message Format:
Byte 1 Byte 2 Byte 3 Byte 4 Byte 5
Card Address       Command                    Point Number                Duration Check                    Byte
AAAAAAAA          23                                XXXPPPPP                   DDDDDDDD                         CCCCCCCC

PPPPP - Point Number
DDDDDDDD - Duration Count
X - Ignored
Action Taken: The output point will cycle on and off at the rate specified in the duration byte.
Response - normal 1 byte ACK
See page 19 of attached .pdf
Is there a way to add this to JMRI or something I can edit to make it do this?
The more cards in a system that are flashing an output also slow the program down quite a bit.
I will send another Debug while I am doing the above to signal 1
Thanks again for your help

Robbie Diehl

closed 2021-04-14 by stale[bot]