JMRI/JMRI #5235: Positional Label Background Colors

dsand47, 2018-04-13:
If a positional label has been assigned a font, the background colors are not retained on the second save.

The first save after creating the label with a font and background includes the background information in the xml file.  A subsequent load is successful.  The next save does **_not_** include the background information.

This affects PE, CPE and LE.

dsand47, 2018-04-14:
When a positionable label has a font assigned, the **PositionablePopupUtil** instance is cloned partway through the XML loading process for the label.  The remaining data is applied to the old instance and is not available the next time the panel xml file is saved.

bobjacobsen, 2018-04-24:
Fixed by #5236

closed 2018-04-24 by bobjacobsen