JMRI/JMRI #5243: 4.11.5 roster/decoder issues

KenC57, 2018-04-17:
Initial mention was from a user saying opening a roster entry in edit mode did nothing. I found that it does open the roster entry, but fails to display the pane. It also has a number of warnings or errors in the console.

In the case of Bachmann E-Z 2 function decoder, it listed a number of variables as warnings:

2018-04-17 10:36:07,525 roster.LocoFile                       WARN  - Did not find locofile variable "Version ID" in decoder definition, not loading [AWT-EventQueue-0]
2018-04-17 10:36:07,531 roster.LocoFile                       WARN  - Did not find locofile variable "Manufacturer ID" in decoder definition, not loading [AWT-EventQueue-0]
2018-04-17 10:36:07,534 roster.LocoFile                       WARN  - Did not find locofile variable "Normal direction of motion" in decoder definition, not loading [AWT-EventQueue-0]
2018-04-17 10:36:07,535 roster.LocoFile                       WARN  - Did not find locofile variable "Analog conversion mode" in decoder definition, not loading [AWT-EventQueue-0]
2018-04-17 10:36:07,535 roster.LocoFile                       WARN  - Did not find locofile variable "Addressing Mode" in decoder definition, not loading [AWT-EventQueue-0]
2018-04-17 10:36:07,537 roster.LocoFile                       WARN  - Did not find locofile variable "User Private ID #1" in decoder definition, not loading [AWT-EventQueue-0]
2018-04-17 10:36:07,539 roster.LocoFile                       WARN  - Did not find locofile variable "User Private ID #2" in decoder definition, not loading [AWT-EventQueue-0]

If I open the same decoder in 'Edit' mode, the pane does not show up (but is in the Windows list), the same warnings happen, but also an error happens:

2018-04-17 10:41:16,144 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.NullPointerException
	at jmri.jmrit.symbolicprog.VariableTableModel.setButtonModeFromProgrammer(VariableTableModel.java:628)
	at jmri.jmrit.symbolicprog.tabbedframe.PaneProgPane.setCvListFromTable(PaneProgPane.java:2191)
	at jmri.jmrit.symbolicprog.tabbedframe.PaneProgPane.makeCvTable(PaneProgPane.java:2180)
	at jmri.jmrit.symbolicprog.tabbedframe.PaneProgPane.newColumn(PaneProgPane.java:1533)
	at jmri.jmrit.symbolicprog.tabbedframe.PaneProgPane.<init>(PaneProgPane.java:268)
	at jmri.jmrit.symbolicprog.tabbedframe.PaneProgFrame.newPane(PaneProgFrame.java:1215)
	at jmri.jmrit.symbolicprog.tabbedframe.PaneProgFrame.readConfig(PaneProgFrame.java:959)
	at jmri.jmrit.symbolicprog.tabbedframe.PaneProgFrame.loadProgrammerFile(PaneProgFrame.java:789)
	at jmri.jmrit.symbolicprog.tabbedframe.PaneProgFrame.<init>(PaneProgFrame.java:465)
	at jmri.jmrit.roster.swing.RosterFrame$4.<init>(RosterFrame.java:1403)
	at jmri.jmrit.roster.swing.RosterFrame.startProgrammer(RosterFrame.java:1403)
	at jmri.jmrit.roster.swing.RosterFrame.lambda$1(RosterFrame.java:296)
	at javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
	at javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
	at javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
	at javax.swing.DefaultButtonModel.setPressed(Unknown Source)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(Unknown Source)
	at java.awt.Component.processMouseEvent(Unknown Source)
	at javax.swing.JComponent.processMouseEvent(Unknown Source)
	at java.awt.Component.processEvent(Unknown Source)
	at java.awt.Container.processEvent(Unknown Source)
	at java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$500(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)

bobjacobsen, 2018-04-17:
> 2018-04-17 10:36:07,525 roster.LocoFile WARN - Did not find locofile variable "Version ID" in decoder definition, not loading [AWT-EventQueue-0]

That message means that the previously-stored locomotive file (the roster entry file) had a “Version ID” variable stored in it, but the current decoder definition doesn’t.  That sounds like something bad happened to the decoder definition, as that variable really should be there.  Or perhaps DecoderPro is somehow confused by the model information in the locofile, and is looking at the wrong decoder definition.  

Could you please post the roster entry file for that loco?

> 2018-04-17 10:41:16,144 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0] [...]

That’s a weird place for an NPE. The only way I can see it happen is if the decoder definition wasn’t completely read in and turned into objects.  But I don’t see how that can happen.

Was there anything else in the log at all?

KenC57, 2018-04-17:
The warning messages and the NPE were all that went to the log.
[Ken_C_NYC_1957.zip](https://github.com/JMRI/JMRI/files/1920457/Ken_C_NYC_1957.zip)
 Yes this is an old decoder, but the roster entry file date is 2011.

KenC57, 2018-04-17:
I would add that similar messages happen if I try other roster entries. The warnings differ, but the NPE seems consistent with edit mode. The warnings happen in ops or service modes too, but no NPE.

dheap, 2018-04-17:
As @bobjacobsen says, the `roster.LocoFile WARN - Did not find locofile variable "xxx" in decoder definition, not loading` is unrelated to this issue. It does occur opening existing entries after variables are renamed/removed.

However I can confirm that the Edit mode is apparently completely broken with V4.11.5 and I got the NPE with any roster entry, using various Simulator and real connections.

PR #5333 (2018-05-13): Protect against DecoderPro NPE in edit mode
Should resolve #5243 and #5330
closed 2018-05-13 by bobjacobsen