JMRI/JMRI #5244: Font identification on certain machines delays startup

bobjacobsen, 2018-04-17:
On some machines, startup can be delayed by 10's of seconds or even minutes while fonts are scanned.

There have been several jmri-developers discussions about this.  Basically, the scan is desired to know whether each font is mono- or proportional-spaced; on some (Windows) machines this requires loading the font from disk, which takes time; on machines with lots of such fonts it takes a _long_ time; although that operation is deferred to another thread, JMRI startup ends up waiting for it.

A thread trace is attached. In it,

- The “PrepareFontListsThread” thread is running. Meanwhile, 147 “FontComboUtil Prepare” threads are blocked. 

 - The “init prefs” thread is blocked in jmri.util.swing.FontComboUtil.prepareFontLists(..) 

 - That’s all tangled up in this wait:
```
"TabbedPreferencesAction actionPerformed" #74 prio=6 os_prio=0
tid=0x0000000022ea2800 nid=0xe4 waiting for monitor entry
[0x000000002817e000]
 java.lang.Thread.State: BLOCKED (on object monitor)
      at apps.gui3.TabbedPreferences.init(TabbedPreferences.java:138)
      - waiting to lock <0x0000000082c3e928> (a
apps.gui3.TabbedPreferences)
      at
apps.gui3.TabbedPreferencesAction.lambda$0(TabbedPreferencesAction.java:87)
      at
apps.gui3.TabbedPreferencesAction$$Lambda$133/386599598.run(Unknown Source)
      at java.lang.Thread.run(Unknown Source)
```

[ThreadDump.txt](https://github.com/JMRI/JMRI/files/1920407/ThreadDump.txt)

bobjacobsen, 2018-04-17:
Part of the problem is due to how  FontComboUtil synchronizes requests, which can generate extra threads. There's a (possible) fix for that in the "[font-synchronization](https://github.com/bobjacobsen/JMRI/tree/font-synchronization)" branch in my bobjacobsen/JMRI repository. It contains one commit: e4d008e75464ae280432a48c5461ec7e3f7f7d78  

I have no way to check whether that helps the problem or not, so won't be creating a PR from it until somebody who can recreate the issue tests it.

KenC57, 2018-04-18:
It seems to avoid the spawn of many threads and the long wait on a slow machine. So a good improvement. I could open tables and preferences without odd delays. No big list of threads in the monitor. So I'd say it passes.

bobjacobsen, 2018-04-18:
PR #5250 (merged) reduces the impact of the font reads on normal operation, at least until something tries to reference the font list.  But it doesn't solve the underlying problem, which is the time needed to do the mono- vs proportional-spacing check when the fonts are not already loaded.

bobjacobsen, 2018-04-25:
From Dick Bronson on jmri-developers:

> OK, running JMRI.4.11.6ish+jenkins+20180419T2149Z+Raca91d7.tgz
> 
> It still hogs resources like they were free for 5-6 minutes, (see attached) but it no longer blocks the opening of the Preferences window. It looks cleaner on the thread monitor as well running as a single thread.
> 
> Thanks for re-arranging whatever you did. You mentioned that the delay might be from disk accesses, but that doesn't seem to be evident. I.e. my disk activity is not high based on the front panel LED.

[image]

bobjacobsen, 2018-04-25:
Re above: 50% might be one thread spin-locking on a two-core system.  It's not the Swing thread, as things are working.  So it's probably some part of the initialization thread(s).  A thread dump should show it if it's that consistent.

mattharris, 2018-04-25:
I _really_ should get back to looking into this (especially as the original instigator of this chunk of code).

As regard the original intent, a form of caching mechanism should greatly assist this - just need to give it some thought.

PR #8380 (2020-04-19): fix: Always use system monospace font in console
This should prevent initializing the FontComboUtil from impacting most users, since the remaining uses of FontComboUtil are related to printing, and I doubt someone has an automatic action to print from JMRI on startup.

Closes #8362; the difference between this PR and #8362 is that this PR allows other code that uses the FontComboUtil to get a list of fonts on the system to continue to do so while #8362 did not.
Fixes #5244 by not cataloging font families on startup.
closed 2020-04-19 by bobjacobsen