JMRI/JMRI #5245: /dist path in JMRI Web Server no longer working

mstevetodd, 2018-04-18:
`/dist` now returns a 404 Not Found. I'm not sure if this is intentional, but the initial home page still lists it as an option.

mstevetodd, 2018-04-18:
Further testing shows that only the `/prefs` path can be accessed directly, `/web`, `/images`, etc. also return the 404. However, you can specify individual files in those folders, such as `/dist/jmri.conf`

bobjacobsen, 2018-05-18:
Sees to have also been true, at least for /dist, in JMRI 4.10.  

Log shows:
```
     [java] 2018-05-18 10:54:46,130 directory.DirectoryService            ERROR - Sending /dist/ for /Users/jake/Documents/Trains/JMRI/projects/JMRI in context / [WebServer-64]
```
But /prefs seems not to work for me (contra above); either /web nor /images.  /dist/jmri.conf does display OK.

bobjacobsen, 2018-06-10:
git bisect found:

```
fa2b80d9ffacfd497e89d5a23c03ea8d6e05c625 is the first bad commit
commit fa2b80d9ffacfd497e89d5a23c03ea8d6e05c625
Author: Randall Wood <randall.h.wood@alexandriasoftware.com>
Date:   Fri Oct 6 05:41:14 2017 -0400

    Update Jetty to current version.

:100644 100644 524bdf2ba116a80bffa2f53a92d1efcd889d601a 67e195ea35fd7df5b6d7898ed9651918872c0229 M	.classpath
:100644 100644 90585f0a92a5dc44efaefcffd6683d117474b4cb ab10ccb0a6d3d5de5ca6746af7da7beaedf43487 M	build.xml
:040000 040000 7744281b375137cb703f4f10b4b85a008a0010d8 23c8432cb2d4e00acca617d5f60cd347186e2af8 M	java
:040000 040000 045d50d2afc097fd3f587365a5c5226b60854819 5a70a5e50c41ec80ca57ef34a34de41d9c596c78 M	lib
:040000 040000 62f8fba0cb526a98bde834ced42fb9c6195e4b5b 3c0740f7e84891b64db45b2fa581942230bb0a5f M	nbproject
:100644 100644 77a3810c5931637574eedec6410d034295ce487f 1d3ad08ed13f86c3a8cdcd0f721092c2912b3a14 M	pom.xml
```

bobjacobsen, 2018-06-10:
In commit fa2b80d9ffacfd497e89d5a23c03ea8d6e05c625 Randall switched around the directory handling a lot, and apparently broke it. 

The 9-line jmri.web.servlet.directory.DirectoryHandler#doDirectory method became the jmri.web.servlet.directory.DirectoryResource.java class. It's not clear how (or if) that's being loaded.

bobjacobsen, 2018-06-10:
#5464 should fix this.

This whole area of the code needs some functional tests.  @pabender, is that something that Cucumber can do?

pabender, 2018-06-11:
@bobjacobsen yes, we certainly can do that. Cucumber is used for behavior driven testing.

The existing cucumber tests are pretty simplistic.  The selenium web driver is actually capable of doing a lot of the UI tests that we do in the java code, which is part of the reason 
I wanted to get that working, I just haven’t had time to go add additional feature tests.

PR #5464 (2018-06-11): Web server providing directory listings again
Restores the ability to provide directory listings.  Resolves #5245. Note that this code has only ctor tests,  so the coverage numbers are quite misleading.
closed 2018-06-11 by bobjacobsen