JMRI/JMRI #5264: Turnout Table Issues

radioflyer72, 2018-04-24:
Hi. I am having an issue with the turnout tables not storing the user names of the turnouts. I have tried multiple times. I save the configuration to the file, and to the panel files, but each time that i close Panel Pro and restart, the user names of the turnouts are missing.  I have tried multiple times with the same result. Any Ideas?  Also the turnout directional thrown/close graphics in the turnout tables are backwards. Currently running Panel Pro Version 4.10, Java 1.8.0

dsand47, 2018-04-24:
Are you loading the panel xml file after re-starting PanelPro?

silverailscolo, 2018-04-24:
I reversed the graphics vs states somehere between 4.11.4 and 4.11.5 as a user reported they were inversed on WebServer panel displays. @radioflyer72 Did you perhaps check the "Reversed" box to quickle solve that back then? If so, what happens if you now uncheck it?
When you change your prefs to text state display, do they match your layout turnouts? (Thrown - Diverted)

dadspc, 2018-08-08:
i am using Panelpro 4.12 and am suffering from the same issue where user names are not restored when Panelpro restarts. I have saved the turnout configuration to the file called panel.xml. Using notepad i can see the user names in this file but they do not appear in the turnout table. Does Panelpro automatically load the configuration file panel.xml when it starts?

dadspc, 2018-08-08:
Problem solved...I am new to JMRI and GitHub and threfore learning rapidly. I have discovered that you must explicitly load the configuration file (panel.xml in my case) by setting it up in the startup preferences.

closed 2021-03-13 by icklesteve