JMRI/JMRI #5265: Yahoo thread: Sensors do not save after editing in 4.11.5

dsand47, 2018-04-24:
In **jmri.managers.configurexml.AbstractSensorManagerConfigXML**, the following line:

        java.util.Iterator<String> iter = tm.getSystemNameAddedOrderList().iterator();

Includes sensor names for deleted sensors.

bobjacobsen, 2018-04-24:
Wow, that's a subtle bug.  Working on it.

bobjacobsen, 2018-04-24:
Thanks for catching this. #5266 is the fix.  I'll add a warning to the 4.11.5 release note.  

I don't have a handle on how significant this is (how often people delete things). Is this significant enough to push out a quick 4.11.6 from HEAD?

dsand47, 2018-04-24:
There has only been the one instance of the problem reported.  The OP is going to manually delete at the XML level for now.  We should be able to recommend the Jenkins release if necessary.

dsand47, 2018-04-24:
This problem also affects Turnouts, Lights, Reporters, Memories and Audio.

PR #5266 (2018-04-25): Deleted NamedBeans (e.g. Sensors) really should go away
Fixes #5265
closed 2018-04-25 by bobjacobsen