JMRI/JMRI #5276: Add New Profile Error in JMRI 4.11.5

CCatania01, 2018-04-25:
When attempting to add a new profile to the preferred Preferences Path I get the following error dialog:
[image]

Using Browse and going up one level, the new profile is created in that selected directory.
[image]

Here are the Preferences Config Profiles screen shots.  
[image]
[image]

I recently added a DecoderPro profile which was not in place earlier.

bobjacobsen, 2018-04-26:
It looks like your "DecoderPro 3" profile was somehow created using /Users/Development/Library/Preferences/JMRI as the profile directory.  Normally /Users/Development/Library/Preferences/JMRI  is the preferences directory that _contains_ one or more profile directories.  

Before you do anything to try to fix this, I _strongly_ recommend you make multiple backups of the complete contents of /Users/Development/Library/Preferences/JMRI 

It might work to delete the "DecoderPro 3" profile.  But that might also delete the entire "DecoderPro 3" profile directory, which is the preferences directory, which would delete all your other profiles.

The other choice is manually removing the parts that make up the "DecoderPro 3" profile,leaving everything else in place.  That will be tricky...

CCatania01, 2018-04-26:
Bob,
Thank you for the direction.

I backed up the full preferences and did a delete from PP. That deleted everything and created a clean, new environment.  I created a new test profile without error.
I found the best way to recreate the profiles was to launch PP, do a new profile with the existing profile name, quit PP, and move the backup profile contents into the new profile.

The environment is back to normal.
Chuck Catania

closed 2018-04-26 by CCatania01