JMRI/JMRI #5294: Test failure in EditCircuitFrameTest

bobjacobsen, 2018-05-03:
`jmri.jmrit.display.controlPanelEditor.EditCircuitFrameTest` is showing occasional errors within the Swing/AWT code. A typical traceback"

```
1) testCTor(jmri.jmrit.display.controlPanelEditor.EditCircuitFrameTest)
java.lang.NullPointerException
	at java.awt.GridBagLayout.GetLayoutInfo(GridBagLayout.java:1118)
	at java.awt.GridBagLayout.getLayoutInfo(GridBagLayout.java:916)
	at java.awt.GridBagLayout.preferredLayoutSize(GridBagLayout.java:736)
	at java.awt.Container.preferredSize(Container.java:1796)
	at java.awt.Container.getPreferredSize(Container.java:1780)
	at javax.swing.JComponent.getPreferredSize(JComponent.java:1664)
	at java.awt.FlowLayout.preferredLayoutSize(FlowLayout.java:414)
	at java.awt.Container.preferredSize(Container.java:1796)
	at java.awt.Container.getPreferredSize(Container.java:1780)
	at javax.swing.JComponent.getPreferredSize(JComponent.java:1664)
	at javax.swing.BoxLayout.checkRequests(BoxLayout.java:484)
	at javax.swing.BoxLayout.preferredLayoutSize(BoxLayout.java:301)
	at java.awt.Container.preferredSize(Container.java:1796)
	at java.awt.Container.getPreferredSize(Container.java:1780)
	at javax.swing.JComponent.getPreferredSize(JComponent.java:1664)
	at java.awt.BorderLayout.preferredLayoutSize(BorderLayout.java:719)
	at java.awt.Container.preferredSize(Container.java:1796)
	at java.awt.Container.getPreferredSize(Container.java:1780)
	at javax.swing.JComponent.getPreferredSize(JComponent.java:1664)
	at javax.swing.JRootPane$RootLayout.preferredLayoutSize(JRootPane.java:920)
	at java.awt.Container.preferredSize(Container.java:1796)
	at java.awt.Container.getPreferredSize(Container.java:1780)
	at javax.swing.JComponent.getPreferredSize(JComponent.java:1664)
	at java.awt.BorderLayout.preferredLayoutSize(BorderLayout.java:719)
	at java.awt.Container.preferredSize(Container.java:1796)
	at java.awt.Container.getPreferredSize(Container.java:1780)
	at jmri.util.JmriJFrame.getPreferredSize(JmriJFrame.java:578)
	at jmri.util.JmriJFrame.pack(JmriJFrame.java:239)
	at jmri.jmrit.display.controlPanelEditor.EditCircuitFrame.makeContentPanel(EditCircuitFrame.java:155)
	at jmri.jmrit.display.controlPanelEditor.EditCircuitFrame.<init>(EditCircuitFrame.java:61)
	at jmri.jmrit.display.controlPanelEditor.EditCircuitFrameTest.testCTor(EditCircuitFrameTest.java:24)
...
	at org.junit.runners.model.FrameworkMethod$1.runReflectiveCall(FrameworkMethod.java:50)
...
	at org.junit.runners.ParentRunner.runLeaf(ParentRunner.java:325)
```

This may be a threading issue in how the test is invoking methods.

bobjacobsen, 2018-05-03:
PR #5295 addresses this, but I'd like to leave this issue open for a bit to see if it recurs.

pabender, 2018-05-03:
@bobjacobsen 

I've got an alternate, probably complementary, proposed solution in #5296 

It looks like about 5 months ago, many of the explicit panel layout settings were commented out.  I don't know how long this test has been periodically failing, but if it started after that, that might be the issue.

closed 2018-05-12 by bobjacobsen