JMRI/JMRI #5300: 4.11.6 Loconet Stationary Decoder Read Sheet fails

n3ix, 2018-05-04:
In 4.11.5 the new LOCONETOPSBOARD setting was working and Read Sheet worked fine with Loconet Stationary decoders.  In 4.11.6 the Read Sheet fails with a "Programmer bus (302)" error.  The loconet log appears to be what is expected with a Loconet Stationary Decoder read:

0:32:04.560: [EF 0E 7C 2F 00 55 79 00 00 01 00 7F 7F 60]  Byte Read on Main Track (Ops Mode): Decoder address 11001: CV2.
10:32:04.560: [B4 6F 00 24]  LONG_ACK: The Slot Write command was rejected.
10:32:04.560: [E7 0E 7C 2F 00 55 79 00 02 01 47 7F 7F 2D]  Programming Response: Byte Read on Main Track (Ops Mode) Was Successful: Decoder address 11001: CV2 value 199 (0xC7, 11000111b).

Test setup is DCS240 + Windows 10.  I also tried setting the Transponding connection option to On and got the same result.

n3ix, 2018-05-12:
New info - the problem is connected to the Track Power State!  I set JMRI to Open Loconet Monitor as the first startup action and saw that JMRI was continually repeating the Loconet Interrogate messages even though the Loconet Stationary Decoders were responding properly.  Looking closer I saw that the command station was sending a LONG ACK right after each interrogate request.  Aha, the DCS240 sends a LONG ACK failure message anytime you send a Switch Request when track power is off.  

Given this info, I retested Ops Mode Reads on the Loconet Stationary decoders and it works fine when track power is ON!  For example,

0:37:35.027: [EF 0E 7C 2F 00 55 79 00 00 01 00 7F 7F 60]  Byte Read on Main Track (Ops Mode): Decoder address 11001: CV2.
10:37:35.027: [B4 6F 40 64]  LONG_ACK: The Slot Write command was accepted blind (no response will be sent).
10:37:35.042: [E7 0E 7C 2F 00 55 79 00 02 01 47 7F 7F 2D]  Programming Response: Byte Read on Main Track (Ops Mode) Was Successful: Decoder address 11001: CV2 value 199 (0xC7, 11000111b).
1

Looking at the sequence shown my first comment above, a different LONG_ACK was sent by the DCS 240 when track power was off.  I should have spotted that.  So it looks like the actual underlying issue is needing some special handling of the DCS 240 LONG_ACK response variation when track power is when sending Switch Requests and Ops Mode Read Requests.  Hope that helps.

devel-bobm, 2018-05-16:
@n3ix, the "track power on" issue is really two different issues, because JMRI handles the LocoNet messaging for the two different cases you mention in two different ways, in two different sections of JMRI code.  And "track power on" may affect other things, such as "OPC_IMM_PACKET" messages, which are handled in at least one other part of JMRI code.

As such, changes to (at least) two different parts of JMRI code would be required to deal with the "track power on" issue.

n3ix, 2018-05-16:
@devel-bobm Hi.  I ran into this with a DCS240 and it has been reported on a Zephyr.  I wonder if all the newer generation Digitrax command stations respond differently to Ops Mode Read and Switch Request messages when track power is off?  

Just tried a Signal Mast, which I believe uses OPC_IMM_PACKET.  With Track Power ON and Track Power OFF the DCS240 sent exactly the same LONG_ACK:

Track Power ON
13:20:20.328: [ED 0B 7F 32 01 25 61 02 00 00 13]  Send packet immediate: 3 bytes, repeat count 2(50)
	DHI=0x01, IM1=0x25, IM2=0x61, IM3=0x02, IM4=0x00, IM5=0x00
	packet: A5 61 02 .
13:20:20.328: [B4 6D 7F 59]  LONG_ACK: the Send IMM Packet command was accepted.

Track Power OFF
13:20:42.594: [ED 0B 7F 32 01 25 61 02 00 00 13]  Send packet immediate: 3 bytes, repeat count 2(50)
	DHI=0x01, IM1=0x25, IM2=0x61, IM3=0x02, IM4=0x00, IM5=0x00
	packet: A5 61 02 .
13:20:42.594: [B4 6D 7F 59]  LONG_ACK: the Send IMM Packet command was accepted

There may be other cases to test, but so far I have seen the issue with Ops Mode Read on Loconet Stationary Decoders and with Loconet Stationary Interrogate during JMRI startup if the Turnout Handling is not set to "One Only".

devel-bobm, 2018-05-16:
@n3ix,

>  I ran into this with a DCS240 and it has been reported on a Zephyr. I wonder if all the newer generation Digitrax command stations respond differently to Ops Mode Read and Switch Request messages when track power is off?

That seems likely to me.  I would -guess- that Digitrax' engineers have decided that it does not make sense to send turnout control commands when track power is off, and that it is -necessary- to respond with a `LONG_ACK(Error)` message when one is seen under that condition.  Similarly, I guess that they have decided that it does not make sense to send "ops-mode programming" operations when track power is off so send `LONG_ACK(Error)`.  And that they decided to send `LONG_ACK(Error)` in the case of an `OPC_IMM_PACKET` message when track power is off.

As you have shown, this is an inappropriate implementation for at least some types of hardware, including some of their own and quite a bit of 3rd-party hardware...

So what is JMRI to do?  For turnouts, nothing much - there is too much history with this particular boundary-condition.  For `OPC_IMM_PACKET`, the jury is out to lunch - I haven't heard any discussion of this case.  I personally am NOT a fan of the "remember anything that fails and retry, in order" club.

For ops-mode programming, there are some obvious options:
- Add another connection-specific option which can make JMRI ignore ignore `LONG_ACK(Error)` replies to Ops-mode programming requests.  Issue with this: If track power is off and the device is "on the main" then programming operations will fail silently - not a good end-user experience, when it happens.
- Add something to the decoder definition XML which states that the decoder does not require track power; JMRI would (somehow) ignore `LONG_ACK(Error)` messages when programming a decoder indicates appropriately.
- Add an option to the symbolic programmer window that can disable the way it pays attention to the `LONG_ACK(Error)`.  Enable that option only for Digitrax systems or other systems which can have a similar circumstance, and only enable when "track-power-off" can block a programming operation.
- Add some sort of text notification that says "hey bozo, turn on the track power!".
- Leave things as-is, with the not-quite-silent failures and "poor user experience".

Other options may exist - I am not coming up with anything at the moment.

n3ix, 2018-05-16:
@devel-bobm  Well I have a different take on several points.

Digitrax did not decide to send LONG_ACK(Error) in case of an OPC_IMM_PACKET when track power is OFF based on my testing.  As I showed in the Loconet log entries above, the DCS240 sends _exactly_ the same LONG_ACK response to OPC_IMM_PACKET whatever the state of track power.

Believe that the decoder definition already has a field that indicates it supports Loconet Ops Mode Read: LOCONETOPSBOARD.  When this is present it seems like the Ops Mode Programmer should treat either:
[B4 6F 40 64] LONG_ACK: The Slot Write command was accepted blind (no response will be sent)
or
[B4 6F 00 24] LONG_ACK: The Slot Write command was rejected.
as a valid command station response and then accept and process the response from the stationary decoder response, i.e.
[E7 0E 7C 2F 00 55 79 00 02 01 47 7F 7F 2D] Programming Response: Byte Read on Main Track (Ops Mode) Was Successful: Decoder address 11001: CV2 value 199 (0xC7, 11000111b).

It is perfectly valid to program a Loconet Stationary Decoder with turn track power either OFF or ON.  I disagree strongly with the idea that the user did something wrong by attempting this with track power OFF.  

JMRI issues a Loconet Stationary Decoder interrogate sequence at startup.  Since this sequence is aimed exclusively at Loconet devices, I think that the 8 Switch Requests that make up this sequence should _never_ be retried regardless of the state of the Turnout Handling setting.  When you start JMRI with track power OFF, if you have lots of stationary decoders and you don't have Turnout Handling set to "One Only" the amount of Loconet traffic generated can be staggering.  I would guess that it is extremely common to have track power OFF when JMRI is started.  There certinaly is no "bozo" factor involved.

devel-bobm, 2018-05-17:
@n3ix, 

Continuing to tilt at windmills:  What assumption did Digitrax make with respect to Ops-mode programming?  It seems like they've assumed that the device being programmed must be "on the track" in order to receive the programming command.  

If that is the case, then any device which does -not- need to be "on the track" to be successfully programmed using Ops-mode programming violates that assumption.

The assumption that the "LocoNet Stationary Decoder interrogate" sequence is aimed exclusively at LocoNet devices is entirely without merit.  This command was put in-place to support the DS54, which ONLY receives control messages from the DCC track signal, not LocoNet.  The DS54 sees the "interrogate" switch control messages on the DCC track signal and responds with messages on LocoNet.  It would be better to call the sequence a "stationary decoder interrogate", or "LocoNet request for stationary decoder interrogation".  

Note that the LocoNet Personal Edition 1.0 specification describes a "Stationary Interrogate Command" as a particular DCC track packet, without any specific discussion of how to trigger it.  The description never even mentions LocoNet.  If one reverse-engineers the spec'd DCC track packet, and compares to a NMRA stationary decoder control packet, one can infer that the "interrogate" it is simply a "special case" of the standard stationary decoder control packet.  From this one can infer that a LocoNet `<OPC_SW_REQ>` message with appropriate parameters will cause the same "interrogate" DCC track packet.  And it does.  And the DS54 responds appropriately to it.  Thus what JMRI is doing is "triggering" the "Stationary Interrogate Command"s by issuing "standard" LocoNet messages.

Again, I will state for the record - I am NOT a fan of the current way JMRI floods LocoNet when a turnout command is rejected by the command station when track power is off.  But this is a historical JMRI implementation.  

I have no interest in "fixing" this JMRI "feature".  It seems that only three or four people (including you and I) think it is broken.  I don't use a command station for anything turnout- or sensor-related, so this JMRI behavior doesn't bother me in the least.  The fact that this JMRI startup-time interrogation occasionally causes havoc for other layout owners annoys me.  But it leaves me with nothing more than an "academic" interest in solving the problem.  An "acceptable" solution, to my way of thinking, requires some non-trivial re-engineering of the "turnout retry" mechanism, and far more time than I am willing to give at this time.  With such a simple set of workarounds (turn the #&%#*^ track power on, set an OpSw in the DCS210/DCS240 (and possibly DCS51)), I just cannot see spending time on developing a better solution!

A JMRI "Bushby Bit" solution might be a long-term, partial solution to the "turnout retry" problem.

n3ix, 2018-05-17:
@devel-bobm Hey I get that there is only so much time, and everyone gets to decide when and where to spend time on JMRI.  No problem if this one doesn't rise to a high enough level to get your interest.

For me this was all triggered by a real world issue - the inability to do an Ops Mode Read for over a week on my test bench.  Since I had been running Jenkins builds I chalked it up to the code being broken when in fact the issue was Track Power was OFF.  Others have reported a problem that sounds exactly like this so I'm not alone.  And I have to support many QuadLN product users and so have good reason to be concerned as the number of new generation Digitrax command stations in the field grows.

What some manufacturer had in mind when they designed something might be of academic interest but from a practical standpoint how their products actually interact in the JMRI environment is of primary concern to me.   Certainly would not want JMRI to get less user friendly due to a change in command station behavior if it can be easily avoided.  When it comes to an Ops Mode Read of a Loconet stationary decoder in JMRI, two things are expected: a specific command station response and then the actual Read response from the decoder.   Currently JMRI rejects a valid Read response from the decoder if the command station does not supply the specific response first.  I do not see why that is necessary and do not see the value in decoding the command station response?  Why couldn't the only issue be whether the decoder responds properly?

(Way back when JMRI required the command station response part of the Ops Mode Read even for a standalone Loconet .  As a result I had to put a configuration setting in the stationary decoder years ago to optionally provide that command station response as well as the Read response.  I believe RR-Cirkits did the same thing.  Perhaps that is not longer true and JMRI now doesn't care about the command station response when Standalone Loconet is specified?  If so maybe the approach could be "for Ops Mode Reads, accept but do not require a command station response of any kind" to cover all Loconet configurations?

As far as the Stationary Interrogate is concerned, I personally have no DCC-connected stationary decoders so I run with Turnout Handling set to "One Only" and find that resolves any issue with Track Power being OFF when JMRI starts.  So I thought perhaps some simple solution might be available, for example have JMRI start with the Turnout Handling setting at "One Only" until the Interrogate Sequence is complete and then switch it to the User-supplied value.

bobjacobsen, 2018-05-21:
I'm having a bit of trouble following all this, sorry.

Is the right thing to do to add a check that track power is on when doing an OpsMode Read, and warning people if it's not?  That's not so hard. Or is something else required?

Should that power check be specific to LocoNet connections, or to specific command station selections, or something else?

n3ix, 2018-05-21:
@bobjacobsen  Hi, sorry for the confusing journey above.  Let me restate the problem, give a couple examples and then offer thoughts on solutions.

Observed behavior: When doing Ops Mode Reads for Loconet stationary decoders such as the Tower Controller or QuadLN, JMRI sometimes discards perfectly valid responses from the decoders.

Loconet messages for Ops Mode Read:  The basic sequence consists of 3 messages - one from JMRI, one from the command station, and finally one from the stationary decoder:
(sent by JMRI): [EF 0E 7C 2F 00 55 79 00 00 01 00 7F 7F 60] Byte Read on Main Track (Ops Mode): Decoder address 11001: CV2.
(sent by DCSxxx) [B4 6F 40 64] LONG_ACK: The Slot Write command was accepted blind (no response will be sent).
(sent by decoder) [E7 0E 7C 2F 00 55 79 00 02 01 47 7F 7F 2D] Programming Response: Byte Read on Main Track (Ops Mode) Was Successful: Decoder address 11001: CV2 value 199 (0xC7, 11000111b).

The sequence above is seen with a DCS100 regardless of whether track power is on or off, and JMRI treats the read as successful.  The sequence above is also seen with a DCS240 when track power is ON.  However when track power is OFF, the DCS240 sends a slightly different LONG_ACK:

[EF 0E 7C 2F 00 55 79 00 00 01 00 7F 7F 60] Byte Read on Main Track (Ops Mode): Decoder address 11001: CV2.
[B4 6F 00 24] LONG_ACK: The Slot Write command was rejected.
[E7 0E 7C 2F 00 55 79 00 02 01 47 7F 7F 2D] Programming Response: Byte Read on Main Track (Ops Mode) Was Successful: Decoder address 11001: CV2 value 199 (0xC7, 11000111b).

Currently JMRI fails the Ops Mode Read in this case based on the command station response, even though the actual read response from the decoder is identical to that in the first case.

Discussion:  JMRI knows that it is attempting to read from a Loconet device, thanks to the new LOCONETOPSBOARD parameter.  The particular type of LONG_ACK sent by the command station (or even the potential total absence of a LONG_ACK in the case of a Standalone Loconet with no command station) appears to be irrelevant?   If the Loconet device responds correctly, as it does it both cases above, my thought is that JMRI should accept the response and the Read should succeed in both cases.

n3ix, 2021-03-30:
This is still a valid issue.  The sequence of messages for an Ops Mode Read for instance is: Byte Read on Main Track; LONG_ACK of some form; Programming Response.  With newer Digitrax command stations, when track power is on the form of the LONG_ACK is "[B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted".  When track power is off the form of the LONG_ACK is "[B4 6F 00 24]  LONG_ACK: The Slot Write command was rejected".  

JMRI effectively aborts the Ops Mode operation when it sees the LONG_ACK 'rejected' response form.  That makes sense for mobile decoders, since track power is required for Ops Mode communication.  That does not make sense for LOCONETOPSBOARD devices since they communicate via Loconet regardless of the track power state.   Therefore for LOCONETOPSBOARD devices, JMRI should treat all forms of the LONG_ACK as valid and accept the Programming Response from the Loconet device that follows any form of LONG_ACK.

closed 2021-05-29 by stale[bot]