JMRI/JMRI #5303: Create duplicate sensors using provideSensor

dsand47, 2018-05-05:
### Test setup - create script
```
import java
import jmri
sensors.provideSensor('IS:xyz')
```

### Run Test
1. Start PanelPro
2. Run script
3. Save panel file
4. Stop/start PanelPro and load panel file
5. Run script
6. Save panel file

### Results
Subsequent loads will fail: Duplicate unique value [IS:XYZ] declared for identity constraint "sensorName" of element "layout-config

### XML Content
```
  <sensors class="jmri.jmrix.internal.configurexml.InternalSensorManagerXml">
    <defaultInitialState>unknown</defaultInitialState>
    <sensor inverted="false">
      <systemName>ISCLOCKRUNNING</systemName>
    </sensor>
    <sensor inverted="false">
      <systemName>IS:XYZ</systemName>
    </sensor>
    <sensor inverted="false">
      <systemName>IS:XYZ</systemName>
    </sensor>
  </sensors>
```

### Question
Is it the application responsibility to use the resulting upper case name when using provideSensor if it is providing the entire system name or should provideSensor handle this situation?

The issue was discovered while investigating the Sections generation feature of the Signal Mast Logic table.  It creates direction sensors for each Section.  If the sections are deleted and re-generated, the duplicate sensor issue occurs.

bobjacobsen, 2018-05-05:
I’ll look at this tomorrow. It’s definitely a bug, perhaps due to pre-rationalization core left over.

bobjacobsen, 2018-05-07:
I haven't been able to recreate this, even though you provided a _great_ writeup.  Could I get a bit more info?

1) Which connection is configured? (I see that this is Internal, but there is perhaps something odd going on with the proxy setup because I can't reproduce...)
2) Which JMRI version? (I'm testing with HEAD, but I don't see much change there)

Sorry for the trouble.

dsand47, 2018-05-07:
The connection is LocoNet Simulator.

I just did the test again using a branch I created earlier today after a git pull.

dsand47, 2018-05-08:
Does the attached change fix the issue for all cases?  It works for my test case using internal sensors and turnouts and the generate sections case which creates internal sensors.

[diff.txt](https://github.com/JMRI/JMRI/files/1981894/diff.txt)

bobjacobsen, 2018-05-08:
Aha!  Got it.  The problem happens when IS:XYZ has already been provided (either directly or via a file).  OK, got this. Sorry to be slow.

bobjacobsen, 2018-05-09:
I think #5319 fixes this _for_ _Sensors_, would appreciate a test.

The larger issue is that people who advocated for "trim and uppercase to reduce the chance of user entry errors" haven't stepped up to code, so there's still a lot undone throughout the code base.

dsand47, 2018-05-09:
I have tested Jenkins build #2460 and the problem is fixed for **sensors**.  An expanded test confirms that the problem affects other objects.

dsand47, 2018-05-11:
Bob,

I can work on this if you have other priorities.

Dave Sand

dsand47, 2018-05-11:
### Update

The root problem is _**getSystemNameAddedOrderList**_.  There may still be issues with provideXXX.  The **_tsys** hash table cannot have duplicate system names but **_originalOrderList** can.

I was enhancing my python script to look for errors and discovered that the size of **getNamedBeanSet** did not change.  But **getSystemNameAddedOrderList** does.  This is what actually causes the duplicate entries after a save.

bobjacobsen, 2018-05-12:
PR #5319 has two commits in it.

d6bd2d1323e11ac was the fix for the problem.  There are two parts:  Adding a `String normalizeSystemName(@Nonnull String inputName)` method to the AbstractSensorManager and then invoking that in AbstractProxyManager as `getBeanBySystemName(String systemName)`

Previous to that, AbstractProxyManager was getting confused as to whether some Sensor with an unnormalized (i.e. lower case, extra spaces, et al) name already existed.

So the actual fix for Turnouts, et al, is to make sure that the AbstractTurnoutManager et al classes have that `String normalizeSystemName(@Nonnull String inputName)`  method.  You can just copy it over.

The work is in the 2nd commit a3b3577f622fd56.  That's the changes to tests that are needed.  There are several types:
- Add `java/test/jmri/jmrix/internal/InternalSensorManagerTest.java` and `java/test/jmri/managers/InternalSensorManagerTest.java` to specifically test the case
- Fix or `@Ignore` the tests that start failing (or fail to fail) due to the change.

dsand47, 2018-05-12:
I think I have a better description of this issue.

### Problem 1 &mdash; Duplicate XML file entries
These are caused by **_jmri.managers.AbstractManager#register_** when system names are added to **_originalOrderList** without checking for duplicates.  The duplicate entries are written to the panel xml file during save.

### Problem 2 &mdash; Duplicate beans created
These are primarily caused by incomplete checks in the provideXXX processes, but could also be caused by other requests to create beans.

The duplicate bean **replaces** the existing bean in **_tsys** (and probably _tuser) but **_NOT_** **_beans**, resulting in subtle issues such as change of state, etc.

Most of the sub classes of **_jmri.implementation.AbstractNamedBean_** include **toUpperCase()** for the system name in the **super** call.  The exceptions are either generated system names or include their own name validation before creating the bean.

bobjacobsen, 2018-05-12:
> Problem 1 — Duplicate XML file entries
> Problem 2 — Duplicate beans created

That description is how, but not _why_. 

The AbstractSensorManager et al classes are responsible for maintaining the collections of objects, including _originalOrderList. 

Yes, duplicates in those collections are bad. The only thing that changed those is the AbstractSensorManager et al implementation, so the proximate cause of any duplicates is due to new/add/delete code in those classes. 

In this case, it’s because the class was using an unnormalized name (that was different from the normalized name), and that led to a bad conclusion as to whether something already existed. 

It’s be slower than necessary (particularly since I think forcing upper case and trimming is just a bad idea), but the safest thing is to do a normalize call on every user name and system name parameter to a method of the class. But that might be the only really reliable way in the end to do this. 

Bob

dsand47, 2018-05-12:
I think the first phase is to fix the register() issue that treats duplicates in 3 different ways.
- **_beans.add(s);**  &mdash; retain the current instance, ignore the new one
- **_tsys.put(systemName, s);**  &mdash; replace the current instance with the new one
- **_originalOrderList.add(systemName);**  &mdash; Add the duplicate system name to the list

My inclination is to use a logical **replace** for all 3 lists and log an error or warn message.  This provides a quick fix to the panel XML duplicate issue.

The second phase is to fix the provideXXX logic to eliminate that source of duplicates.  Other sources will have to be dealt with when discovered.

bobjacobsen, 2018-05-12:
> _beans.add(s); — retain the current instance, ignore the new one
> _tsys.put(systemName, s); — replace the current instance with the new one
> _originalOrderList.add(systemName); — Add the duplicate system name to the list

Please don’t spend time doing this. More to follow. 

Bob

bobjacobsen, 2018-05-14:
Started a discussion on the larger direction for name-normalization on jmri-developers: https://sourceforge.net/p/jmri/mailman/message/36316791/

PR #5319 (2018-05-09): Properly (sort-of) normalize system name for internal sensors
This should fix #5303 

Long ago we (mistakenly, IMHO) decided to process system names to strip leading and following spaces and (in most, but not all cases) convert them to upper case.  This is done via various calls in Manager and NamedBean that are now present, but not always invoked when Strings are processed into SystemNames.

This fixes a couple places with that problem in the ProxyManagers, and associated tests. There are still more that need to be fixed....
closed 2021-04-14 by stale[bot]