JMRI/JMRI #5313: Panel Editor image issues

mstevetodd, 2018-05-07:
With the latest master, new panel, in Panel Editor, trying to add a background image or icon does not show the "Catalogs:" panes at all. I tried maximizing the window, etc.  Same test, in 4.10, shows the "Catalogs:" panes properly.

Also noticed that the "Find Icons", "Search File System for Icons" has no way to select hidden folders (starting with "." in Linux). Since the user prefs folder defaults to ~/.jmri, this seems to be a problem.

Note: I am not familiar with Panel Editor at all, I was testing it regarding https://groups.yahoo.com/neo/groups/jmriusers/conversations/messages/149015

cledersontp, 2018-05-10:
I found the same problem in 4.11.5:

[image]

silverailscolo, 2018-05-10:
I suspect something changed that caused ``jmri.jmrit.display.IconAdder`` to come up blank. Could be the partly replacement of ``ImageIndexEditor`` by ``CatalogTreeManager``. It occurs in all Panel Editor > Change Icon and pop-up Edit Icon... applications.
Looking at it.
The yellow strip is all that the IconAdder contains:
[image]
[image]

silverailscolo, 2018-05-11:
Got it; cause were the details of placing items on the recently improved split pane, with the last item replacing the tree. To do some clean up and testing.
[image]

mstevetodd, 2018-05-31:
@silverailscolo, Yes I tested with 4.11.6 and the problem is gone. Thank you.
P.S. I apologize for the delay. I missed this update completely.

closed 2018-05-31 by mstevetodd