JMRI/JMRI #5316: Possible issue with multiple time settings in Lights

bobjacobsen, 2018-05-08:
From jmriusers:

I'm using Ethernet connection to control arduino GPIO. My base script is IOT/JRMI_TcpPeripheral with some modifications. 
I've configured internal Lights in JMRI and everything works as expected when I turn on the specific light manually (JMRI is sending to the socket this command: [OUT:board_number:channel:state_on_off] for example [OUT:40:16:1] is sent to switch ON channel 16 on board with address 40) 
The problem occurs when I setup the LIGHT in JMRI to be switched on by Fast Clock with multiple times: 
By Fast Clock ON at 00:00, OFF at 01:00 
By Fast Clock ON at 02:00, OFF at 03:00 

Whent the Fast Clock time is 00:00 or 02:00 JMRI is sending to the socket multiple commands: 
[OUT:40:16:1] 
[OUT:40:16:0] 
[OUT:40:16:1] 
[OUT:40:16:0] 
[OUT:40:16:1] 
[OUT:40:16:0] 
[OUT:40:16:1] 
[OUT:40:16:0] 
[OUT:40:16:1] 
[OUT:40:16:0] 
[OUT:40:16:1] 
[OUT:40:16:0] 

And arduino is getting crazy switching on/off the specific channel. 

When I have only one Light Control input (f.eg. By Fast Clock ON at 00:00, OFF at 01:00) everything works as expected, the light is turned on at 00:00 and off at 01:00 (only one command is sent by JMRI. 

Why JMRI is sending the LIGHT state so many times with different status when triggered by multiple fast clock entries?

KenC57, 2018-05-08:
I sent a user the following. He replied back that my method (using Logix)
worked great.
---------
I don't use multiples via the light command. I use Logix.

You setup a condition for each time period and use the OR feature.

The Action is to turn on when true, off when false.

It works great. I do not recall the code supporting multiple light controls
that conflict. So I suspect what you have happening is one is doing the on,
but the other sees that and tries to turn it off.

One layout is using over 50 Logix to run that many lamp circuits for
building and effect lights. I think the total number of changes are a couple
of hundred different changes. Some lamps, only one cycle per day, others
might have two to eight cycles.

-Ken Cameron, Member JMRI Dev Team

icklesteve, 2019-05-23:
To get multiple Light Controls working a bit better with each other, 
the actual  if / or needs to be established at AbstractLight level, not LightControl level.

At present, each LightControl acts independently, controlling the light directly using light.setState(Light.ON); so clashes aren't resolved.

I've no intention of adding full if / or logic statements to this, Logix would be more appropriate for that sort of thing.
I think here we just need to make it obvious what it does.

In my mind, if any of the controls have their decision with light on, then it's light on, no matter if one of the controls goes decision light off.
The light should only go off when all of the controls are all decision light off.

This enables multiple cycles to be set, no chance of logic clashes, and should be fairly easy to communicate to users, ie all of the LightControls are in an OR loop to switch on.

Just so I code this right, consider the following time controls . . .

[image]

In the example above, the light would remain on from 1650 to 1655 ( without flicker ).

If a Turnout / Sensor / 2 Sensors control went Light.On or Light.Off between 1650 and 1655 this would not affect the light status.

Sounds ok?

The other option to stop the FastClock flickering would be to make the timed LightControls act as one with all others acting independently, 
though imho some LightControl types acting independently and some as a group is odd, it's easier for the user to understand if it's a simple rule.

Thoughts appreciated @KenC57 @bobjacobsen 
Steve

icklesteve, 2019-05-23:
I've re-thought this, too much potential to mess up existing logic.

If I confine logic change purely to multiple fast-clock LightControls acting as 1 control,
that'll sort the flickering out.

KenC57, 2019-05-23:
I've used Logix for any case of multiple time references. I always felt the
options built into the light control was only done because Logix had not
been created at that time. Due to all the additional features of Logix, I
felt that the light controls were left only for legacy users.

The only feature I've used on a regular basis is the min/max and ramp rates
for variable lights. Everything else is in Logix. One layout we've got 50+
building lighting circuits where we created set of Logix for patterns like:
a circuit for the bathroom for old folks/young babies. It goes on an off
about 6 times over the night for a few minutes each. Other circuits for
early worker houses, late night owls, etc... It does make a very interesting
view.

icklesteve, 2019-05-25:
Hi @KenC57 I've previously worked on some v large DMX512 installs mixing subliminal fades with effect lighting, it can take a while to get right but well worth the effort imho.
I'm mulling through a separate tweak to make that much easier in LightControls.

To get the multiple LightControls using FastClock flicker sorted, I'm now fairly sure the light should go OFF at 16:54 in the example above as this is the "last" Control command sent to the Light in time order.

[image]

At present, if the above dialogue was the only LightControl, 
if the light is found to be OFF at 01:31 for whatever reason, it'll be switched back ON.
Likewise at 02:26, if found ON for whatever reason it'll be switched OFF.
This is actually a pretty aggressive logic done very simply, I can see why people base Logix around it!

Note this differs to the support page http://jmri.org/help/en/html/tools/Lights.shtml which I'll update

> The Light will be turned on when the Fast Clock reaches the time entered in 'Time On' and will be turned off when the Fast Clock reaches the time entered in 'Time Off'.

geowar1, 2019-05-25:
The toggles at 01:31 and 02:26 are somewhat counter intuitive. Why does it do that?

Did you have to do anything special to get your Windows machine to access the USB to DMX512 dongle? I’ve had several users report problems that eventually fixed them by changing the USB port protection. Not being a Windows user I don’t know what that means or how to document it to help others that might have this issue.

icklesteve, 2019-05-25:
> The toggles at 01:31 and 02:26 are somewhat counter intuitive. Why does it do that?

Long term might be good to re-name this LightControl from "By FastClock" to "By FastClock Follower" ?
When the FastClock minute change event is sent, if the LightContol thinks the Light is in the wrong state, it changes it to whatever state it thinks should be .

I can imagine this being useful for the night-time random bathroom light scenarios, or even star shimmers, you'd only need 1 Sensor / Turnout event to trigger the Light On, knowing that the FastClock Follower will reset it back to OFF pretty soon.

Where multiple FastClock Followers are used, I'm proposing that the very most recent FastClock Follower time is authoritative, rather than each follower command the light individually ( hence no flicker ).

I'm thinking about a new simple LightControl FastClock time-based non-follower, though 1st things 1st, get the flicker sorted.

> Did you have to do anything special to get your Windows machine to access the USB to DMX512 dongle?

Sorry but I'm using Internal Lights atm for testing.
The last time I hooked up something with DMX512 to a PC it was using RS232 and Win 3.1 ;-)

There's a discussion on the MERG CBUS forum about brightness and the first CBUS Aurduino LED module capable of variable intensity has appeared, hence my recent interest in this!

Steve

icklesteve, 2021-03-12:
Closing issue pending further info.
Improved code for multiple LightControls with crossover / matching times has been in place and unit tested for a while now.

closed 2021-03-12 by icklesteve