JMRI/JMRI #5321: AWT/Swing event queue not empty in tests

bobjacobsen, 2018-05-09:
We occasionally see messages like this causing failures in Travis graphical CI:

```
[ERROR]   MemoryIconTest.testShowBlank:87 Â» TimeoutExpired Wait event queue staying empt...
```

The same message has happened in multiple places.  They trace back to code like:
```
        jf.pack();
        jf.setVisible(true);
        new org.netbeans.jemmy.QueueTool().waitEmpty(100);
```
where the fault was on the last line.

The message is telling is that the AWT/Swing event queue never got empty; something was keeping it busy.

Perhaps we've been misreading the [Jemmy JavaDoc](http://atetric.com/atetric/javadoc/org.netbeans/jemmy/2.2.7.5/org/netbeans/jemmy/QueueTool.html#waitEmpty-long-) 
> `public void waitEmpty(long emptyTime)`
> Waits for system event queue be empty for emptyTime milliseconds. Uses "QueueTool.WaitQueueEmptyTimeout" milliseconds to wait.

to mean that `emptyTime` should be a small value, e.g. 100, because we just need the queue to be empty for `emptyTime`.  Perhaps we should read that as "we can wait up to `emptyTime` for the queue to be empty, so make it longer to be sure you don't have false negatives" e.g. 5000. (The existence of the no-argument `waitEmpty()` seems to argue for the latter)

I'm not sure which is the correct reading, but the [Jemmy code is here](http://grepcode.com/file/repo1.maven.org/maven2/org.netbeans/jemmy/2.2.7.5/org/netbeans/jemmy/QueueTool.java#QueueTool.waitEmpty%28long%29) in case somebody wants to dig in and figure it out.  Or maybe experiment?

bobjacobsen, 2018-05-16:
From reading the code, it’s how long the queue _remains_ inactive.  

Perhaps we should be using `waitEmpty()`, i.e without an argument, instead?

pabender, 2018-05-16:
I always look at this as we are waiting for the event queue to settle, which means it has to be empty for some time period.

AfteFor most instances, it probably is ok for us to use the no argument waitEmpty, since we are really using this as a synchronization tool to make sure the actions we started are finished.

pabender, 2018-05-17:
@bobjacobsen
I went ahead and looked at all the instances of the QueueTool.waitEmpty.

Most of them had the time set to 100ms.  I changed all of those to an empty time.

There were two tests, which were waiting for signal related events, that have the time set to 250ms.  I left those alone.

When running the tests locally, one of the LocoNet Monitor tests failed, but otherwise everything passed.  I'm re-running the tests with that one set back to 100ms, but otherwise leaving all the parameter lists empty.  Assuming that passes, I'll push that up and PR.

Paul

pabender, 2018-05-17:
The result of my second local test run included another failure of a different test.

So, we may be able to remove the time parameter from some tests, but I think a blanket removal of the time parameter makes the tests less reliable.

bobjacobsen, 2018-05-17:
Having QueueTool.waitEmpty wait costs a bit of elapsed time, which isn't really that big of a deal yet, but it also sometimes fails entirely.  That's what I don't really understand.  

Once the queue is empty for, say, 10msec, during a test, it seems to me that it should stay that way:  There are no user interactions happening.  So I don't understand why stay-empty-for-a-long-time requests should ever fail.  Yet they do.

pabender, 2018-05-17:
@bobjacobsen 

I'm sure we still have timers running that are adding events to the event queue.  

That probably isn't the whole story however.

bobjacobsen, 2018-09-05:
Some further testing on this: `waitEmpty(10000)` extends a test by 10 seconds. It seems likely that the argument means "the queue has to be empty for this long".  

Also, a javax.swing Timer ticking keeps the waitEmpty from expiring.  I think @pabender's hypothesis about left-over Swing timers (sometimes) causing intermittent problems is correct.  Not sure how best to hunt those down...

closed 2020-01-08 by bobjacobsen