JMRI/JMRI #5346: LocoNet-using scripts need updating

bobjacobsen, 2018-05-16:
From jmriusers:

> The code was updated to allow more than one LocoNet. That makes some of those longer. The current line that says 

```
jmri.jmrix.loconet.LnTrafficController.instance().addLocoNetListener(0xFF,ln) 
```
> needs to be updated (in this case, to use the 1st LocoNet connection): 
>
```
jmri.InstanceManager.getList(jmri.jmrix.loconet.LocoNetSystemConnectionMemo).get(0).getLnTrafficController().addLocoNetListener(0xFF, ln) 
```

There are multiple script files with this issue.

silverailscolo, 2018-05-16:
I noticed, except for ``jython/InitLocoNetSensors.py`` which has 8. I think using a named variable makes it easier to add a comment and always put it at the start of the script, like:
```
# set the intended LocoNet connection by its index; when you have just 1 connection index = 0
connectionIndex = 0
```

silverailscolo, 2018-05-19:
Fixed via PR #5348. Issue can be closed.

closed 2018-05-19 by bobjacobsen