JMRI/JMRI #5359: Inconsistent handling of null Active Profile

bobjacobsen, 2018-05-19:
`ProfileManager.getActiveProfile()` can return null if there is no current profile.

`ConnectionConfigManager.savePreferences(Profile profile))` overrides `PreferencesManager.savePreferences(@Nonnull Profile profile)`, so it requires a non-null argument.

There are lots of occurrences of 
```
InstanceManager.getDefault(jmri.jmrix.ConnectionConfigManager.class)
                .savePreferences(ProfileManager.getDefault().getActiveProfile()
);
```
which takes the could-be-null value from `getActiveProfile()` and passes it unchecked to `savePreferences(..)` hence can cause an error.

The Profile system isn't well documented at a high-level (there's only a bit in [the preferences page](http://jmri.org/help/en/html/doc/Technical//AppPreferences.shtml), so a good starting point on this might be to figure out & document how Profiles propagate in the code and when the active one can be null.  Is that relevant any more?

danielb987, 2018-08-03:
Is it a valid state to not have a current profile?

A new profile is created when installing JMRI. And as far as I can see, the current profile cannot be deleted from within JMRI.

The current profile can be deleted by the user outside of JMRI, but that could be checked by JMRI upon startup.

So my suggestion is that JMRI requires a valid current profile at startup and ask the user what to do if it doesn't exist (create new, add existing or exit). And that ProfileManager.getActiveProfile() throws an exception instead of returning null if a current profile does not exists.

danielb987, 2018-08-03:
I created a branch that makes null active profile invalid.
https://github.com/danielb987/JMRI/tree/inconsistent_handling_of_null_active_profile_5359

This will break ProfileManagerTest at line 35 and at line 49.

Is it recommended to create a PR for discussions? Or is it better to only link to my branch?

bobjacobsen, 2018-08-03:
Great that you're looking into this.

IIRC, the reason `null` was left there for no-profile is that users who were updating from pre-profile versions of JMRI needed that to be able to get past the initialization.  Profiles were added in early 2015. It's been three years and six production versions since then.  We're telling people with older versions to update to 4.12 first, before updating to 4.13.1 or later.  So I think that's not a constraint on improving this code any more.

Working from a visible (i.e. on GitHub) branch is fine.  Eventually a WIP PR gets you automated Travis, AppVeyor and coverage tests, but it's not necessary until you're happy with a proposed change.

Clearly, updating `ProfileManagerTest.java` to handle the new behavior is appropriate.  Are there other tests that this will break?  Perhaps we'll have to make sure that some tests get more initialization so that there's really a profile, instead of just ignoring that.  But it would be worth it to get this right.  I opened the original Issue because there were multiple spots where the possible null value was generating FindBugs/SpotBugs warnings.  The goal is to get rid of all those so it's easier to see when real problems creep into the code.

Also, as you work this through, you'll probably learn stuff about how the preferences are working.  If would be great to add what you figure out to the [preferences documentation](http://jmri.org/help/en/html/doc/Technical/AppPreferences.shtml), the [start-up scripts page](http://jmri.org/help/en/html/doc/Technical/StartUpScripts.shtml) or even to start a new page on profiles.

Thanks again!

danielb987, 2018-08-03:
I found the ProfileManagerTest.java by doing a usage search of ProfileManager.setActiveProfile() and the test calls setActiveProfile((String)null) and setActiveProfile((String)null). Other calls from other classes to setActiveProfile() seems to have a valid Profile as parameter.

I have not yet found another test that fails because of my changes, but I will create a WIP PR to be able to run the tests since I have problems running the tests on my local computer.

I'm concerned about AppsBase.configureProfile() at line 191. If there is no profile and there is no GUI, I get the impression that the application starts without a profile. Is this correct and if so, how should this situation be handled? Print an error message to the console and abort the program?

danielb987, 2018-08-03:
PR #5604
I get a lot of errors that activeProfile is null.
https://travis-ci.org/JMRI/JMRI/jobs/411764082

Should I create an empty profile in JUnitUtil.setUp() ? In order to create a new Profile, I need to give a path. Should I add a path, and in that case which path? Or should I create a sub class of Profile that isn't able to store the settings to disk?

bobjacobsen, 2018-08-03:
It's not as many places as I feared.  Most of them are in ConfigureXML classes, which can probably be fixed once bypassing the Profile reference (it's not really needed for anything for those tests).  Will take a look in more detail later today.  

I'm a bit hesitant to recommend doing something for _every_ test (i.e. in setUp)  just because of the length of time it'd take if we're doing I/O for every test.

bobjacobsen, 2018-08-03:
I was able to make (most of?) those test failures happy by adding

```
          jmri.util.JUnitUtil.resetProfileManager()
```

in these files:
```
	modified:   java/test/jmri/configurexml/LoadAndCheckTest.java
	modified:   java/test/jmri/configurexml/LoadAndStoreTestBase.java
	modified:   java/test/jmri/configurexml/SaveMenuTest.java
	modified:   java/test/jmri/configurexml/StoreXmlAllActionTest.java
	modified:   java/test/jmri/configurexml/StoreXmlConfigActionTest.java
```

That's now in the `inconsistent_handling_of_null_active_profile_5359` branch in the bobjacobsen/JMRI GitHub repository if you'd like to pick it up.

danielb987, 2018-08-04:
Thanks!

The PR #5604 now passes Travis and AppVeyor tests.

closed 2018-08-05 by bobjacobsen