JMRI/JMRI #5371: Can't use signal files from User Files location?

bobjacobsen, 2018-05-22:
From [jmriusers]():

> Rather than adding 'Approach' as a speed name I've edited BR-2003 aspects.xml to use existing speed names.  I've had to edit the version in the application directory as I cannot get JMRI to use the version I've put in the User Files Location as defined in preferences despite documentation suggesting that this should be possible.  Unfortunately this means I have to re-apply the edits for new releases of JMRI.

dsand47, 2018-05-22:
There was a discussion on this issue about a year ago.  I am not able to find it.

A review of the code confirms the results of the discussion.

The signal systems are loaded with the JMRI supplied systems first followed by any **unique** user systems in preference:resources/signals.  This includes the aspects.xml file.  When an **appearance** is selected, the user files location is checked first.  This way a custom appearance file can override the supplied one.

Common changes include a different set of icons or changes to the AspectMapping rules.

pnbgit, 2018-06-08:
Dave helped me a while back with having a custom signal system in preference:   In this I fixed the BR2003 aspect speeds as well as  having more icon choices.

The aspect speeds in the standard BR2003 signal system as supplied do need fixing, every aspect is se to the none existent speed "Approach". I have a fixed aspects.xml file - how should I submit it for inclusion in the next release?

dheap, 2018-06-10:
@bobjacobsen In JMRI we generally prefer a file in a user's directory over one in a system directory.
>DefaultSignalSystemManager looks in both system and user directories for aspects.xml. It will take a signal system from either, but if there are two of the same name (which is the directory name), it'll prefer the system one. (Should we change that, i.e. prefer the users version of a signal system?)

KenC57, 2018-06-11:
I think the general concept we've told users is to not touch the system
space and that stuff in the user local space would supersede that. The goal
is letting them replace parts with new (or old) copies they have privately
in the user space. And the user space mirrors the paths in the system space.

At least that's the way I've seen us try to move things. And generally
that's what we've been leading the users to expect over the years. Now we've
known this is a shifting target and while the direction, some things we've
not had or taken the time to make work that way fully.

So continuing that move seems the right way to be heading.
-ken c

closed 2021-04-14 by stale[bot]