JMRI/JMRI #5381: System running OpenLCB gets pointless warning about 'all internal' selected

KenC57, 2018-05-26:
And the preferences pops up too. So if only a single connection, OpenLCB, there is nothing but Internal, so it is correct. I wonder if it is possible for it to know this is ok?
-ken c

SpudsMcCat, 2018-06-30:
I'm assuming that where Decoderco reads/write programming to/from.  There's been talk about an LCC command station, but IDK if ever got added to the standard or not. Perhaps put in a stub "put LCC command station code here" ?

SpudsMcCat, 2018-06-30:
alternativly, an acknolagable popup/message "you have no method chosen that supports programming mobile decoders" ? Though this may quickly scope-creep.

pabender, 2018-06-30:
@KenC57 does the same thing happen if you set the system up as C/MRI or any of the other systems that only support sensors and turnouts?

pabender, 2018-10-09:
This appears to be a simple configuration issue on OpenLCB.  OpenLCB provides both a programmer and a throttle, so those should be selectable as defaults.

There are two classes that handle configuration of the OpenLCB interface
https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrix/openlcb/OlcbConfigurationManager.java
https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrix/openlcb/OlcbSystemConnectionMemo.java

The OpenLCB Throttle Manager constructor expects both a memo and a configuration manager to be passed at construction time.

At first glance, I think the problem is that the memo should defer to the configuration manager for both the get and provides operations.  I am not sure that is all there is to this fix however.

pabender, 2018-10-12:
@KenC57 PR #5982 (now merged) makes it possible for OpenLCB systems to enable the Throttle manager as a default, so this warning is no longer pointless, as a result, I am going to close this issue.

closed 2018-10-12 by pabender