JMRI/JMRI #5382: Tools > Check Unnecessary Anchors False Positive on Decorations

icklesteve, 2018-05-26:
Originally posted in https://groups.io/g/jmriusers/message/148226

Layout Editor

Tools > Check Unnecessary Anchors 
False Positive on Decorations ( in my case tunnels )
Where 2 lengths of track both assigned to the same block but with different decorations.
One stretch of track is inside the tunnel, the other outside the tunnel, not decorated.
Spotted in v 4.11.7ish

icklesteve, 2018-06-04:
Tested in recent Jenkins build, confirm issue resolved, thanks @geowar1 :-)

closed 2018-06-04 by icklesteve

PurkissA, 2020-08-02:
This problem appears to have returned in the same situation.

icklesteve, 2020-08-04:
Hi @PurkissA 

I'm unable to reproduce the documented issue in current head.

Please raise a new issue ( as there could well be something unrelated happening ).
If you could attach to it a minimal panel demonstrating the problem that would help to track down,

Steve.

PurkissA, 2020-08-04:
This is not a new problem it is the same one mentioned. If you draw three horizontal segments, make outside 2 continuous lines and the middle dashed line. Then do check for unnecessary anchors it doesn't show them. Agreed. However if you snap network to grid, i.e. make sure there are no bends at all at the anchor points then both middle anchors are shown as unnecessary in the check. It's no good making the lines horizontal by eye. Conclusion the check code only checks the angles not being 180 degrees, not the line types. Latest production version. This has been reproduced in new panel so there are no outside influences.

icklesteve, 2020-08-05:
Hi @PurkissA 

> This is not a new problem it is the same one mentioned.
The outcome ( a false positive ) may be the same, however the fix will definitely not be the same, otherwise it would be working now.

The issue I had was to do with mixed decoration blocks, was irrespective of track orientation, did not continuous outside lines, did not use dashed lines and did not consider snap to grid.

Please raise a new PR to log your issue, the code has changed a lot in over 2 years since this incidence of the false positive was resolved.

Thanks,
Steve.

PurkissA, 2020-08-05:
I'm sorry if this is not the same problem. I am a new user of JMRI and I do not know what is meant by decorations hence the original description appears to describe the issue.

geowar1, 2020-08-05:
Dashed track segments are not implemented via decorations; decorations are bridge, tunnel, end bumpers and end arrows. I can add the dash matching exception to the check case so these aren't listed if their dashed attributes don't match.