JMRI/JMRI #5407: Block values do not completely clear on unoccupied

icklesteve, 2018-05-29:
Hi Folks,
v 4.11.7ish

Initially raised on JMRI Users https://groups.io/g/jmriusers/message/148294

When a block becomes unoccupied, the value is often being set to "  " ( excluding the quotes, it's 2 spaces ).

When rendered in a Layout Editor panel, this is being drawn with background as a few dots

[image]

The block contents renders as square with a few dots at the bottom.
The same Block Contents icon with a completely clear value is not displayed, hence the track line is normally continuous.

This issue is only visible with the Block Contents Icon background set to 0 transparency ( solid background so is readable without the track line getting in the way ).

I imagine fix would be in trimming the block contents before writing to the block table, 
or trimming before checking if need to display the block contents icon on panel.
The value is already trimmed for the JMRI Webserver version of the panel which displays correctly ( ie does not display the empty box ).

Ta, Steve

dsand47, 2018-05-30:
Steve,

When a block becomes unoccupied, the block routing logic sets the value to null.  Do you have any other processes, such as python, managing the block content?

What other attributes have been set on the block contents icon?  Borders, fixed size, etc.

Dave Sand

icklesteve, 2018-05-30:
Hi Dave,

No transponding used.

You might be onto something, I do use Dispatcher + Transits

I can't recall this happening when I've had my layout in DC mode when I've been operating trains myself, I always use dispatcher on DCC!

Ta,
Steve

icklesteve, 2018-05-30:
Ok, tested with just a throttle and manually entering the block value in to start a test run for 1 hour, no issues.
Exactly the same loop but being drive by Autotrains / dispatcher and the spaces in the block values appeared after the 1st block became unoccupied, so this definitely a Dispatcher issue,
Steve

dsand47, 2018-05-30:
Steve,

I have found the source for the two-space block value: jmri.Section#clearNameInUnoccupiedBlocks.

Almost two years ago, this code was changed to clear block values instead of the optional memory variable.  The spaces had been used to clear the memory variable.  The correct value for an unoccupied block is **_null_**.  

The fix is easy but there will be a side effect.  The default block contents label is invisible when unoccupied (null).  When it has any other value, including spaces, it is visible.  People might be relying on that bug.  I use borders or a fixed size to maintain visibility when needed.

Dave Sand

icklesteve, 2018-05-30:
Hi Dave and well done for finding :-)

Personally I think fix it,

Null describes the block value a lot more accurately than the 2 spaces.
Having 2 different ways of having an unoccupied block value encourages inconsistency!

In the Webserver panel, it's already invisible.

If the block describers disappear because someone has been relying on the bug, they would already be disappearing when switching to non Auto-active.
They're not permanently hidden for an upgrading user as they will re-appear when a train enters the block which will enable them to locate and edit as appropriate.

The invisible label when unoccupied is exactly the effect I'm after to have a modern IECC look to my panels.
With multiple running lines it's so much easier to see which train is which rather than placing the icon at the side of the track.
As I found out in testing yesterday, there's no other way of achieving this look purely through style options with the 2 spaces still in place!

That's my 2 pence worth but of course I will give way to more experienced users.
An easy workaround to achieve the look would be handy if the current 2 spaces system is to be maintained.

Thanks again for looking into this,
Steve

dsand47, 2018-05-30:
PR #5409 has been created to resolve this issue.

icklesteve, 2018-06-08:
Tested now working as expected, thanks all :-)

closed 2018-06-08 by icklesteve