JMRI/JMRI #5416: Decoder definition for Märklin mSD3

finnef, 2018-06-02:
Here is my decoder definition for use with DecoderPro (file>import decoder file) to program a Märklin mSD3 decoder. It should also work with an mLD3, although the sound panels won't do anything, and the function mapping references to sounds won't work. 
Someone might create a separate mLD3 definition from this mSD3.

I tested this with my mSD3 decoder and it works fine.
[Marklin mSD3 JMRI decoder definition.zip](https://github.com/JMRI/JMRI/files/2065304/Marklin.mSD3.JMRI.decoder.definition.zip)

AlanUS, 2018-06-02:
@finnef 
Thank you for your contribution, but this decoder is already part of JMRI since version 4.11.5 a couple of months ago (#5193).

AlanUS, 2018-06-02:
I have not checked in detail your own definition, but I just want to point our that it won't work fully with JMRI because Märklin is NOT a recognized brand on its own by the NMRA i.e. CV8 will actually read 'Trix'.

You are very welcome to check and enrich the definitions that I have created.
Currently the available languages for this decoder are German, English and French. I you want to add more, I'm sure that several users will be grateful. Märklin manual is available in many more languages than these 3.

finnef, 2018-06-02:
ah shame - double effort... I did not check the test version - just stable. But it looks like your definition is more complete so unless I find anything that you could add from my version - ignore this one.

AlanUS, 2018-06-02:
Well yes I guess it took you some time, as this is not a simple decoder...
Also stay tuned on the JMRI user list (now on groups.io) as I clearly announced that I was working on this one and when I released it.

finnef, 2018-06-02:
Yeah I joined the yahoo group but there was so much activity that I left again... I'll try groups.io though

finnef, 2018-06-02:
The only thing that is slightly easier in my version is that I read and write the function mapping lines directly, so they are updated when you write a line. So I don't need to re-read the full pane. I ignore CV's 33-37. According to the manual I found on the Marklin site you can read/write directly to the lines in DCC mode. I tested this and it seems to work fine.

AlanUS, 2018-06-02:
I do not own this decoder myself and the definition was tested by another user, who did not report this possibility. It is quite easy to set all function mapping CVs back to read write.

closed 2019-05-31 by bobjacobsen