JMRI/JMRI #5422: LE Decorations – Buffer stop width negative value causes LE Freeze

icklesteve, 2018-06-03:
Hi Folks,

I found that I am reliably able to freeze Layout editor by setting buffer width to a negative number.

Error log growing too fast for Raspberry Pi to keep up with, even after checking do not scroll the error log text was all over the place.  I managed to grab the 1st bit of it :

	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
2018-06-03 02:19:51,616 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.IllegalArgumentException: negative width
	at java.awt.BasicStroke.<init>(BasicStroke.java:194)
	at java.awt.BasicStroke.<init>(BasicStroke.java:249)
	at jmri.jmrit.display.layoutEditor.TrackSegment.drawDecorations(TrackSegment.java:2415)
	at jmri.jmrit.display.layoutEditor.LayoutEditor.drawDecorations(LayoutEditor.java:9851)

Was able to close JMRI via main PanelPro menu.
Re-started JMRI and found able to reproduce the issue by re-setting the negative number,

Steve

icklesteve, 2018-06-04:
Thanks for that Ken C, have located where the log files are kept.

Have just tested all other decoration attributes, found 2 more which could do with having a minimum value of 0 or 1 :

Bridge Line Width negative value also freezes LE
Tunnel Line Width negative value also freezes LE

Steve

geowar1, 2018-07-30:
There is no buffer width… Can you post screen show where you set this?

geowar1, 2018-07-30:
The arrow, bridge & tunnel width/lengths have already been fixed (PR?).
If you can confirm the "buffer width" as fixed then this can be closed.

icklesteve, 2018-07-30:
Still issue with Bufferstop negative line width causing freeze, have been accessing the inputs for these via the track segment right click menu.
Buffers

[image]

[image]

Bridge
[image]

Tunnel
[image]

All of these cause the java.lang.IllegalArgumentException: negative width though it seems the buffer stops effect me more as I have more of these than tunnels or bridges,

Steve.

geowar1, 2018-08-13:
I think all of these have been fixed… Close if so… comment if not… Thanks.

icklesteve, 2018-08-15:
Hi George,

Have recently been running via ant so to rule out clean issues deleted the profile directory and installed latest Jenkins v 4.13.3ish.

To confirm still able to reproduce the issues with negative numbers entered for
Bridge line width, 
Bufferstop line width,
Tunnel line width.
Arrows, tested ok on 4th June still fine :-)

Would it help if I uploaded a test panel?

Thanks,
Steve.

geowar1, 2018-08-15:
See PR #5651

icklesteve, 2018-10-28:
Thanks George, have tested with merged commit and has fixed the issue,
Steve.

closed 2018-10-28 by icklesteve