JMRI/JMRI #5423: Some Json tests fail to find schema

pabender, 2018-06-03:
I’ve seen both Travis and Appveyor fail on multiple PRs with ( this is from an Appveyor build ):

    [junit] Testcase: testOnMessage(jmri.server.json.schema.JsonSchemaSocketServiceTest):	Caused an ERROR
3275    [junit] java.net.UnknownHostException: json-schema.org
3276    [junit] com.networknt.schema.JsonSchemaException: java.net.UnknownHostException: json-schema.org
3277    [junit] 	at com.networknt.schema.JsonSchemaFactory.getSchema(JsonSchemaFactory.java:203)
3278    [junit] 	at com.networknt.schema.RefValidator.getRefSchema(RefValidator.java:65)
3279    [junit] 	at com.networknt.schema.RefValidator.<init>(RefValidator.java:45)
3280    [junit] 	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
3281    [junit] 	at com.networknt.schema.ValidatorTypeCode.newValidator(ValidatorTypeCode.java:121)
3282    [junit] 	at com.networknt.schema.JsonMetaSchema.newValidator(JsonMetaSchema.java:241)
3283    [junit] 	at com.networknt.schema.ValidationContext.newValidator(ValidationContext.java:24)
3284    [junit] 	at com.networknt.schema.JsonSchema.read(JsonSchema.java:116)
3285    [junit] 	at com.networknt.schema.JsonSchema.<init>(JsonSchema.java:56)
3286    [junit] 	at com.networknt.schema.JsonSchema.<init>(JsonSchema.java:49)
3287    [junit] 	at com.networknt.schema.PropertiesValidator.<init>(PropertiesValidator.java:35)
3288    [junit] 	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
3289    [junit] 	at com.networknt.schema.ValidatorTypeCode.newValidator(ValidatorTypeCode.java:121)
3290    [junit] 	at com.networknt.schema.JsonMetaSchema.newValidator(JsonMetaSchema.java:241)
3291    [junit] 	at com.networknt.schema.ValidationContext.newValidator(ValidationContext.java:24)
3292    [junit] 	at com.networknt.schema.JsonSchema.read(JsonSchema.java:116)
3293    [junit] 	at com.networknt.schema.JsonSchema.<init>(JsonSchema.java:56)
3294    [junit] 	at com.networknt.schema.JsonSchema.<init>(JsonSchema.java:49)
3295    [junit] 	at com.networknt.schema.JsonSchema.<init>(JsonSchema.java:44)
3296    [junit] 	at com.networknt.schema.JsonSchemaFactory.newJsonSchema(JsonSchemaFactory.java:144)
3297    [junit] 	at com.networknt.schema.JsonSchemaFactory.getSchema(JsonSchemaFactory.java:208)
3298    [junit] 	at jmri.server.json.schema.JsonSchemaServiceCache.getSchema(JsonSchemaServiceCache.java:120)
3299    [junit] 	at jmri.server.json.schema.JsonSchemaServiceCache.validateJsonNode(JsonSchemaServiceCache.java:162)
3300    [junit] 	at jmri.server.json.schema.JsonSchemaServiceCache.validateMessage(JsonSchemaServiceCache.java:154)
3301    [junit] 	at jmri.server.json.schema.JsonSchemaServiceCache.validateMessage(JsonSchemaServiceCache.java:148)
3302    [junit] 	at jmri.server.json.JsonMockConnection.sendMessage(JsonMockConnection.java:61)
3303    [junit] 	at jmri.server.json.schema.JsonSchemaSocketService.onMessage(JsonSchemaSocketService.java:34)
3304    [junit] 	at jmri.server.json.schema.JsonSchemaSocketServiceTest.testOnMessage(JsonSchemaSocketServiceTest.java:61)
3305    [junit] Caused by: java.net.UnknownHostException: json-schema.org
3306    [junit] 	at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:184)
3307    [junit] 	at java.net.PlainSocketImpl.connect(PlainSocketImpl.java:172)
3308    [junit] 	at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
3309    [junit] 	at java.net.Socket.connect(Socket.java:589)
3310    [junit] 	at java.net.Socket.connect(Socket.java:538)
3311    [junit] 	at sun.net.NetworkClient.doConnect(NetworkClient.java:180)
3312    [junit] 	at sun.net.www.http.HttpClient.openServer(HttpClient.java:463)
3313    [junit] 	at sun.net.www.http.HttpClient.openServer(HttpClient.java:558)
3314    [junit] 	at sun.net.www.http.HttpClient.<init>(HttpClient.java:242)
3315    [junit] 	at sun.net.www.http.HttpClient.New(HttpClient.java:339)
3316    [junit] 	at sun.net.www.http.HttpClient.New(HttpClient.java:357)
3317    [junit] 	at sun.net.www.protocol.http.HttpURLConnection.getNewHttpClient(HttpURLConnection.java:1220)
3318    [junit] 	at sun.net.www.protocol.http.HttpURLConnection.plainConnect0(HttpURLConnection.java:1156)
3319    [junit] 	at sun.net.www.protocol.http.HttpURLConnection.plainConnect(HttpURLConnection.java:1050)
3320    [junit] 	at sun.net.www.protocol.http.HttpURLConnection.connect(HttpURLConnection.java:984)
3321    [junit] 	at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1564)
3322    [junit] 	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1492)
3323    [junit] 	at java.net.URL.openStream(URL.java:1045)
3324    [junit] 	at com.networknt.schema.url.StandardURLFetcher.fetch(StandardURLFetcher.java:14)
3325    [junit] 	at com.networknt.schema.JsonSchemaFactory.getSchema(JsonSchemaFactory.java:186)
3326    [junit] 
3327    [junit] 
3328    [junit] Testcase: testOnMessage(jmri.server.json.schema.JsonSchemaSocketServiceTest):	Caused an ERROR
3329    [junit] Unexpected ERROR or FATAL messages emitted
3330    [junit] java.lang.AssertionError: Unexpected ERROR or FATAL messages emitted
3331    [junit] 	at apps.tests.Log4JFixture.tearDown(Log4JFixture.java:38)
3332    [junit] 	at jmri.util.JUnitUtil.tearDown(JUnitUtil.java:118)
3333    [junit] 	at jmri.server.json.schema.JsonSchemaSocketServiceTest.tearDown(JsonSchemaSocketServiceTest.java:36)

I assume this is a transient error, but we need to provide a local copy of the information being requested from Json-schema.org.

This test will fail on any host running it that is disconnected from the internet.  

It is possible this will cause runtime errors on live systems that are disconnected from the internet as well.

pabender, 2018-06-03:
This does not appear to be a transient error.  Json-schema.org appears to no longer exist.

pabender, 2018-06-03:
I will try to get this resolved this afternoon ( US eastern time ).

bobjacobsen, 2018-06-03:
Looks like they lost their domain name entry.  

There's been recent activity on their [git hub project](), so maybe they'll figure it out and come back. But we do need an alternative for testing and running w/out internet connections.

@pabender Thanks!

pabender, 2018-06-04:
I didn’t figure out how to actually fix this test so that it doesn’t look for json-schema.org, because the actual request appears to be deep inside a library that doesn’t advertise as throwing exceptions.

For now, #5428 disables the one test that was failing.  

I also went ahead and notified the json-schema.org folks that the site was down: https://github.com/json-schema-org/JSON-Schema-Test-Suite/issues/229

bobjacobsen, 2018-06-10:
[json-schema.org](http://json-schema.org) seems to be back up.

pabender, 2018-06-12:
PR #5472 re-enables this test with an assume check that the host is reachable.  That should significantly reduce the risk of this causing a problem in the future.

I am going to go ahead and close this issue.

closed 2018-06-12 by pabender

Relequestual, 2021-08-17:
@pabender I'm curious, why are you making requests to json-schema.org as part of tests?

pabender, 2021-08-19:
> @pabender I'm curious, why are you making requests to json-schema.org as part of tests?

This is a very old Issue.  I had to remind myself of what was happening.  The issue addressed here was with a test that someone else wrote.  That issue was breaking builds, and the last PR referenced fixed the build.

There might be other ways to fix this, but the bottom line is I would rather have some test of the functionality than an ignored test.  You are welcome to re-write the test ( ans/or the underlying class under test ) to make the process better.

Relequestual, 2021-08-19:
> You are welcome to

Yeahhhh I got my own work to be doing 😅
Building JSON Schema.