JMRI/JMRI #5424: LE Disabled turnout not respected in Web Server

icklesteve, 2018-06-03:
Hi Folks,

JMRI v 4.11.7ish 20180525
Java v 1.8.0_65

Within LE the disable turnout options work as expected.
[image]

When panel viewed within a web browser the circles appear around the turnouts, enabling them to be changed.
[image]

Same issue when tested with disable when occupied, web server CAN change the turnouts when occupied.
JMRI re-started after changing LE disabled turnout settings to ensure web server has fresh layout version.

Steve

mstevetodd, 2018-06-03:
@icklesteve, please attach an example panel showing this issue and I'll take a look at it.

icklesteve, 2018-06-03:
Hi @mstevetodd many thanks for having a look at this, have attached xml : 

[test-panel-turnouts-web-server.zip](https://github.com/JMRI/JMRI/files/2066128/test-panel-turnouts-web-server.zip)

Just realised this test panel created in 4.11.6, however same behaviour as 4.11.7,

Steve

bobjacobsen, 2018-06-04:
PR #5430 by @mstevetodd, now merged, is meant to fix this.

icklesteve, 2018-06-04:
Hi @mstevetodd 

Tested using latest build 20180604T1452

Disabled turnout now not working in webserver panel view, perfect :-)

I then tested the disabled when occupied box, unfortunately this doesn't operate the same as LE :
[image]

I imagine this is a more difficult fix so shall I raise on new issue?

( test xml file posted above includes an internal block sensor for the panel )
nb, JMRI re-started + browser cache cleared after changing to disable when occupied.

Thanks,
Steve.

mstevetodd, 2018-06-04:
@icklesteve 
You are correct, that the occupancy check is a bit more involved, since the disabled only need to be set when the panel was loaded. The occupancy check will need to be handled each time. Doable.
I assume I'll need to modify your panel to test this? I see the disableWhenOccupied flag, but it is set to "no".
No need for an additional issue.

icklesteve, 2018-06-04:
Hi @mstevetodd 

Correct, only change needed to test panel turnout is 
disabled="no" disableWhenOccupied="yes"

Major apologies for creating more work for you, I appreciate it could be a fair bit of doing but if it works like the rest of the browser panel I'm sure the end result will be worth it.
imho as a user may be relying on this as a safety feature better to report sooner rather than later!

Steve

mstevetodd, 2018-06-05:
@icklesteve This change is now in the latest Jenkins build, which I've loaded on my demo server. I also changed all layoutturnouts on my panel to disableWhenOccupied="yes", and one of them "W Industry (76)" to disabled. You can fiddle with them at http://jmri.mstevetodd.com:1080/panel/Layout/Waccamaw%20Coast%20Line
Please give it a go and then close this issue if it looks good to you.

icklesteve, 2018-06-05:
Hi @mstevetodd 
That's timing for you, I just spotted that the Jenkins build was ready and had tested the mini panel I posted, all working as expected :-)
Loving your demo server, I haven't had a play with Frames but your demo shows off the capabilities of them really well :-)

Thanks again for the quick fixes,
Steve

closed 2018-06-05 by icklesteve