JMRI/JMRI #5426: Deadlock during startup

bobjacobsen, 2018-06-03:
Ken Cameron reported a deadlock on startup via the jmri-developers list.  Posts to follow.

bobjacobsen, 2018-06-03:
On May 21, 2018, at 8:03 AM, Ken Cameron <kcameron@twcny.rr.com> wrote:

I'm getting as far as picking a profile, but after that, it hangs. Using
jvisualvm, I got the following thread dump. Can somebody figure it out?

2018-05-21 11:00:49
Full thread dump Java HotSpot(TM) 64-Bit Server VM (25.171-b11 mixed mode):

"AWT-EventQueue-0" #18 prio=6 os_prio=0 tid=0x000000001db1d800 nid=0xab0
waiting for monitor entry [0x000000001fd6d000]
 java.lang.Thread.State: BLOCKED (on object monitor)
      at jmri.InstanceManager.getInstances(InstanceManager.java:765)
      - waiting to lock <0x0000000082eb8980> (a java.lang.Class for
jmri.UserPreferencesManager)
      at jmri.InstanceManager.getInstance(InstanceManager.java:256)
      at jmri.InstanceManager.getNullableDefault(InstanceManager.java:230)
      at jmri.InstanceManager.getDefault(InstanceManager.java:201)
      at
jmri.managers.configurexml.DefaultUserMessagePreferencesXml.load(DefaultUser
MessagePreferencesXml.java:50)
      at
jmri.configurexml.ConfigXmlManager.loadOnSwingThread(ConfigXmlManager.java:6
99)
      at
jmri.configurexml.ConfigXmlManager.lambda$1(ConfigXmlManager.java:610)
      at
jmri.configurexml.ConfigXmlManager$$Lambda$49/1028780142.run(Unknown Source)
      at java.awt.event.InvocationEvent.dispatch(Unknown Source)
      at java.awt.EventQueue.dispatchEventImpl(Unknown Source)

 Locked ownable synchronizers:
      - None

"main" #1 prio=5 os_prio=0 tid=0x0000000005203800 nid=0x1374 in
Object.wait() [0x000000000516e000]
 java.lang.Thread.State: WAITING (on object monitor)
      at java.lang.Object.wait(Native Method)
      - waiting on <0x00000000831029f0> (a
java.awt.EventQueue$1AWTInvocationLock)
      at java.lang.Object.wait(Unknown Source)
      at java.awt.EventQueue.invokeAndWait(Unknown Source)
      - locked <0x00000000831029f0> (a
java.awt.EventQueue$1AWTInvocationLock)
      at java.awt.EventQueue.invokeAndWait(Unknown Source)
      at javax.swing.SwingUtilities.invokeAndWait(Unknown Source)
      at
jmri.configurexml.ConfigXmlManager.load(ConfigXmlManager.java:608)
      at
jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.j
ava:231)
      at
jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.j
ava:170)
      at
jmri.managers.JmriUserPreferencesManager.readUserPreferences(JmriUserPrefere
ncesManager.java:867)
      at
jmri.managers.JmriUserPreferencesManager.initialize(JmriUserPreferencesManag
er.java:1133)
      at jmri.InstanceManager.getInstance(InstanceManager.java:318)
      - locked <0x0000000082eb8980> (a java.lang.Class for
jmri.UserPreferencesManager)
      at jmri.InstanceManager.getNullableDefault(InstanceManager.java:230)
      at jmri.InstanceManager.getOptionalDefault(InstanceManager.java:369)
      at jmri.util.JmriJFrame.setFrameLocation(JmriJFrame.java:169)
      at jmri.util.JmriJFrame.<init>(JmriJFrame.java:115)
      at jmri.util.JmriJFrame.<init>(JmriJFrame.java:124)
      at
jmri.jmrix.loconet.hexfile.HexFileFrame.<init>(HexFileFrame.java:37)
      at
jmri.jmrix.loconet.hexfile.configurexml.ConnectionConfigXml.load(ConnectionC
onfigXml.java:68)
      at
jmri.jmrix.ConnectionConfigManager.initialize(ConnectionConfigManager.java:1
06)
      at
jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigur
ationManager.java:264)
      at
jmri.implementation.JmriConfigurationManager.lambda$3(JmriConfigurationManag
er.java:260)
      at
jmri.implementation.JmriConfigurationManager$$Lambda$33/1983025922.accept(Un
known Source)
      at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(Unknown
Source)
      at java.util.stream.ReferencePipeline$Head.forEach(Unknown Source)
      at
jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigur
ationManager.java:259)
      at
jmri.implementation.JmriConfigurationManager.lambda$3(JmriConfigurationManag
er.java:260)
      at
jmri.implementation.JmriConfigurationManager$$Lambda$33/1983025922.accept(Un
known Source)
      at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(Unknown
Source)
      at java.util.stream.ReferencePipeline$Head.forEach(Unknown Source)
      at
jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigur
ationManager.java:259)
      at
jmri.implementation.JmriConfigurationManager.lambda$1(JmriConfigurationManag
er.java:183)
      at
jmri.implementation.JmriConfigurationManager$$Lambda$32/1906808037.accept(Un
known Source)
      at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(Unknown
Source)
      at java.util.stream.ReferencePipeline$Head.forEach(Unknown Source)
      at
jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.j
ava:182)
      at
jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.j
ava:170)
      at apps.Apps.<init>(Apps.java:252)
      at apps.PanelPro.PanelPro.<init>(PanelPro.java:40)
      at apps.PanelPro.PanelPro.main(PanelPro.java:120)

 Locked ownable synchronizers:
      - None

bobjacobsen, 2018-06-03:
On May 21, 2018, at 11:52 AM, Ken Cameron <kcameron@twcny.rr.com> wrote:

Here is a trace with NCE USB, it is the right port, but not responding
either. It looks like the 'blocked at getInstances' like the others. One was
a NCE sim, I could see the send and receive threads, but also waiting at
getInstances.

-ken c

2018-05-21 14:44:17
Full thread dump Java HotSpot(TM) 64-Bit Server VM (25.171-b11 mixed mode):

"Swing-Shell" #24 daemon prio=5 os_prio=0 tid=0x000000001eadf000 nid=0x267c
waiting on condition [0x00000000228ff000]
  java.lang.Thread.State: WAITING (parking)
       at sun.misc.Unsafe.park(Native Method)
       - parking to wait for  <0x0000000083b604a8> (a
java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject)
       at java.util.concurrent.locks.LockSupport.park(Unknown Source)
       at
java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject.await(
Unknown Source)
       at java.util.concurrent.LinkedBlockingQueue.take(Unknown Source)
       at java.util.concurrent.ThreadPoolExecutor.getTask(Unknown Source)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown
Source)
       at sun.awt.shell.Win32ShellFolderManager2$ComInvoker$3.run(Unknown
Source)
       at java.lang.Thread.run(Unknown Source)

  Locked ownable synchronizers:
       - None

"TimerQueue" #20 daemon prio=5 os_prio=0 tid=0x000000001d97e800 nid=0x22f8
waiting on condition [0x00000000210ae000]
  java.lang.Thread.State: WAITING (parking)
       at sun.misc.Unsafe.park(Native Method)
       - parking to wait for  <0x00000000833745a8> (a
java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject)
       at java.util.concurrent.locks.LockSupport.park(Unknown Source)
       at
java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject.await(
Unknown Source)
       at java.util.concurrent.DelayQueue.take(Unknown Source)
       at javax.swing.TimerQueue.run(Unknown Source)
       at java.lang.Thread.run(Unknown Source)

  Locked ownable synchronizers:
       - <0x00000000833745f0> (a
java.util.concurrent.locks.ReentrantLock$NonfairSync)

"AWT-EventQueue-0" #18 prio=6 os_prio=0 tid=0x000000001c003000 nid=0x99c
waiting for monitor entry [0x000000001fb1d000]
  java.lang.Thread.State: BLOCKED (on object monitor)
       at jmri.InstanceManager.getInstances(InstanceManager.java:765)
       - waiting to lock <0x0000000082e6af48> (a java.lang.Class for
jmri.UserPreferencesManager)
       at jmri.InstanceManager.getInstance(InstanceManager.java:256)
       at jmri.InstanceManager.getNullableDefault(InstanceManager.java:230)
       at jmri.InstanceManager.getDefault(InstanceManager.java:201)
       at
jmri.managers.configurexml.DefaultUserMessagePreferencesXml.load(DefaultUser
MessagePreferencesXml.java:50)
       at
jmri.configurexml.ConfigXmlManager.loadOnSwingThread(ConfigXmlManager.java:6
99)
       at
jmri.configurexml.ConfigXmlManager.lambda$1(ConfigXmlManager.java:610)
       at jmri.configurexml.ConfigXmlManager$$Lambda$48/168366.run(Unknown
[... 236 lines cut ...]
"GC task thread#1 (ParallelGC)" os_prio=0 tid=0x0000000004fca000 nid=0x3bd0
runnable 

"GC task thread#2 (ParallelGC)" os_prio=0 tid=0x0000000004fcb800 nid=0x13f0
runnable 

"GC task thread#3 (ParallelGC)" os_prio=0 tid=0x0000000004fce000 nid=0x3a18
runnable 

"GC task thread#4 (ParallelGC)" os_prio=0 tid=0x0000000004fd0800 nid=0x20f4
runnable 

"GC task thread#5 (ParallelGC)" os_prio=0 tid=0x0000000004fd1800 nid=0xd98
runnable 

"GC task thread#6 (ParallelGC)" os_prio=0 tid=0x0000000004fd4800 nid=0x3b2c
runnable 

"GC task thread#7 (ParallelGC)" os_prio=0 tid=0x0000000004fd6000 nid=0x2edc
runnable 

"VM Periodic Task Thread" os_prio=2 tid=0x000000001bf14800 nid=0x1d2c
waiting on condition 

JNI global references: 907

KenC57, 2018-06-04:
Built using bob's branch, testing with an NCE simulator, here is the thread dump. It is after the selection of the profile. Still seeing hang.

2018-06-04 10:46:36
Full thread dump Java HotSpot(TM) 64-Bit Server VM (25.171-b11 mixed mode):

"Swing-Shell" #24 daemon prio=5 os_prio=0 tid=0x000000001830d000 nid=0x3440 waiting on condition [0x000000001c0ce000]
   java.lang.Thread.State: WAITING (parking)
        at sun.misc.Unsafe.park(Native Method)
        - parking to wait for  <0x00000000d802cd50> (a java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject)
        at java.util.concurrent.locks.LockSupport.park(Unknown Source)
        at java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject.await(Unknown Source)
        at java.util.concurrent.LinkedBlockingQueue.take(Unknown Source)
        at java.util.concurrent.ThreadPoolExecutor.getTask(Unknown Source)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
        at sun.awt.shell.Win32ShellFolderManager2$ComInvoker$3.run(Unknown Source)
        at java.lang.Thread.run(Unknown Source)

   Locked ownable synchronizers:
        - None

"TimerQueue" #20 daemon prio=5 os_prio=0 tid=0x00000000187ae800 nid=0x1f94 waiting on condition [0x000000001becf000]
   java.lang.Thread.State: WAITING (parking)
        at sun.misc.Unsafe.park(Native Method)
        - parking to wait for  <0x00000000d803cce0> (a java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject)
        at java.util.concurrent.locks.LockSupport.park(Unknown Source)
        at java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject.await(Unknown Source)
        at java.util.concurrent.DelayQueue.take(Unknown Source)
        at javax.swing.TimerQueue.run(Unknown Source)
        at java.lang.Thread.run(Unknown Source)

   Locked ownable synchronizers:
        - <0x00000000d803cd70> (a java.util.concurrent.locks.ReentrantLock$NonfairSync)

"AWT-EventQueue-0" #18 prio=6 os_prio=0 tid=0x00000000188cf800 nid=0x49e0 waiting for monitor entry [0x0000000019bbd000]
   java.lang.Thread.State: BLOCKED (on object monitor)
        at jmri.InstanceManager.getInstances(InstanceManager.java:765)
        - waiting to lock <0x00000000d802d490> (a java.lang.Class for jmri.UserPreferencesManager)
        at jmri.InstanceManager.getInstance(InstanceManager.java:256)
        at jmri.InstanceManager.getNullableDefault(InstanceManager.java:230)
        at jmri.InstanceManager.getDefault(InstanceManager.java:201)
        at jmri.managers.configurexml.DefaultUserMessagePreferencesXml.load(DefaultUserMessagePreferencesXml.java:50)
        at jmri.configurexml.ConfigXmlManager.loadOnSwingThread(ConfigXmlManager.java:699)
        at jmri.configurexml.ConfigXmlManager.lambda$load$1(ConfigXmlManager.java:610)
        at jmri.configurexml.ConfigXmlManager$$Lambda$44/2109874862.run(Unknown Source)
        at java.awt.event.InvocationEvent.dispatch(Unknown Source)
        at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
        at java.awt.EventQueue.access$500(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.awt.EventQueue.dispatchEvent(Unknown Source)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.run(Unknown Source)

   Locked ownable synchronizers:
        - None

"AWT-Windows" #13 daemon prio=6 os_prio=0 tid=0x00000000184c8000 nid=0x34dc runnable [0x0000000018dff000]
   java.lang.Thread.State: RUNNABLE
        at sun.awt.windows.WToolkit.eventLoop(Native Method)
        at sun.awt.windows.WToolkit.run(Unknown Source)
        at java.lang.Thread.run(Unknown Source)

   Locked ownable synchronizers:
[... 114 lines cut ...]

   Locked ownable synchronizers:
        - None

"VM Thread" os_prio=2 tid=0x0000000015339800 nid=0x2034 runnable 

"GC task thread#0 (ParallelGC)" os_prio=0 tid=0x000000000353a000 nid=0x2620 runnable 

"GC task thread#1 (ParallelGC)" os_prio=0 tid=0x000000000353b800 nid=0x4b48 runnable 

"GC task thread#2 (ParallelGC)" os_prio=0 tid=0x000000000353d000 nid=0xe5c runnable 

"GC task thread#3 (ParallelGC)" os_prio=0 tid=0x0000000003540800 nid=0x3cdc runnable 

"GC task thread#4 (ParallelGC)" os_prio=0 tid=0x0000000003542000 nid=0x3b58 runnable 

"GC task thread#5 (ParallelGC)" os_prio=0 tid=0x0000000003543000 nid=0x4ba4 runnable 

"GC task thread#6 (ParallelGC)" os_prio=0 tid=0x0000000003546000 nid=0x36d8 runnable 

"GC task thread#7 (ParallelGC)" os_prio=0 tid=0x0000000003547800 nid=0x4c10 runnable 

"VM Periodic Task Thread" os_prio=2 tid=0x00000000172ff000 nid=0x3c5c waiting on condition 

JNI global references: 852

KenC57, 2018-06-04:
Here is the log, for whatever it might be worth.

     [java] 2018-06-04 10:39:49,120 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
     [java] 2018-06-04 10:39:49,377 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\kcameron\JMRI\log\messages.log [main]
     [java] 2018-06-04 10:39:49,380 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\kcameron\JMRI\log\session.log [main]
     [java] 2018-06-04 10:39:49,405 apps.Apps                             INFO  - PanelPro version 4.11.7ish+kcameron+20180604T1439Z+Recaa0baaac starts under Java 1.8.0_171 on Windows 10 amd64 v10.0 at Mon Jun 04 10:39:49 EDT 2018 [main]
     [java] 2018-06-04 10:40:16,700 apps.Apps                             INFO  - Starting with profile PanelProNCESim.3f703698 [main]
     [java] 2018-06-04 10:40:18,003 node.NodeIdentity                     INFO  - Using jmri-jlmXlbB7bhNiaaGkAi0s-H-3f703698 as the JMRI Node identity [main]

bobjacobsen, 2018-06-04:
Thanks! Much appreciated.  That's a different deadlock, I think.  Unfortunately, there may be several issues here that have to be uncovered one-at-a-time.  Will look into this one shortly.

KenC57, 2018-06-04:
I'll keep my copy ready to pull if you find other things to try.

bobjacobsen, 2018-06-04:
Could you try it again with this line in default.lcf please?
```
log4j.category.jmri.managers.JmriUserPreferencesManager = TRACE
```

Thanks.

bobjacobsen, 2018-06-04:
On 2nd thought, let me add two more:
```
log4j.category.jmri.managers.JmriUserPreferencesManager = TRACE
log4j.category.apps.Apps = TRACE
log4j.category.apps.AppsBase = TRACE
```

KenC57, 2018-06-04:
Added the others, but this is all the console showed:

     [java] 2018-06-04 11:35:48,157 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
     [java] 2018-06-04 11:35:48,236 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\kcameron\JMRI\log\messages.log [main]
     [java] 2018-06-04 11:35:48,240 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\kcameron\JMRI\log\session.log [main]
     [java] 2018-06-04 11:35:48,262 apps.Apps                             INFO  - PanelPro version 4.11.7ish+kcameron+20180604T1535Z+Recaa0baaac starts under Java 1.8.0_171 on Windows 10 amd64 v10.0 at Mon Jun 04 11:35:48 EDT 2018 [main]
     [java] 2018-06-04 11:35:48,271 apps.Apps                             DEBUG - Config file was specified as: PanelProConfig2.xml [main]
     [java] 2018-06-04 11:35:48,430 apps.Apps                             DEBUG - Prepare font lists... [PrepareFontListsThread]
     [java] 2018-06-04 11:35:51,970 apps.Apps                             INFO  - Starting with profile PanelProNCESim.3f703698 [main]
     [java] 2018-06-04 11:35:52,988 node.NodeIdentity                     INFO  - Using jmri-jlmXlbB7bhNiaaGkAi0s-H-3f703698 as the JMRI Node identity [main]
     [java] 2018-06-04 11:35:52,992 managers.JmriUserPreferencesManager   DEBUG - start load user pref file: C:\Users\kcameron\JMRI\PanelProNCESim\UserPrefsProfileConfig.xml [main]

NOTE: I've found that removing the preference file lets a normal start happen. So it is something about loading preferences.

bobjacobsen, 2018-06-04:
Thanks.  That helps.  Yes, it's a problem with loading (some of) the user preferences that's causing the conflict.  Could you attach your current UserPrefsProfileConfig.xml file?  Thanks.

KenC57, 2018-06-04:
Here is the file.

[UserPrefsProfileConfig - Copy.txt](https://github.com/JMRI/JMRI/files/2069065/UserPrefsProfileConfig.-.Copy.txt)

bobjacobsen, 2018-06-04:
Thanks!

Just pushed some changes to improve logging.  Could you pull those and try again with 

```
log4j.category.jmri.managers.JmriUserPreferencesManager = TRACE
log4j.category.apps.Apps = TRACE
log4j.category.apps.AppsBase = TRACE
log4j.category.jmri.configurexml.ConfigXmlManager = TRACE
```

Thank you.  We're getting there...

KenC57, 2018-06-04:
With the changes, still hang. Here is the console and thread dump.
[session.log](https://github.com/JMRI/JMRI/files/2069323/session.log)

[threaddump-1528133400836.txt](https://github.com/JMRI/JMRI/files/2069349/threaddump-1528133400836.txt)

KenC57, 2018-06-05:
I pulled your next change down. No deadlock, but things like the tables won't open. I get the empty window (if I click it in the Windows list). All time for tonight.

bobjacobsen, 2018-06-05:
Did the earlier sets of changes in this PR have the same problem?

KenC57, 2018-06-05:
Here is a threaddump of it while I'm waiting for the exit to happen.

[threaddump-1528202137391.txt](https://github.com/JMRI/JMRI/files/2072363/threaddump-1528202137391.txt)

KenC57, 2018-06-05:
I don't think so, but I didn't try the table every single time. So I'm not
sure where in the updates this might have changed. The fail to exit, that
would be new on the last set of changes (I think) but I had loaded a
different panel file this time and that might somehow tie into the exit
hang.

KenC57, 2018-06-05:
Bob,
Got a handy git command or two so I can step back and find where the exit and tables got their hang?
BTW, it still hasn't exited after a couple of hours.

bobjacobsen, 2018-06-05:
Thanks for all your patience with sorting this out.  It's a tricky one.

You can get any specific commit from git with a checkout command that references the hash for the commit.  For example

```
git checkout eb6452b
```

will check out the first commit in the branch that might have fixed it.  You can see the full list here:  https://github.com/JMRI/JMRI/pull/5427/commits  (the numbers on the right)

The "git bisect" command (See http://jmri.sourceforge.net/help/en/html/doc/Technical/GitFAQ.shtml ) can systematically work through the commits between a good and bad version, but for something this short it probably easier to do it manually.

Thanks for all your patience with sorting this out.  It's a tricky one.

bobjacobsen, 2018-06-05:
> Here is a threaddump of it while I'm waiting for the exit to happen.

Just got a chance to look at that.  it’s not showing anything happening:  No locks, no waits that shouldn’t be there.

Was there anything in the log?  It looks like something crashed out, and it’s going to wait forever.

KenC57, 2018-06-05:
Here is the console log.
[session.log](https://github.com/JMRI/JMRI/files/2073359/session.log)

It may be a issue with the panel file, it was from a user, and I had picked at random. So it may be the original and I fixed in another version.

KenC57, 2018-06-05:
I split the distance, so picked 6102184 for first try. Doing 'ant realclean' between checkouts. The exit was working until one of the last updates I think. And simple start then exit does work here. But trying to open tables, does not populate, it has been 40 minutes and still blank window. Will try stepping back again after table does populate.

KenC57, 2018-06-05:
After an hour, the table populated. Will try eb6452b next.

KenC57, 2018-06-05:
How can I tell if doing the git checkout really worked? As I'm rolling back, I'm questioning if the code really is changing.

KenC57, 2018-06-05:
Something isn't right. I've rolled all the way back to ecaa0ba and get the same results. As I recall the first couple in the list didn't work. But I'm getting the main menu, but then tables doesn't work and exit hangs.

Were there other commands with git I should be doing besides the checkout?

dheap, 2018-06-05:
git log [-1|-10]

No arguments display commit history with less. Others show last or last 10.

Sent from my iPad
-- 
Dave

> On 6 Jun 2018, at 6:35 AM, Ken Cameron <notifications@github.com> wrote:
> 
> How can I tell if doing the git checkout really worked? As I'm rolling back, I'm questioning if the code really is changing.

bobjacobsen, 2018-06-06:
The following is what it looks like when I jump around, starting at the head of the branch (the "%" is my command line prompt). If you're seeing these messages, git is doing the right thing.

```
% git checkout ensure-preferences-creation-order
Switched to branch 'ensure-preferences-creation-order'
Your branch is up to date with 'origin/ensure-preferences-creation-order'.

% git checkout ecaa0ba
Note: checking out 'ecaa0ba'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at ecaa0baaac... ensure UserPreferencesManager is in InstanceManager to prevent deadlock

% git checkout e04adfe
Previous HEAD position was ecaa0baaac... ensure UserPreferencesManager is in InstanceManager to prevent deadlock
HEAD is now at e04adfea26... better tests
```

KenC57, 2018-06-06:
Ok, I'm seeing similar results. Now to go and look at a couple of files to see if it proves the version.

KenC57, 2018-06-06:
Looks right in the files that I am at the first point. As I recall, that didn't work to startup. But now it is working. I checked the timestamp, the UserPref file is unchanged. So I'm not sure what is happening. But the profile I was using for testing is now working. Luckly I've got lots of profiles and found another that is failing to start with ecaa0ba.

KenC57, 2018-06-06:
With respect to the tables and exit, I don't see the line in the build.xml (I'm launching with ant panelpro) that set the option to fix the joal issue with startup. Or was there a different fix in the end?

bobjacobsen, 2018-06-06:
I don't understand the current status.  Is there one problem here, or two, or more?

PR #5427  is from a very recent HEAD of master, so any fix in 4.11.6 or before should be present; if not, that's  a regression to track down and fix.

klk32003, 2018-06-06:
> With respect to the tables and exit, I don't see the line in the build.xml (I'm launching with ant panelpro) that set the option to fix the joal issue with startup. Or was there a different fix in the end?

@KenC57  I'm also launching with ant panelpro. In this case the Java joal option is to be set in ```local.properties``` (syntax: jvm.args=-Djogamp.gluegen.UseTempJarCache=false).

KenC57, 2018-06-06:
When I was rolling back, it seems something in that profile had changed and
the deadlock was not happening with any of the versions. But another profile
is hanging nicely with the first version, now to move forward.

But the second issue is the tables vs audio issue, once it did startup.
Looking at the build.xml, I didn't see the line that made the setting that
worked around the joal issue.

KenC57, 2018-06-07:
After a suggestion from klk32003  I creatied a local.properties file, and loading it with:
jvm.args=-Djogamp.gluegen.UseTempJarCache=false

The tables work fine. At this point I think the exit was tied to the tables hang, which was related to the above.

At this point I think the start hang seems fixed. I will try some more profiles using the latest from this branch.

KenC57, 2018-06-07:
It now seems that after a profile is started with this version, older (4.11.6) starts up fine and does not hang. So it 'fixes' the profile. I don't know if it is worth figuring out what it changes.

dheap, 2018-06-07:
I prefer working with a branch (as per @bobjacobsen below) rather than a detached HEAD.

I also frequently use "git log -1" to keep an eye where I am.

I also now am comfortable with "git bisect" and it's nuances. Great tool for finding what broke something.

On 6 Jun 2018, at 10:39 PM, Bob Jacobsen <notifications@github.com> wrote:
> 
> If you want to create a new branch to retain commits you create, you may do so (now or later) by using -b with the checkout command again. Example:
> git checkout -b <new-branch-name>
> 
>  git checkout e04adfe

PR #5427 (2018-06-07): Ensure UserPreferencesManager is in InstanceManager to prevent deadlock
Meant to resolve #5426.
closed 2018-06-07 by bobjacobsen