JMRI/JMRI #5433: Transits > Sections > Edit Action > Hold / Release Signal : Swap over text

icklesteve, 2018-06-04:
Hi Folks,

Transits > Edit Transit > Sections in Transit > View / Add / Edit Actions > Add New Action > What > Hold ( or Release ) Signal

Select one of Signal Head ( Signal MAST list appears ) or Mast ( Signal HEAD list appears )

[image]

I use Signal Mast Logic, they're all listed in the left hand select, with just 1 Signal Head defined, displayed in the right hand select.

The actions do work as expected on the selected signal mast, so I think it's just a case of swapping the text over in the dialogue box?

Thanks,
Steve.

silverailscolo, 2018-06-05:
I see, you mean the labels displayed in front of the drop downs should be swapped? It seems that the right hand drop down initially should be empty. Is that item already selected when you first open this pane?

Egbert

Courier-Online, 2018-06-05:
Hi Egbert, 

Correct, 
I think the labels need swapping, 
I created one signal head purely for testing this issue, select box on right was not populated and was blank previously, no issue with the state of the select dropdowns, just the labels :-)

Steve

icklesteve, 2018-06-05:
Hi Egbert,

Yes, just the text labels "Signal Head : " and " or Signal Mast" need changing.

No issues with the select states, I just created a signal head for testing this issue,

Thanks, Steve.

silverailscolo, 2018-06-05:
Fix ready:
[image]

icklesteve, 2018-06-11:
Many Thanks @silverailscolo , 
Just tested on Jenkins release, looking good :-)

Steve

closed 2018-06-11 by icklesteve