JMRI/JMRI #5440: LE Dispatcher tables Section Name Display

icklesteve, 2018-06-06:
Hi Folks,

Sections were setup with automatic system names.
I'm now struggling to see which section is which in the dispatching tables, eg

[image]

I really don't want to add new sections with more descriptive system names as I have a few transits now setup.

Would it be possible to use the section username in these + the other dispatching tables?
If no username then display system name?

or

New option in dispatcher to select either username or internal name in the tables?

Personally I don't like adding more settings, the 1st approach seems more user intuitive?

Thanks,
Steve

mstevetodd, 2018-06-06:
The current format is SystemName(UserName), with (UserName) omitted if they are the same. See example from my layout. Your system names are much longer than mine.
[image]
Perhaps UserName(SystemName), suppressing if the same, would work for both of us?

icklesteve, 2018-06-06:
Hi @mstevetodd that'd work for me :-)

Have not modified system names, just checked the auto name box on creation,
Steve

Pugwash1, 2018-06-19:
So, this is what I've done. 
[image]

icklesteve, 2018-06-19:
Wowzers, that looks like a major improvement @Pugwash1  :-)
Above and beyond what would fix my section names issue, fair play for your time taken on this.
Looking forward to when this is added to a Jenkins release so I can confirm resolved.

Pugwash1, 2018-07-10:
[build](http://jmri.tagadab.com/jenkins/job/Development/job/Packages/) 2573 and beyond. As they are dev builds your mileage may vary!!!!

icklesteve, 2018-07-26:
Many thanks @Pugwash1 working really well.
Not sure if local issue but Signal Name not being displayed in the active table ( just the Signal ID ) using SML, have asked JMRI Users if it's just me!

But issue closed, I can definitely read the section names, the extras were pure bonus.

closed 2018-07-26 by icklesteve

Pugwash1, 2018-07-26:
Hi Steve
Yep, noted, I will fix.
Steve G

On July 26, 2018 5:30:17 PM EDT, steve young <notifications@github.com> wrote:
>Many thanks @Pugwash1 working really well.
>Not sure if local issue but Signal Name not being displayed in the
>active table ( just the Signal ID ) using SML, have asked JMRI Users if
>it's just me!
>
>But issue closed, I can definitely read the section names, the extras
>were pure bonus.

-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.