JMRI/JMRI #5443: Duplicate LocoNet2 menu created when autostarting LocoNetOverTCPServer

mstevetodd, 2018-06-07:
When I auto-start LocoNetOverTCPServer, a duplicate LocoNet2 menu is created on the main PanelPro dialog. This menu has identical options as the LocoNet menu. 
It does not appear if I take the "Start LocoNetOverTCP Server menu option instead of auto-start.
I tested using a new profile with no other auto-start actions, on Windows 10, Raspbian and Ubuntu.
This occurs at 4.11.6 and current Jenkins build, but does not occur at 4.10.

devel-bobm, 2018-06-07:
@mstevetodd, 

I am guessing that this directly stems from code changes related to commit by @silverailscolo in April of this year, which (in part?) added `LocoNetSystemConnectionMemo` as a parameter to the `LnNetworkPortController` and to those classes which extend that class.  This includes/requires a similar change to the `LnTcpDriverAdapter`, which is the heart of the LocoNet over TCP server:

`public LnTcpDriverAdapter(LocoNetSystemConnectionMemo m) {`

The c'tor was parameterless before this commit.

Since the "start-up" actions have no knowledge of any existing LocoNet connection memo, the start-up action cannot pass one in.

For completeness, the commit in question is at [https://github.com/JMRI/JMRI/commit/27dc8de04ccc763a5939ddf427cf71abe81e847e](url) .

I do not understand the reason for the changes in the above commit, so I cannot recommend any code changes to resolve the problem.

pabender, 2018-06-08:
even though it's bedtime, I dug around...  The action needs to implement:

https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrix/swing/SystemConnectionAction.java

silverailscolo, 2018-06-09:
Thanks for the pointer, Paul
I will try to add that in the coming days.

Egbert

> Op 8 jun. 2018, om 07:32 heeft Paul Bender <notifications@github.com> het volgende geschreven:
> 
> even though it's bedtime, I dug around... The action needs to implement:
> 
> https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrix/swing/SystemConnectionAction.java <https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrix/swing/SystemConnectionAction.java>

silverailscolo, 2018-06-09:
What I understand is that this LnOverTcp server should connect to an existing menu, meaning an existing Memo instead of creating a new one (plus its menu). As long as you don't autostart it and start it from the menu of choice, that works again: see branch [eb-loconet](https://github.com/silverailscolo/JMRI/blob/eb-loconet/java/src/jmri/jmrix/loconet/loconetovertcp/LnTcpDriverAdapter.java#L32).
But the startup sequence neither is told what connection to connect to, nor is that connection up and running by the time the action is called on startup. At least, when I manually edit the startup action prefix, it can't find that connection at start-up and says: ``"Connection "L" does not exist and cannot be assigned to action ...`` and creates a new one.
I think that in apps.startup.AbstractActionModel you somehow need to call ``setSystemPrefix("L")`` and use it to get the correct memo in the action.

devel-bobm, 2018-06-09:
@silverailscolo, On my machine, it seems like the LocoNet (simulator) connection is alive by the time the start-up action is triggered.  I suspect this because I started the LocoNet Monitor with a start-up action before the LocoNetOverTcpServer start-up action, and it didn't complain and there wasn't any error or warning in the console log.

I did some experimenting the other day with the mechanism @pabender recommended and quickly got swamped with complexity and "squeezing the balloon" syndrome.  I had to make more stuff public and add more features to somewhat related LocoNet-generic functionality and even some connection-type-specific functionality, just to get things to build.  And I wasn't making any real progress on the original problem, so I gave up that particular branch as a "lost cause".

devel-bobm, 2018-06-09:
I would think that an -easy-, but not necessarily "correct", answer to this might be to create a second c'tor for `loconet.loconetovertcp.LnTcpDriverAdapter`.  

That c'tor would not have a `LocoNetSystemConnectionMemo` parameter.  It would perform the necessary work to "figure out" which LocoNet connetion to attatch to, then get it's Memo, and attach to it.  This would get past the original problem posted by the user.

bobjacobsen, 2018-06-09:
There's some discussion of how the system connection pieces all fit together here: http://jmri.org/help/en/html/doc/Technical/SystemStructure.shtml  with some additional info here:  http://jmri.org/help/en/html/doc/Technical/MultiConnectionUpdate.shtml

bobjacobsen, 2018-06-09:
> I think that in apps.startup.AbstractActionModel you somehow need to call setSystemPrefix("L") and use it to get the correct memo in the action.

Please don’t do that.  The memo needs to be passed in at the time the menu item is created.  Please don’t try to use something else at runtime to figure it out.

silverailscolo, 2018-06-10:
@bobjacobsen This issue is with the ``LnTcpServer`` as an auto start-up action in prefs in 10.11.7ish. 

Having the LnOverTcpServer implement ``SystemConnectionAction.java``, as suggested by @pabender should allow us to ask for a prefix and figure out the memo from that. But currently the startup action on save does not store the prefix in the profile. But should it?

To understand the Server concept, let's look at a multi-connection example:
[image]
I have a setup with 2 connections (and ditto menus): LocoNet Simulator connection (LocoNet1, prefix L1) and LocoNet Simulator connection (LocoNet2, prefix L2). In JMRI prefs on the "LocoNet over TCP" tab I select to start up (and save):
[image]
After restart, there's a 3rd LocoNet menu and the LnTcpServer is bound to that:
[image]
Question now is: would you expect a different LnServer instance to start, depending on the connection menu from which you select "Start LocoNetOverTCP Server"? Or is it connection-less?
And if that is the intention, for what connection am I setting the _single_ Prefs tab port number?
And when my Digitrax connection adapter is set to "LocoNet Over TCP", would that make any difference? Because all LnOverTcpServer implementations use the same ``LnTcpDriverAdapter``.

In my branch the menu item was updated to use a specific memo by including it as a param:
``        add(new jmri.jmrix.loconet.loconetovertcp.LnTcpServerAction(Bundle.getMessage("MenuItemLocoNetOverTCPServer"), memo));
``
That's useful for the TCP (hardware) connection, but might force us to handle the (abstract) LnOverTcpServer in another way. Any creative thoughts?

bobjacobsen, 2018-06-10:
I think maybe it's time to make this work like (almost) every other action in the program.  Instead of the special "LocoNet over TCP" preference pane having a "Start automatically with application" check box (and perhaps eventually multiple ones for multiple connections), it should work like e.g. the LocoNet Monitor or LocoNet Slot Monitor or really anything else:

- You go to Start Up in the preferences and select the "Perform Action.." (which also allows selection of when, etc)

- As part of that, you're asked to specify the connection:

[image]

- That automatically gets stored in the "profile/profile.xml" file:

```
    <startup xmlns="http://jmri.org/xml/schema/auxiliary-configuration/startup-4-3-5.xsd">
        <perform xmlns="" class="apps.configurexml.PerformActionModelXml" name="jmri.jmrit.beantable.SignalMastTableAction" type="Action">
            <property name="systemPrefix" value=""/>
        </perform>
        <perform xmlns="" class="apps.configurexml.PerformActionModelXml" name="jmri.jmrix.loconet.locomon.LocoMonPane$Default" type="Action">
            <property name="systemPrefix" value="L2"/>
        </perform>
    </startup>
```

That ends up simpler for both the user (who doesn't need to know how to do special things in a special way) and in the code (ditto).

This leaves another problem, which is that you can't start two LocoNetOverTCP servers on two separate connections that use the same port.  (There might be other issues, too, as there are multiple static items in the jmrix.loconet.loconetovertcp package)  But that seems to be a question for another day.

silverailscolo, 2018-06-10:
Great, certainly the way to go. It occurred to me we had to get the connection (prefix) stored somehow. This looks easy, but still has to be created.
Sorry I can’t spend more time on this for the coming 3 weeks. Will save a few my included GUI tweaks for later.

Egbert

> Op 10 jun. 2018, om 17:13 heeft Bob Jacobsen <notifications@github.com> het volgende geschreven:
> 
> I think maybe it's time to make this work like (almost) every other action in the program. Instead of the special "LocoNet over TCP" preference pane having a "Start automatically with application" check box (and perhaps eventually multiple ones for multiple connections), it should work like e.g. the LocoNet Monitor or LocoNet Slot Monitor or really anything else:
> 
> You go to Start Up in the preferences and select the "Perform Action.." (which also allows selection of when, etc)
> 
> As part of that, you're asked to specify the connection:
> 
>  <https://user-images.githubusercontent.com/2774117/41202936-561990ce-6c85-11e8-8ada-1072d084a3ea.png>
> That automatically gets stored in the "profile/profile.xml" file:
>     <startup xmlns="http://jmri.org/xml/schema/auxiliary-configuration/startup-4-3-5.xsd">
>         <perform xmlns="" class="apps.configurexml.PerformActionModelXml" name="jmri.jmrit.beantable.SignalMastTableAction" type="Action">
>             <property name="systemPrefix" value=""/>
>         </perform>
>         <perform xmlns="" class="apps.configurexml.PerformActionModelXml" name="jmri.jmrix.loconet.locomon.LocoMonPane$Default" type="Action">
>             <property name="systemPrefix" value="L2"/>
>         </perform>
>     </startup>
> That ends up simpler for both the user (who doesn't need to know how to do special things in a special way) and in the code (ditto).
> 
> This leaves another problem, which is that you can't start two LocoNetOverTCP servers on two separate connections that use the same port. (There might be other issues, too, as there are multiple static items in the jmrix.loconet.loconetovertcp package) But that seems to be a question for another day.
>

silverailscolo, 2018-07-22:
Yes, original issue is fixed. Will create a new one for the additional config raised.

closed 2018-07-22 by mstevetodd