JMRI/JMRI #5447: Layout Editor - new track styles need to display backwards compatible

pnbgit, 2018-06-08:
The new track style displays in LE are great, and I have been making use of them.

However, I think it is important that when loading existing panels, they should display ok using the settings from the panel file.

There are several problems which I shall create as separate issues.  They generally manifest when the track is set to a higher width, for display on a large format display on the railway room wall, for example.

The first problem is that the track segment selection circles have become very small, making it difficult to click on them.

Here is an example, using a track width setting of 4.  Here it is in 4.10:

[image]

And here it is in 4.11.6, as you can see, whilst the turnout circles are still a good size, the track segment selection circles are much smaller:

[image]

The problem gets worse with wider tracks.  In this example, which is used on a wide screen wall mounted TV, I have used a mainline track width of 10, which gives a good display.

Here it is in 4.10, with good size selection circles:

[image]

And here it is in 411.6.  The track segment selection circles have now disappeared completely within the track width, so it is impossible to see where to right click for the pop-up menu:

[image]

Hopefully this is an easy fix, just to increase the circles back to the size they used to be.

Thanks,

Pete

dsand47, 2018-06-09:
This is easy to fix by reverting the track segment selection circle to the turnout circle size.

Dave Sand

dsand47, 2018-06-10:
PR #5461 has been created to use the turnout circle size instead of a fixed size.

pnbgit, 2018-06-11:
Thanks Dave.

pnbgit, 2018-06-12:
I can confirm this is fixed in 4.11.7 #1, looks good and nice that you can now adjust the track segment circle size with the turnout circle size.  I will close this issue now.

closed 2018-06-12 by pnbgit