JMRI/JMRI #5448: Layout Editor: new track styles need to display backwards compatible

pnbgit, 2018-06-08:
With the new track styles, when an existing panel from 4.10 or earlier is loaded, any track not within a block displays very narrow and pale gray, with side track only having a width of 1.  When using wide track settings, it can look as those the tracks not in a block have disappeared.

I think it is important that users existing panel files should display the same, as far as possible, as they did before.  (see also #5447) There are a very great many users out there who use JMRI just as a "glass control panel" to operate signals, turnouts etc. and do not use any of the features based on blocks and block routing. These users do not know anything about putting track segments into blocks and should have no need to do so.

Here is an example, with mainline track width set to 10 and side track width set to 5.

Here is the screenshot from 4.10.  The siding displays correctly as side track with width 5:

[image]

But in 4.11.6, the siding shows very thin, and in pale gray, almost invisible:

[image]

This is because the siding is not in a block (and has no need to be).

The problem arises because the new track drawing options set the block line width correctly to the track line width set in the loaded panel, which means that track in a block draws correctly, but then sets the rail widths to minimums of 2 for mainline and 1 for side track, and the rail colour to pale gray, which is what is used to draw track that is not in a block.

Could I suggest two options to get tracks that are not in a block to display the same as they did in 4.10 when loading a panel created in an earlier version?

1. Display tracks that are not in a block with the block line width 
     or
2. When reading in an old panel (predating the new track styles), set the initial values, for the track drawing options, so that both track width and block line width are the same as the old panel's track line width settings. Also set the rail colour the same as the block's track colour, or the default track colour for a segment not in a block (black if the user hasn't changed it). This should be separately applied to both main track and side track.

Option 2 seems to me to be favourite, when I make those changes I see no visible effect on my panel for segments in a block, but the segments not in a block now revert to how they looked in 4.10.  The user can they apply track drawing presets or customise the settings as they wish from there.

Thanks,

Pete

dsand47, 2018-06-13:
I consider the default rail sizes to be a feature.  It is obvious which track does not have a block assigned.  This is important for signaling, entry/exit and automation.

Due to the scope of the changes, it will be important to make the changes known.  This will need to be more than just the release note.  The current plan includes a separate posting to the user group along with the release 4.12 announcement and a summary document will posted to the files area.

dsand47, 2018-06-13:
There is no need to add blocks.  A one time change to rail width and rail color addresses the issue.

We don't know the relative numbers of people who will need fix the rail size versus number of people who will need to undo a default that masks unblocked track.

In one sense, the default narrow track width triggers people to **_think_** about how they want to display the track that does not have a block assigned.  Some people will use the old values, others may want to use a different size and/or color to indicate unblocked track so that the dispatcher knows not to assume that a train can be routed into that area.

bobjacobsen, 2018-06-13:
When was the underlying change made?  If it was a while ago, have we heard any user comments from people encountering it in the test releases?

dsand47, 2018-06-13:
The track changes were in 4.11.4.  Prior to that there were some comments about turnout position visibility from people using the Jenkins builds.

I made a lot of changes prior to 4.11.4 to clean up issues with hidden track, dashed track, turnout line drawing, etc.

I am not aware of any issues being raised with 4.11.4 or 4.11.5.

pnbgit, 2018-06-14:
This problem will affect mainly users who are using JMRI in a simple way, not the power users.  So the type of user most affected by this is also the type of user who is more likely to stick to release versions and not try out the test versions, and be far less able to work out what has gone wrong and how to fix it.

For me, the overriding factor is that we should maintain backwards compatibility and not break existing panels.

Whilst it may only need changing 2 numbers to fix any panels broken by this, the type of users that will be most affected will not know that, they will have no idea what has caused the problem, and no idea that there is a new display settings window and that they can change two numbers to fix the problem.

We have no need to force such users to think about how to display track without a block assigned, they are not using blocks at all so why should we force them to be concerned about it by breaking their existing panel displays?

The solution is so simple - if we change those 2 numbers in the default settings, then the new version will no longer break all those existing panel displays, and power users who are using blocks can change those numbers (or choose a preset as I suggested) so that unblocked track is displayed differently.

Surely not breaking existing panel displays should be our top priority here?

dsand47, 2018-06-14:
I give up.  PR #5481.

bobjacobsen, 2018-06-15:
I'm not an extensive LayoutEditor user, so still trying to get my head around this.

If this is a migration issue, can we treat it that way?  I.e.

- When a panel file is read that doesn't have settings for the items being discussed here, have the ConfigureXML reader code provide the _old_ defaults:  Old panel files will look the same
- Have the defaults for new panels (i.e. in newly-created icons) be the new values.

The outcomes are:
- Old panels stay the same
- New panels are created in the new form:  People will be reminded to set blocks, and if they're not using those, they can set options as they'd like the appearance to be. (Not sure I got that part right)

I'm probably missing something, though, as I don't really use these features. If so, what?

dsand47, 2018-06-15:
As of 4.11.4, Layout Editor Panel loading is a two step process.

1) **LayoutEditorXml** loads the panel and LayoutEditor sets up the initial **LayoutTrackDrawingOptions** using the track width to set the block width.  The **_rail_** width and colors are default values in **LayoutTrackDrawingOptions**.

2) **LayoutTrackDrawingOptionsXml** updates the **LayoutTrackDrawingOptions** that was created in step 1.

Pete's request is to change step 1 to use the track width and colors for **_rails_** as well as **blocks**.  Pre 4.11.4 files do not have a LayoutTrackDrawingOptions element and therefore do not perform step 2. They are left with the values from step 1.  

PR #5481 un-comments the code to set the rail width and colors.  These were commented earlier due to issues with the rail and block sizes being the same.  Subsequent changes appear to have resolved those issues.

There is no migration as such, just the lack of step 2.

New panels will be treated the same way since they effectively have just a step 1.  It should be possible to add a dialog to recommend a review of the track drawing options, or maybe even automatically open the track drawing options dialog.  Or do nothing.

pnbgit, 2018-06-16:
Thanks Dave - I am convinced this is the right decision, and will avoid having to deal with the support issues of users asking why their panels didn't display as they used to.

bobjacobsen, 2018-07-02:
Can we close this?

pnbgit, 2018-07-04:
Yes, it works correctly now on 4.11.8.

closed 2018-07-04 by pnbgit