JMRI/JMRI #5449: LE Decorations: End Bumper issues

pnbgit, 2018-06-08:
The new decorations feature is useful, but I have found  few issues with end bumpers.

I am creating each as a separate issue. See also: 

1. Naming of end bumper/buffer stop not consistent.

When you create an end bumper or buffer stop, it chooses the name based on the Java locale. So US users get an end bumper, and UK users get a buffer stop.

[image]

However, the right click menu for the new decorations always uses the term end bumper, so UK users get inconsistent terminology:

[image]

Would it be possible for the decorations menu to get the naming from the same place as the buffer stop/end bumper itself?

geowar1, 2018-07-30:
Just needed to add EndBumperMenuTitle and EndBumperMenuToolTip to the LayoutEditorBundle_en_GB.properties file.

geowar1, 2018-08-15:
If you can conform this as fixed then this can be closed.

pnbgit, 2018-09-16:
Just tested in 4.13.3, I can confirm it is fixed, words buffer stop now showing in decorations popup menu with UK localisation.

Just #5451 still outstanding of these 3 related issues.

I have now closed this issue.

closed 2018-09-16 by pnbgit