JMRI/JMRI #5455: Expose Car "Status" via JSON endpoints

mmillican, 2018-06-09:
I'm currently working on a mobile app using the JMRI APIs (roster/operations focused) and would like to get the car "status" exposed via the API. By status, I'm referring to "Out of service" and "Location unknown."

I believe this method is what would need to be updated for the `/json/cars` endpoint: https://github.com/JMRI/JMRI/blob/master/java/src/jmri/server/json/operations/JsonUtil.java#L95

I'm happy to submit a PR if interested / opening to adding it.

mstevetodd, 2018-06-09:
@mmillican , adding a new element to an existing json string should be a very low-impact change. Please submit a PR.

mmillican, 2018-06-12:
@mstevetodd thanks - will try to get to it this week. Any idea how long then it would take to get into a release?

mstevetodd, 2018-06-12:
@mmillican Once you create the PR and the automated testing completes, the PR should be reviewed and merged within a day or so. The three release levels and their scheduling is described here: http://jmri.org/download/index.shtml

mstevetodd, 2018-08-14:
@mmillican I have added car status to the json/car node. It is now in the latest Jenkins build, and will be in future test and production builds. You can see an example here: 
http://jmri.mstevetodd.com:1080/json/car/SSW46
Note that I simply added the output of Car.getStatus() to the node. 
Is that what you were looking for?

Note: I plan to add the expected values and their descriptions to the json schema, but I haven't figured out the best syntax for that, and I have no JMRI example to follow.
I also fixed the operations json lists to properly set up the list and individual change listeners.

mmillican, 2018-08-15:
@mstevetodd Great, this will do the trick I think! Thanks!

Sorry I never got around to the PR for this - had some other commitments come up.

closed 2018-08-15 by mstevetodd