JMRI/JMRI #5466: Routes missing IR silently lost on load

KenC57, 2018-06-11:
I have a user report and the issue was the panel file had simple numbers for route systemName fields. The file loads ok, and the XML check says ok too, but the table (or a save) shows no route entries.

The user is using 4.10, and I tested with 4.11.6 (where I got the results above). His reports had some NPE's but they were either common to 4.10 (like a Audio NPE) or something related to disposing of a pane, not sure if related here.

The example panel file is:

[SSLgood.xml.txt](https://github.com/JMRI/JMRI/files/2090949/SSLgood.xml.txt)

It would seem that it should have either failed the file or loaded the routes (and correct the system name on save). But not be silent while never loading them. I thought this issue has been found before. But I'm not finding an open case.

dsand47, 2018-06-12:
The failure is not entirely silent:

```2018-06-11 22:41:59,033 configurexml.DefaultRouteManagerXml   ERROR - failed to create Route: 50 [AWT-EventQueue-0]```

The route xml process catches an **IllegalArgumentException** which occurs when the bean with the numeric system name is added to the _beans TreeSet.  When this occurs, the xml process returns and ignores the remaining entries.

The solution is to manually add the IR prefix to each route system name.

bobjacobsen, 2018-07-02:
The history (bottom of file; this is why we keep that stuff!) shows that the file was read-with-errors (which were ignored) in Feb 2018. Before that, it was written last in June 2013 by JMRI 2.10 (not a typo).

Two things about migration:
 - if you have a file that’s shown errors when read, and you write it out, you can’t assume that all the errors are _fixed_. 

 - We only test backward about four major releases (two years) for file readability. Five years and 9 production releases is a long time, too long to imagine that everything can be cleanly processed. But, of course, it _did_ tell the user there was a problem, in several ways. 

One thing we should do is 
 - [ ] update the schema check so that files like these just don’t load at all, instead of loading with errors. That has the benefit of being unambiguous.

KenC57, 2018-07-02:
I hadn't noticed the history part. Wow that is way back.

Yes, making the schema check catch and explain is better. And the check is
much more obvious to the users than the system log.
-ken c

closed 2021-04-14 by stale[bot]