JMRI/JMRI #5480: PanelPro splash screen shrinking

bobjacobsen, 2018-06-14:
From Dick Bronson on jmri-developers:

> I have been frustrated for a time that the splash window often opens too short to fully display the [Help] and [Quit] buttons. I drag the bottom of the window down, and then exit, only to have it be to short again the next time it opens.
> 
> Someplace someone else also claimed the same issue and was told to simply resize the window and it would remember when saved. As an experiment I did that, and discovered that apparently there is a bug in the save because every time it is saved and then reopened again it is shorter than the last. I went a hunting and discovered that the data is saved in user-interface.xml. Running a diff shows the following changes for three consecutive closes.
> 
> <window xmlns="" class="apps.gui3.dp3.DecoderPro3Window:DecoderPro" height="440.0" locX="2085.0" locY="75.0" width="745.0">
> 
> <window xmlns="" class="apps.gui3.dp3.DecoderPro3Window:DecoderPro" height="413.0" locX="2085.0" locY="103.0" width="739.0">
> 
> <window xmlns="" class="apps.gui3.dp3.DecoderPro3Window:DecoderPro" height="386.0" locX="2085.0" locY="131.0" width="733.0">
> 
> It appears that the PanelPro splash screen loses 27 pixels height and 6 pixels width each time it is closed.
> 
> Version 4.11.7
> Ubuntu Linux 16.4

bobjacobsen, 2018-06-14:
From Dick Bronson on jmri-developers:

> I find it an interesting coincidence that the attached image is exactly 27 pixels high. This is the title bar portion from the PanelPro splash screen. Maybe Windows and Linux count the title part differently when reporting window size.

[image]

Pugwash1, 2018-06-19:
@bobjacobsen maybe in jmrijframe savewindowsize or getmaximumsize

danielb987, 2019-01-02:
PR  #6147 seems related to this.

danielb987, 2019-01-02:
With "related" I mean that the PR has the same kind of error.

closed 2021-04-14 by stale[bot]