JMRI/JMRI #5507: Install TrainCrew Error java.io.FileNotFoundException

icklesteve, 2018-06-30:
v 4.11.8

Problem with Win7 installs?
No problem found when installing TrainCrew on Raspbian ( where full file list copied was shown in console ), 

Debug > Install TrainCrew Application

2018-06-30 14:22:50,534 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]
2018-06-30 14:22:50,541 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\Steve\JMRI\dc\ [main]
2018-06-30 14:22:50,542 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\Steve\JMRI\dc\ [main]
2018-06-30 14:22:50,542 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\Steve\JMRI\ [main]
2018-06-30 14:22:50,543 util.FileUtilSupport                  INFO  - File path home: is C:\Users\Steve\ [main]
2018-06-30 14:22:50,543 util.FileUtilSupport                  INFO  - File path scripts: is C:\Users\Steve\JMRI\dc\ [main]
2018-06-30 14:22:58,163 PanelPro.PanelPro                     INFO  - Main initialization done [main]
2018-06-30 14:29:53,839 TrainCrew.InstallFromURL              INFO  - Will install from https://github.com/ekapus/TrainCrew/archive/master.zip   to web/TrainCrew [AWT-EventQueue-0]
2018-06-30 14:29:56,703 util.PhysicalLocation                 INFO  - Unzip file TrainCrew-master/ to C:\Program Files (x86)\JMRI\web\TrainCrew\TrainCrew-master [AWT-EventQueue-0]
2018-06-30 14:29:56,706 util.PhysicalLocation                 ERROR - Problem creating Folder [AWT-EventQueue-0]
2018-06-30 14:29:56,707 util.PhysicalLocation                 INFO  - Unzip file TrainCrew-master/LICENSE.txt to C:\Program Files (x86)\JMRI\web\TrainCrew\TrainCrew-master\LICENSE.txt [AWT-EventQueue-0]
2018-06-30 14:29:56,709 util.PhysicalLocation                 ERROR - Error unpacking zip file [AWT-EventQueue-0]
java.io.FileNotFoundException: web\TrainCrew\TrainCrew-master\LICENSE.txt (The system cannot find the path specified)
	at java.io.FileOutputStream.open0(Native Method)
	at java.io.FileOutputStream.open(Unknown Source)
	at java.io.FileOutputStream.<init>(Unknown Source)
	at java.io.FileOutputStream.<init>(Unknown Source)
	at jmri.util.UnzipFileClass.unzipFunction(UnzipFileClass.java:67)
	at apps.TrainCrew.InstallFromURL.actionPerformed(InstallFromURL.java:72)
	at javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
	at javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
	at javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
	at javax.swing.DefaultButtonModel.setPressed(Unknown Source)
	at javax.swing.AbstractButton.doClick(Unknown Source)
	at javax.swing.plaf.basic.BasicMenuItemUI.doClick(Unknown Source)
	at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(Unknown Source)
	at java.awt.Component.processMouseEvent(Unknown Source)
	at javax.swing.JComponent.processMouseEvent(Unknown Source)
	at java.awt.Component.processEvent(Unknown Source)
	at java.awt.Container.processEvent(Unknown Source)
	at java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$500(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)
2018-06-30 14:29:56,794 TrainCrew.InstallFromURL              INFO  - Complete! [AWT-EventQueue-0]

bobjacobsen, 2018-06-30:
```
2018-06-30 14:29:53,839 TrainCrew.InstallFromURL INFO - Will install from https://github.com/ekapus/TrainCrew/archive/master.zip to web/TrainCrew [AWT-EventQueue-0]
2018-06-30 14:29:56,703 util.PhysicalLocation INFO - Unzip file TrainCrew-master/ to C:\Program Files (x86)\JMRI\web\TrainCrew\TrainCrew-master [AWT-EventQueue-0]
```

The above looks OK. This next line is the core error:

```
2018-06-30 14:29:56,706 util.PhysicalLocation ERROR - Problem creating Folder [AWT-EventQueue-0]
```

It happens while trying to unzip the 1st file:
```
2018-06-30 14:29:56,707 util.PhysicalLocation INFO - Unzip file TrainCrew-master/LICENSE.txt to C:\Program Files (x86)\JMRI\web\TrainCrew\TrainCrew-master\LICENSE.txt [AWT-EventQueue-0]
2018-06-30 14:29:56,709 util.PhysicalLocation ERROR - Error unpacking zip file [AWT-EventQueue-0]
java.io.FileNotFoundException: web\TrainCrew\TrainCrew-master\LICENSE.txt (The system cannot find the path specified)
```

So something is going wrong when JMRI attempts to create the `C:\Program Files (x86)\JMRI\web\TrainCrew\TrainCrew-master` directory.  

Some questions to consider:

- Does Windows somehow protect against attempts to create something in the program directory? I.e. does JMRI need some special privileges to do that?

- Did the Traincrew directory get created OK?  Perhaps there's an issue with the 2nd level create?

#5384 talks about how the unpacking is configured.  You can perhaps unpack to a different location to see if that works.

KenC57, 2018-06-30:
Yes Windows needs to privilege session to install in the program space. Now
since this is being loaded/unzipped from the user session, it can't. If JMRI
had been run 'as admin' which is a right click function, I think it would
have worked.

Other option would be allowing it as a local user space addition to the
program, like how we read scripts and panels from the user space.
-ken c

icklesteve, 2018-06-30:
Thanks Both,

Running PanelPro with Admin Privileges enabled the install fine.
Have added a note to the JMRI TrainCrew install page which I'll upload near future,

Steve

closed 2018-06-30 by icklesteve