JMRI/JMRI #5517: WiThrottle Server to use Turnout.provide() instead of Turnout.get()

mstevetodd, 2018-07-04:
See discussion on JMRI Users Group: https://groups.io/g/jmriusers/topic/22999140

Currently, WiThrottle Server (intentionally) only operates turnouts that are already defined to JMRI. 

After this change, WiThrottle Server will create a turnout in JMRI when a valid system name is requested that is not known to JMRI. Plan is to use Turnout.provide() instead of Turnout.get().

Testing will need to verify that the "Allowed Controls" Preference is still respected.

pabender, 2018-07-07:
I have two minor concerns.

1. We have been trying very hard to move away from system names in the user interface where possible.  This change means that WiThrottle and EngineDriver will be making those more visible.

2. The system names need to be validated at some point in this process, and if that validation fails, the clients need to be notified of the reason why there is a failure ( this is part of the problem with the current implementation)

The second item is more important imo.

mstevetodd, 2018-07-22:
Added a new "Create Turnout" preference to the WiThrottle "Allowed Controls" section. It defaults to OFF, so as to match current behavior. Users that want this functionality will need to change the preference.
Even if enabled, the turnout will first be checked against the known list and the action prevented if user has marked this turnout as not allowed.
If checked, and not in the known list, provideTurnout() is used to attempt to create the turnout, and the error message returned to the client.

mstevetodd, 2018-07-22:
Note: the current turnout list-handling behavior is unchanged (per discussion with Brett).

closed 2018-08-03 by mstevetodd