JMRI/JMRI #5521: Group issue:  Control Panel (Add Item) Sort Fails

dsand47, 2018-07-05:
Reported by Phil Abernathy on 7/2.

Using JMRI 4.11.8 (LocoNet Simulator) with Java 1.8.0_171 on Win 10, I found that in my Control Panel, I was frustrated because I seemed to be consistantly adding the wrong sensor, even though I highlighted what I wanted then dragged the icon into the panel. But I realized that it (wrong sensor) only happened after I did a sort on the columns of sensors, and I found that the sensor added to the panel was always the sensor that had been in that position in the list before "sorting", not the sensor that was highlighted.

For example, in a small test panel with two user (internal) sensors (and ISCLOCKRUNNING), if I try to add a sensor called "StopListener", the three sensors show in the Add Items/Item Palette in System Name order with IS1, IS2 and ISCLOCKRUNNING in that order.  If I click on a line to highlight the sensor I want, then click on the column heading for System Name (or User Name), when the items are re-ordered, the highlighted item remains highlighted (in the new position). However, if I now drag the icon onto the panel, the new item placed on the panel is the sensor that had been in that position on the list, not the highlighted one.

---

If no one else is working on this problem, this issue can be assigned to me and I will finish up my probable fix.

Dave Sand

bobjacobsen, 2018-07-05:
@dsand47 Thanks!  Sounds to me like it would be good to get this in, if possible and it's not an extensive code change.

dsand47, 2018-07-05:
So far I have only tested two of the CPE Pallete tabs but it looks promising.
```
-        return _model.getBeanAt(row);
+        return _model.getBySystemName((String) _table.getValueAt(row, 0));
```
I should be able test the rest of them tomorrow.

dsand47, 2018-07-05:
PR #5525 contains the fix for this issue.

closed 2018-09-23 by bobjacobsen