JMRI/JMRI #5522: attempting to get speed for invalid name: 'Approach'

pnbgit, 2018-07-05:
This problem is encountered by users trying to use dispatcher with the BR-2003 signalling system.

It is caused by the aspects.xml file for BR-2003 having an invalid speed "approach" set for every signal aspect.

I have attached a file with correct signal speeds for all the aspects.

[aspects-BR2003.zip](https://github.com/JMRI/JMRI/files/2165352/aspects-BR2003.zip)

Could somebody check this and then submit it for the next test release?

Thanks,

Pete

icklesteve, 2018-07-05:
Thanks Pete, 
Was v annoying issue for UK users!

I'll have a look over the updated aspects + test on my system then submit.
Bob - Please assign issue to me, though Pete needs the credit for this,

Steve.

icklesteve, 2018-07-06:
Worked well ta Pete, modified xml file uploaded above with revision history + updated the htmls file supporting the definition

closed 2018-07-27 by bobjacobsen