JMRI/JMRI #5523: Create Test Release 4.11.9

bobjacobsen, 2018-07-05:
(If) we need to add changes before releasing JMRI Production Release 4.12, they go here first.

- [PRs labelled with the 4.13.1 milestone](https://github.com/JMRI/JMRI/milestone/57)

- [Issues and PRs (currently) marked as "Add to Next Release"](https://github.com/JMRI/JMRI/pulls?q=is%3Apr+is%3Aopen+label%3A%22Add+to+Next+Release%22)

- [Issues (currently) marked as "Add to Next Release"](https://github.com/JMRI/JMRI/issues?q=is%3Aissue+is%3Aopen+label%3A%22Add+to+Next+Release%22)

Final contents: (to be filled out)

bobjacobsen, 2018-07-06:
Merged #5514, #5525, #5512

bobjacobsen, 2018-07-06:
[Jenkins Packages 2568](http://jmri.tagadab.com/jenkins/job/Development/job/Packages/2567/) or later will have those updates.  Git hash is d6e27a4562d31f163f2478fea3f8b725f2576574

dsand47, 2018-07-06:
#5512 and #5525 work properly in Jenkins 2568.

Pugwash1, 2018-07-06:
> ---------- Original Message ----------

Hi
#5514 looks good, in 2568.
Steve G.

bobjacobsen, 2018-07-07:
@dsand47 @Pugwash1 Thank you!

bobjacobsen, 2018-07-09:
[image]

closed 2018-07-13 by bobjacobsen