JMRI/JMRI #5528: TCS WOW with caching and/or confirm writes

mmosher5501, 2018-07-06:
TCS WOW four CV programming fail if either
Allow caching when writing index CVs for read or write operations
When possible, confirm CV writes by following them with a read
is enabled.  
Would it be possible to disable those while doing a T2CV or T3CV operation then restore the settings after finished?
If not then I can add note telling user to disable those settings.

bobjacobsen, 2018-07-16:
The run-time decision to use or not use confirm-reads is in java//src/jmri/jmrit/symbolicprog/tabbedframe/PaneProgFrame.java at line 479, in a block called "set the programming mode”

The caching decision starts a few lines later and gets passed into the jmri.implementation.ProgrammerFacadeSelector. It could be intercepted in PaneProgFrame or handled lower down, but I recommend the former.

So I’d implement this by adding another item to the <programming> element to control it, which is then picked up there to control those two options.  

But I don’t have time to do it for a while.

mmosher5501, 2018-07-30:
OK, I've added note to WOW definitions to turn those features off.  Once an automatic temporary disable of those features is implemented, I will modify the note for that.

closed 2019-09-03 by mmosher5501