JMRI/JMRI #5529: Sensor debounce input fields use inconsistent forms

KenC57, 2018-07-07:
Within the sensor edit, and possibility other places, it displays with a separator anytime the value is more than 999. The spinner (inc/dec arrows) don't work when it goes past 999. Peter reported that there is also a field in the block edit that has similar issues.

The entry field in the sensor edit, debounce tab, the box size it fixed and limits the display to under 10 seconds.

The sensor table display doesn't use the separators and allows larger number entries.

Pugwash1, 2018-07-07:
I took a look see and found:
Tools->table->sensors. The headings are wrong, they use the word "Delay", its not a delay its a "De-bounce" or "Noise Filter'.and using it as a delay is actually dangerous as a fast train in a short block may never be seen, typical values to do its job are normally sub second. Directly editing the table in the grid is horrible, it crashes unless you use only pure long numeric, so 1,000 or 1000.1 gets it twisted underwear. Negative values can be entered, turning the air blue around me.
Using the "Edit Button" uses the same panel as editing the block in the Layout Editor. The wording is confused, it uses both "De-bounce" and "Delay". It only allows values between 1,000 and 1 (unless you start with -10,000, that gives it a teeny weeny headache.
 I would propose that the wording be cleaned up, "Noise Filter Tolerance" might be a better description, if less succinct, at least in English.  If people need a delay on reporting its state that should be an added attribute. The max, min, realistic values should all come from a single source, like, maybe, the appropriate sensor manager, and editing 101 rules applied.

devel-bobm, 2018-07-07:
@Pugwash1,

In my opinion, "Delay" is actually an applicable term here.  Logic cannot predict whether a "bounce" will occur.  This is true for both hardware and software debouncing solutions.  That means that an input value change -cannot be propagated- until it is known that it has stabilized for a given period of time.  

That "given period of time" is effectively a -minimum- delay from the initial input value change to the value being propagated to the logic which looks at the "debounced" output.

In my opinion, it might be more sensible to -clearly document- what we JMRI does for "debouncing", and provide some easy mechanism for users to get to that documentation while running the program, such as a link on the sensor configuration sheet which allows configuration of sensor "debouncing".

pnbgit, 2018-07-07:
For block detectors, I use an active debounce of 100mS and an inactive of 2 seconds.  This debounces locos losing contact on frogs etc. (whilst continuing on their keep alives) and provides a  sensible holdoff as the train leaves the block. It has the beneficial side effect that non-detectable stock triggers the sensor every second or so as the wheels bridge the block join, keeping the block detector active as the "non-detectable" stock passes through. As the on value is sub-second, no risk of the train not being seen in the block. So for that application, the option to set debounce figures over any desired range should be fixed - and there are valid cases, such as this, for values of several seconds.

pabender, 2018-07-07:
Sent from my iPhone

> On Jul 7, 2018, at 12:44 PM, Bob Milhaupt <notifications@github.com> wrote:
> 
> In my opinion, it might be more sensible to -clearly document- what we JMRI does for "debouncing", and provide some easy mechanism for users to get to that documentation while running the program, such as a link on the sensor configuration sheet which allows configuration of sensor "debouncing".
> 
I think the documentation does pretty clearly state 

http://jmri.sourceforge.net/help/en/html/tools/Sensors.shtml

The documentation pretty clearly states that debounce is intended to deal with noise when the input signal changes state.

Longer delays really should be put into place using higher level logic, not by manipulating the debounce settings on the sensors.

pnbgit, 2018-07-07:
> Longer delays really should be put into place using higher level logic, not by manipulating the debounce settings on the sensors.

Why?  Using a longer delay on the debounce settings works perfectly - I don't see why its use should be disallowed. If we disabled doing so, then we would break many user's perfectly good panels for no good reason.

pabender, 2018-07-07:
Sent from my iPad

> On Jul 7, 2018, at 3:03 PM, Pete Brownlow <notifications@github.com> wrote:
> Why? Using a longer delay on the debounce settings works perfectly - I don't see why its use should be disallowed. If we disabled doing so, then we would break many user's perfectly good panels for no good reason.
> 
Debouncing is about making sure the input does not oscillate between high and low.  Inputs to logic that bounce re not good for making logical decisions.  Users can set longer delays, but we don’t guarantee they will work for all desired situations.

As noted by Ken in his thread on the developers list, JMRI generates warning messages that tell you the input isn’t stabilizing during the debounce period.  The correct solution to applications such as the one Ken outlined is to use debouncing of the input AND a higher level logic timer.

Pugwash1, 2018-07-07:
I totally agree.
But, as most people have found long off debounce tends to work as a delay that they find benifical. Long ons can result in the train never being recorded as ever being present in the block. I see no reason to limit the input values, beyond not being negative, I just think adding a specific setting for delay before reporting would be benificial and fairly simple to implement. 
So, dont report unless consistant for x time, and when consistant dont report for y time, but still report even if sensor has since changed. X would normally be small, but could be any value, and y as big or small as you like. 
Its not difficult to do, any one can manipulate as they like. 
Steve G.

On July 7, 2018 3:43:34 PM EDT, Paul Bender <notifications@github.com> wrote:
> Sent from my iPad
>
> wrote:
> I don't see why its use should be disallowed. If we disabled doing so,
> then we would break many user's perfectly good panels for no good
> reason.
> Debouncing is about making [...]

-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

pabender, 2018-07-07:
@Pugwash1 we design tools to do a specific job.  If someone chooses to use that tool for some other purpose, we can't guarantee that will work.  ( it's like using a screwdriver for s hammer.  Sure, there are cases where that works, but there are more cases where it doesn't.)

In any case, the real issue here is about making sure the user interface is consistent, not about changing the underlying behavior.

There isn't anything we

devel-bobm, 2018-07-07:
"So I think, before this thread goes any further: is there any evidence that there's really a problem here that has been encountered by a user in normal use?"

@bobjacobsen - yes there are two real JMRI problems reported in the original post: 
- a problem when editing a sensor and playing with the "inactive delay" and/or "active delay".  JMRI is apparantly limiting the value to 1000 maximum.  If one wanted a larger "active delay" or "inactive delay", one -could not- set it using the JMRI "edit sensor" feature - it seems to get rounded down to the maximum value of "1,000".

- a problem where a sensor (created by an (unknown) older JMRI version, or read of an XML) has an "inactive delay" and/or "active delay" greater than 10000.  In this case, the `JTextField` portion of the `JSpinner` is too narrow to show the entire value.  On my machine, "10,000" is entirely visible, but "100,000" would likely not be entirely visible.

These are reproducible problems in current "head of master" code.

Whether the use-case of "debounce delay" of more than 999 mSec or more than 10000 mSec or more than 100000 mSec) are "normal use" is debatable.  Some posts on this "issue" reveal users with "inactive delay" values greater than 1000 mSec.  

Is 1000mSec a resonable upper limit?  It seems not.  And it seems silly to me to limit the values to a maximum of 1000mSec whenever entered via the JSpinner, but to support values greater than 1000mSec when they are configured by some mechanism other than the GUI `JSpinner`.

But what should the upper limit be? I'll leave that debate to others.

KenC57, 2018-07-08:
I would say the tool works fine, even if I'm using a number of zeros in the numbers. I only wanted the out to change when the input had remained stable for some period. So function was doing just fine.

The input fields didn't allow the number sizes I needed. As others have stated, delays of seconds is very common for many of our inputs. I've known users that were using 5 to 15 seconds for the off time in many applications to make up for not detecting all stock. But this was before this was coded. They were using my script to do the debouncing.

Having taught real time programming using the model of the payroll cycle, thinking that concepts like that should be very short, like less than a second to be real time are limiting themselves very badly. My class examples just used different time scales for many things. Some of the items used minutes, hours, days, weeks, months and years. Yet all filled the concept of real time, if you don't meet the time constraint, bad things happen. Miss getting the weekly checks out on Thursday is just as bad as the spark for an engine being 100 mSec behind schedule.

KenC57, 2018-07-08:
The one thing I wanted to add is I suspect it didn't show up before because
something changed elsewhere that introduced the separators into the number
display of the spinner. Since some fields use separators and some don't, I
suspect other common changes were behind this. It would take some time to
back step to see which major changes had the values without separators (or
spinners). So I'm guessing one of the presentation cleanups that have cycled
through from time to time.

In general, I think it would be nice if the spinner didn't have the
separator issue. I'm thinking some other parts of the code may have more
than 999 possible in a spinner.
-ken c

bobjacobsen, 2018-07-08:
The upper bound on those [JSpinners](https://docs.oracle.com/javase/8/docs/api/javax/swing/JSpinner.html) are loaded when they're created in java/src/jmri/jmrit/beantable/beanedit/SensorDebounceEditAction.java
```
        sensorDebounceInactiveSpinner.setModel(
                new SpinnerNumberModel(Long.valueOf(0L), Long.valueOf(0L), Long.valueOf(1000L), Long.valueOf(1L)));

        sensorDebounceActiveSpinner.setModel(
                new SpinnerNumberModel(Long.valueOf(0L), Long.valueOf(0L), Long.valueOf(1000L), Long.valueOf(1L)));
```

You can either change those `Long.valueOf(1000L)` values to something larger, or provide `null` to have no limit at all.  (Providing an arbitrary really large value is not compatible with the Spinner UI guidelines)

bobjacobsen, 2018-07-08:
Looking back at this, I think there's confusion about what "debounce" means.
[image]

Consider two different input sequences, shown in black above.  One starts low, bounces back and forth and ends up high, as on the left.  The other starts low, bounces back and forth, and ends up low, as on the right.

The middle line in blue is what JMRI outputs as "debounced".  From the 1st edge, it waits to see if the sequence ends up changing after Delta T.  If so, it provides the changed Sensors output, as on the left.  If it doesn't as shown on the right, the state provided by the Sensor doesn't change.  This is _debouncing_:  The changing values on the real input are suppressed on the output.  But that output's change is delayed by Delta T to make sure.

Long values of Delta T are generally not appropriate here.  You should _not_ have multi-second debouncing on your occupancy inputs, for instance, because then your signals will respond un-prototypically slowly.

The red line shows a different behavior, called "stretching" in some control systems.  On the left, it responds immediately, which would seem to be a good thing.  And it "stretches" the short, transient transitions on the right.  This can be useful for making sure that transient inputs extend for long enough to be noticed; perhaps long times are appropriate here.    But it's _NOT_ debouncing, because a bouncing contact in the right-case wasn't debouncing but rather the opposite: _More_ attention was paid, not less.

A "stretch" behavior is a perfectly fine thing; there are clearly use cases for it. But it's not what this feature does.

Can somebody please point to a use case for having a debounce time over a second?

silverailscolo, 2018-07-08:
I generally understand what's already been said. I made the changes 11 months ago to make use of current Java features that take care of entry checking and i18n.
Inconsistency in display format can be fixed (in 2 ways ;-) and the wording in the help page should be improved.
Being able to edit directly in the table seems a useful feature, but it needs at least some entry validation (the Edit Block length in the Block Table has that in place; JSpinners take care of this on the Sensor and Block Edit panes and could also be used as table cell editors).
The thousands separator (automatically localized to display as a period (.) in many other Locales) is especially usefull for higher values, and just like the decimal separator, could also be applied to the Sensor Table view. It's not stored and you don't have to enter themAFAIK.
Using the word "delay" regarding debounce settings certainly causes some confusion. Would Debounce _Period_ or - _Time_ be an improvement? I agree it is not the function of the Sensor item to add some Stretching ("delay") before reporting - _after_ determining a state change was detected. However in many places in JMRI the user can select a Sensor to trigger eg. Turnout feedback, but not a Stretch/Delay value for that trigger. So we either have to add that or guide towards using an Internal sensor that listens to the hardware Sensor and does add the Stretch/Delay.
It would be easy to increase the max value in both places, when we agree on a value. For nearly a year it is limited to 10 x what one would expect for hardware button or block detector bouncing characteristics.

pnbgit, 2018-07-08:
Bob said:

> So I think, before this thread goes any further: is there any evidence that there's really a problem here that has been encountered by a user in normal use?

No problem with the sensor debounce implementation - please leave it as it is.

The only problem is a regression in the UI that no longer allows values of more than 1000 to be entered in the spin boxes.  If we remove that restriction, which seems a simple fix, then everything is as it was before and all will be well.   

If users choose to use larger values to get an effective delay after the last actual state change, and it works fine for those users, then there is no need for us to try and stop them by putting in artificial restrictions.  

For example, as already mentioned, I use sensor off debounce values of 2 to 3 seconds to hold off the release of block occupancy, it works perfectly and forcing me to add higher level logic to do this would just add unnecessary complexity when the current simple solution works well.

devel-bobm, 2018-07-08:
A slightly larger view around the code @bobjacobsen posted reveals the width of the `JTextField` associated with the `JSpinner`.  Tweaking the parameters there could resolve @KenC's second issue (visibility of values with more than 4 digits).

```
        sensorDebounceInactiveSpinner.setModel(
                new SpinnerNumberModel(Long.valueOf(0L), Long.valueOf(0L), Long.valueOf(5000L), Long.valueOf(1L)));
//        sensorDebounceInactiveSpinner.setValue(Long.valueOf(0L)); // reset from possible previous use
        sensorDebounceInactiveSpinner.setPreferredSize(new JTextField(5).getPreferredSize());
        basic.addItem(new BeanEditItem(sensorDebounceInactiveSpinner, Bundle.getMessage("SensorInActiveDebounce"), Bundle.getMessage("SensorInActiveDebounceText")));
        sensorDebounceActiveSpinner.setModel(
                new SpinnerNumberModel(Long.valueOf(0L), Long.valueOf(0L), Long.valueOf(1000L), Long.valueOf(1L)));
//        sensorDebounceActiveSpinner.setValue(0L); // reset from possible previous use
        sensorDebounceActiveSpinner.setPreferredSize(new JTextField(5).getPreferredSize());
```

devel-bobm, 2018-07-08:
I am not convinced that the @bobjacobsen drawing is either complete or correct for "debouncing".  And my review of JMRI code leaves me wondering exactly what behavior JMRI has when its "debouncing" features are in-use.
I will file this as a separate issue, to avoid cluttering this issue.

pabender, 2018-07-08:
@devel-bobm, @bobjacobsen’s description of debounce matches my understanding of the process, along with the reasons we want debouncing.

Incidentally, I have been going through the documentation and it appears that the sensor table documentation and the block edit documentation conflate the idea of delay and debounce.  These are not the same thing, and shouldn’t be treated as such.

Also, does the block edit window sensor tab change the sensor or does it add additional properties?  It seams to me that blocks should be able to aggregate sensors, but there is a one to one mapping.  ( you actually can aggregate detection sections into a single layout editor block, which is not the same as the named bean blocks )

pabender, 2018-07-08:
Just for reference here is a YouTube video that shows how debouncing can be handled in an arduino.

https://m.youtube.com/watch?v=jYOYgU2vlSE

I am posting this because it includes an oscilloscope trace showing the debounce behavior on the left hand side of @bobjacobsen’s drawing

devel-bobm, 2018-07-08:
@pabender, @bobjacobsen,

The JMRI implementation, and the @bobjacobsen diagram shown above, do not reject "spurious input changes".  In an electrically-noisy environment, such spurious changes would result in JMRI perceiving a change from the input, and allow triggering of dependant logic.

The "debouncing" implementations I worked with were intended to address"noisy" environments, and would reject "spurious input changes".  One might not expect problems from "electrical noise" on pushbutton input signals in small personal electronic devices, but it is indeed a real problem, and our functional requirements forced design implementations to "filter-out" such spurious noise.

Note that I have seen JMRI problems from "spurious noise" issues.  The JMRI "debounce" implementation was ineffective in identifying or resolving the issues.  A Logix implementation was able to resolve the issues.  Examination of the inputs with an oscilloscope helped identify and resolve the noise sources, and the Logix has been unneeded since.  I believe that a JMRI implementation of "debouncing" similar to the ones I had previously worked with in silicon would have avoided the "spurious noise sensitivity" issues altogether...

silverailscolo, 2018-07-08:
To keep the discussion on the subject of debounce, the YouTube video very to the point states: "We could wait _a few milliseconds_ in order to make sure the (hardware) button was pressed". Using the term "regression" in this thread sounds quite negative to describe the effect of code changes and disregards its intent. Let's call it an unexpected side effect for a different use case, deliberate how we wish to address it and move on. Where would a Wait Delay timer ideally be set, separate from the hardware sensor?

pnbgit, 2018-07-08:
Actually, if you look the original opening of this issue by Ken, it is not about the JMRI implementation of debounce, what words we should use to describe what, or what people think we should be allowed to use it for.   The issue as opened is simply the regression that the spin boxes don't work as they used to, they now add a comma over 1000, are now not wide enough to display the value properly and have a new arbitrary maximum validation attached.  If we can fix that regression, so the spin boxes work properly as they used to, then this issue as opened is dealt with.  The semantics of debounce can then be discussed separately.

bobjacobsen, 2018-07-08:
> The JMRI implementation, and the @bobjacobsen diagram shown above, do not reject "spurious input changes". In an electrically-noisy environment, such spurious changes would result in JMRI perceiving a change from the input, and allow triggering of dependant logic

Wrong.  See the right hand behavior of the blue line on the diagram I posted.

> The "debouncing" implementations I worked with were intended to address"noisy" environments, and would reject "spurious input changes". One might not expect problems from "electrical noise" on [...]

Right. That’s what the implementation is intended to do. That’s what it does, and does quite nicely, with a short timeout.  

Long timeouts result in long delays before inputs are noticed.  That’s not a problem with the algorithm nor implementation, it’s a problem of misuse of the feature for an unintended purpose.

> Note that I have seen JMRI problems from "spurious noise" issues. The JMRI "debounce" implementation was ineffective in identifying or resolving the issues.

I’d be interested in getting more detail on that.

> A Logix implementation was able to resolve the issues. Examination of the inputs with an oscilloscope helped identify and resolve the noise sources, and the Logix has been unneeded since. I believe [...]

And on that, too.  How does that “previously worked with” algorithm work?

pabender, 2018-07-08:
> On Jul 8, 2018, at 5:52 PM, Bob Jacobsen <notifications@github.com> wrote:
> Long timeouts result in long delays before inputs are noticed. That’s not a problem with the algorithm nor implementation, it’s a problem of misuse of the feature for an unintended purpose.

I agree with @bobjacobsen.  

Also, to address this point and @pnbgit’s concern about how we are straying from the original topic, the reason the documentation becomes an issue is that the conflation of delay and debounce makes it seem that longer delays are acceptable, when they really aren’t.  We NEED to make sure users understand that.  Part of that is by fixing the user interface ( yes, to limit the time to under 1 second ) another part of this is fixing the documentation.  

There may be a third part of the fix which is making sure users understand how they should be getting delays either in the active or inactive side of the input.  We have tools for that already ( scripts, logix, etc ) but that doesn’t mean there isn’t room for some other way to do it.  I feel pretty strongly that action delays based on change of state should be something determined by the consumer of the information provided by the sensor, not the sensor itself.  Debouncing is not about delayed action, but to prevent unintended outputs from occurring.

> > Note that I have seen JMRI problems from "spurious noise" issues. The JMRI "debounce" implementation was ineffective in identifying or resolving the issues.
> 
> I’d be interested in getting more detail on that.

I would be too, especially if that information provides us with a repeatable test we can add to the test suite.

To that end, I am in the middle of refactoring the sensor tests, so we have a single place to put this.  I should be able to finish that work tonight.

The existing abstract sensor test has tests for debounce, but it only handles single changes of state.  I am planning on adding tests that verify what happens with both an even and odd number of state changes in the debounce period.

Paul

dheap, 2018-07-09:
> @devel-bobm The "debouncing" implementations I worked with were intended to address"noisy" environments, and would reject "spurious input changes". One might not expect problems from "electrical noise" on pushbutton input signals in small personal electronic devices, but it is indeed a real problem, and our functional requirements forced design implementations to "filter-out" such spurious noise.

My understanding of "debounce" (based partly on 21 years of my working life designing and building custom electronics equipment) was just as @bobjacobsen drew in his left-hand drawing. The term "debounce" was based on the fact that pushbutton (and other switch) contacts often (should be assumed to) actually bounce on closure, causing spurious transitions. In many a design I had to consider and (often) include debounce for the pushbuttons. This was not necessarily "spurious noise" filtering, although the two may be combined in a filter design.

In my view:
- "spurious noise filtering" is the removal of unwanted transitions (usually due to electrical noise) that are not really a valid state change. See @bobjacobsen RHS drawing.
- "debounce" is the prevention of unwanted transitions (due usually to mechanical bounce issues) after a valid state change from registering as multiple transitions/counts. See @bobjacobsen LHS drawing.

pabender, 2018-07-09:
@dheap In model railroads, we get something that looks like bob's RHS drawing for mechanical reasons.

For current detection, this noise is caused by wheels rotating, especially when they hit dirty spots on the track.

For optical detection, this is caused by the gaps between cars.

dheap, 2018-07-09:
@pabender Agreed, but thats a candidate for stretching as per @bobjacobsen lower drawing rather than debounce as per middle drawing.

But in our application, judicious application of stretching time may cover both cases and avoid the on-delay of Bob's middle row. But we have to consider the case you mentioned where the output is high to start...

My head needs a break tonight. Been in problem-solving mode all day...

pnbgit, 2018-07-09:
> longer delays are acceptable, when they really aren’t. We NEED to make sure users understand that. 

I absolutely disagree.

Longer OFF delays are working perfectly on layouts I have helped with.  It is just a conceptual problem people are having with this.  They are perfectly acceptable IMHO. Why break something that is working well for many just over the semantics of what we call it?

pabender, 2018-07-09:
@dheap,

Sent from my iPad
> On Jul 9, 2018, at 3:18 AM, Dave Heap <notifications@github.com> wrote:
> @pabender Agreed, but thats a candidate for stretching as per @bobjacobsen lower drawing rather than debounce as per middle drawing.
> 
> But in our application, judicious application of stretching time may cover both cases and avoid the on-delay of Bob's middle row. But we have to consider the case you mentioned where the output is high to start...
> 

The on delay is intentional. That is the only way to tell the difference between the right and left hand sides in debouncing.

Stretching is not the right behavior.  As an example, consider a single car with a single detectable wheelset rolling through a current sensing bock.  

If the car rolls through a dead frog on a turnout, the raw sensor state SHOULD switch from active to inactive while the detectable wheelset is on the frog.

If you use debouncing, as it is currently implemented in JMRI, the known sensor state will stay active so long as the car is moving fast enough that it traverses the dead spot before the end of the debounce period.

If we were using stretching in this case, the known sensor state would switch from active to inactive as soon as the wheelset hit the frog, and it would stay that way for either 1) the length of time it takes to traverse the frog or 2) the stretch time, whichever is longer.

So stretching really isn’t the answer.

closed 2018-07-27 by bobjacobsen