JMRI/JMRI #5530: Selecting Edit is unreliable on rotated text, and the string is truncated - Layout Editor 4.11.8

DaveTur, 2018-07-09:
Panel Pro Layout Editor v 4.11.8
7 July 2018

Label text that has been rotated using Orientation -> Vertical Up is very difficult to select (using right-mouse or Shift-right-mouse click) to display the options menu.  On a 14 char upper case string "ABCDEFGHIJKLMN" the string has to be clicked towards the top (on or after the "M") to display the options menu.  This string is displayed trucated, the top 70% part (after rotation) of the final letter 'N' is not displayed.  
When rotated to 'Vertical Down' most of the 'N' is not displayed and the string has to be clicked on or above the 'B' to display the options menu.
Padding the string with 2 trailing spaces allows all the letters to be displayed but does not alter the length of the string that can be clicked to display the options menu.
Un-rotated label text is also displayed truncated, but the options menu can be displayed by clicked anywhere on the string, and (incorrectly) up to about 4 chars to the right.

dsand47, 2018-07-09:
The calculation for height and width of a text label does not check the orientation of the label.  This results in the selection box being at the top at of the label and to the right.  The rotation option works properly.

The fix will be in the 4.13.1 test release.

I am not able to re-create the truncation problem.  When I change font size, the image changes as expected.

dsand47, 2018-07-09:
Pull request #5535 has been submitted.

bobjacobsen, 2018-07-10:
Fixed by #5535

closed 2018-07-10 by bobjacobsen