JMRI/JMRI #5531: Inconsistent editing of slips - Layout Editor 4.11.8

DaveTur, 2018-07-09:
Inconsistent editing of slips
Layout Editor 4.11.8
Java 1.8.0_171 (en-AU)
Windows 10
	
I have been attempting to change the geometry of a slip (single or double) to reduce the crossing angles. I have not found instructions about which keystrokes should be used, so I have assumed that the published keystrokes for normal turnouts should be used.  However right mouse click editing of slips seems buggy :-

Both right click and shift-right click act the same, but not the same as for turnouts. Dragging the whole slip is OK, but after this all left or right mouse clicks cause one of the connection points to jump to that position, after which any further editing of the panel is impossible without terminating then re-invoking the program.

geowar1, 2018-07-09:
This is a know issue that only happens after you command-drag a slip; after that attempting to move the 4th (“D”) leg gets stuck and follows the mouse… DaveS and I have it in our queues to investigate further.

dsand47, 2018-07-09:
A slip has 4 points that can be moved.

[image]

To change the geometry of the slip drag either A or B.  C and D will move automatically to maintain the angle and length.  Or use C and D and A and B will follow.

Changing the geometry of a level crossing is exactly the same.  Drag the end points.

As George mentioned, if you drag a slip, the D point is stuck to the mouse.  After dragging, drag the D point a tiny bit and the mouse will be released.

geowar1, 2018-07-09:
What DaveS said plus… the option[/alt] key will invert the “align to grid” setting during drags.

DaveTur, 2018-07-11:
Thanks to George and DaveS for taking this issue on board.  I now see that point D can be unstuck from the mouse by simply returning it to the original position using right drag.  Also when using Alt-right drag to invert “align to grid”, point D does not follow the mouse.  Without looking at the code, looks like a break could be missing from a switch statement.

DaveTur, 2018-07-11:
Oops, I'm a newbie and close the issue in error - now reopened.

closed 2021-04-14 by stale[bot]