JMRI/JMRI #5540: Move Sensor User Name from one system name to another is (still) Broken.

bobjacobsen, 2018-07-12:
From David Parks <bearwestern@comcast.net> on jmriusers:

> Move User Name updates Logix references, but does not update the Panel Icons resulting in an inconsistent panel file.  This has been broken for several JMRI versions.  Move User name is the primary method of changing an external connection since this is tied to the system name prefix and it is not editable in the user Tables.

A quick look at the code shows inconsistent use of NamedBeanHandles.  Not sure whether it ever worked or not.

KenC57, 2018-07-12:
That sounds the opposite of what had been happening. At one point I recall
that it worked EXCEPT in Logix. For a long time it worked well in panels and
tables other than Logix. But that was about a year or two ago when I last
recall this topic.
-ken c

dsand47, 2018-07-12:
PR #5367 fixed a problem with named bean handles.  Since then Logix, sensor icons, etc. all seem to work.  

The sensor icon does use NamedBeanHandles.

I suspect that the issue is that the **_move_** does not **_appear_** to work since the tool tip was not changed.

dsand47, 2018-07-12:
Is there any value in having an automatic tool tip capability?

I have created a proof of concept that appears to work.
- A NamedBeanHandle field is added to ToolTip along with a method to update the field.
- If the field is not null, the standard **_user name (system name)_** format is returned using the current bean contents.  Otherwise the current text value is returned.
- When a tool tip is created, the handle will be set if appropriate for the item type.
- If the tool tip is manually changed, the handle field will be set to null.
- During XML loading, the handle will be set depending on whether the calculated tool tip matches the xml supplied tool tip.  This prevents replacing a manual tool tip that has been saved.

Dave Sand

KenC57, 2018-07-12:
I'd say it is nice to have. Saves having to open a table or right click edit
on a panel item to find out what it is.
-ken c

closed 2021-04-14 by stale[bot]